flipperdevices/flipperzero-firmware #3219: file rfal_rf.h not found

dineshkrishna9999, 2023-11-16:
### Describe the bug.

flipperzero-firmware$ ./fbt fap_seader
        CC      applications_user/seader/scenes/seader_scene_card_menu.c
        CC      applications_user/seader/scenes/seader_scene_credential_info.c
        CC      applications_user/seader/scenes/seader_scene_delete.c
        CC      applications_user/seader/scenes/seader_scene_delete_success.c
        CC      applications_user/seader/scenes/seader_scene_file_select.c
        CC      applications_user/seader/scenes/seader_scene_read_14a.c
        CC      applications_user/seader/scenes/seader_scene_read_card_success.c
        CC      applications_user/seader/scenes/seader_scene_read_picopass.c
        CC      applications_user/seader/scenes/seader_scene_sam_info.c
        CC      applications_user/seader/scenes/seader_scene_sam_missing.c
        CC      applications_user/seader/scenes/seader_scene_sam_present.c
        CC      applications_user/seader/scenes/seader_scene_sam_wrong.c
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_delete.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_file_select.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
compilation terminated.
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_sam_info.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_delete.o] Error 1
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_sam_info.o] Error 1
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_file_select.o] Error 1
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_delete_success.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_delete_success.o] Error 1
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_card_menu.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_card_menu.o] Error 1
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_read_14a.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_read_card_success.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_read_14a.o] Error 1
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_read_card_success.o] Error 1
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_sam_present.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_read_picopass.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_sam_present.o] Error 1
compilation terminated.
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_read_picopass.o] Error 1
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_credential_info.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_credential_info.o] Error 1
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_sam_wrong.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_sam_wrong.o] Error 1
In file included from applications_user/seader/scenes/../ccid.h:10,
                 from applications_user/seader/scenes/../seader_i.h:40,
                 from applications_user/seader/scenes/seader_scene_sam_missing.c:1:
applications_user/seader/scenes/../seader_worker_i.h:12:10: fatal error: rfal_rf.h: No such file or directory
   12 | #include <rfal_rf.h>
      |          ^~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-D/.extapps/seader/scenes/seader_scene_sam_missing.o] Error 1

********** FBT ERRORS **********
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_delete.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_sam_info.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_file_select.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_delete_success.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_card_menu.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_read_14a.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_read_card_success.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_sam_present.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_read_picopass.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_credential_info.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_sam_wrong.o: Error 1
build/f7-firmware-D/.extapps/seader/scenes/seader_scene_sam_missing.o: Error 1

### Reproduction

.

skotopes, 2023-11-16:
@bettse looks to be your customer ;-)

closed 2023-11-16 by skotopes

bettse, 2023-11-16:
💯 Addressed here: https://github.com/bettse/seader/issues/3#issuecomment-1813816195