flipperdevices/flipperzero-firmware #3222: Black Magic wont compile to wifi devboard

aidenhyde, 2023-11-16:
### Describe the bug.

So i am using your guide to install black magic to the wifi devboard WD.1
When i run the command py -m ufbt devboard_flash
i get the error of wifi board not being found. It shows up in my device manager.

### Reproduction

1. Install uFBT.
2. Connect the wifiboard to pc
3. run the command py -m ufbt devboard_flash

### Target

Wifi Devboard

### Logs

```Text
PS C:\Users\HydeAiden>  $PSVersionTable

Name                           Value
----                           -----
PSVersion                      5.1.22621.2506
PSEdition                      Desktop
PSCompatibleVersions           {1.0, 2.0, 3.0, 4.0...}
BuildVersion                   10.0.22621.2506
CLRVersion                     4.0.30319.42000
WSManStackVersion              3.0
PSRemotingProtocolVersion      2.3
SerializationVersion           1.1.0.1

PS C:\Users\HydeAiden> py -m ufbt devboard_flash
scons: Entering directory `C:\Users\HydeAiden\.ufbt\current\scripts\ufbt'

fbt: warning: Folder app: manifest application.fam is missing
LoadAppManifest, line 33, in file "C:\Users\HydeAiden\.ufbt\current\scripts\fbt_tools\fbt_apps.py"
python C:\Users\HydeAiden\.ufbt\current\scripts/wifi_board.py
2023-11-16 10:09:09,804 [ERROR] WiFi board not found
2023-11-16 10:09:09,804 [INFO] Please connect WiFi board to your computer, hold down BOOT button and press RESET button
scons: *** [phony_devboard_flash] Error 1
PS C:\Users\HydeAiden>
```

hedger, 2023-11-16:
Please show how does your device manager displays it.

aidenhyde, 2023-11-16:
[image]

ThirteenTX, 2023-12-29:
I have the same issue, and it also shows under "Other Devices" as ESP32-S2 with an error icon.
[image]
[image]
Under the ports section od device manager, mine looks the same, except it is COM9 instead of 6. I confirmed the ID it shows is an ESP ID in the device manager. Hope this helps. Not sure what to try next.

skotopes, 2024-01-10:
@hedger @DrZlo13 any progress?

CookiePLMonster, 2024-01-29:
Just chiming in as I have the exact same issue.
[image]

closed 2024-03-01 by hedger