flipperdevices/flipperzero-firmware #3225: SubGHZ receive while viewing a key does not work, despite blue light flashing.

LeeroysHub, 2023-11-17:
### Describe the bug.

When you are receiving keys that are decoded in SubGHZ Read, you can view them. When you are viewing a key, the light is still flashing blue, indicating its still receiving. 

The keys are not received and added to the list. I can see that the SubGHZ Read worker is still running, and the radio is acutally on. But the user does not care if "the radio is on," they care if they are receiving keys! 

Either the light should be turned off while viewing a key, or the code fixed that keys are added to the list even when on other scenes. 

### Reproduction

1) Go into SubGHZ Read.
2) Receive a key, watch it add to the list.
3) View the key, note the blue light on indicating receive.
4 ) Transmit another key.
5) Press back and view received keys. Key transmitted while viewing will not be on the list. 

### Target

Flipper Zero

### Logs

```Text
No logs, behaviour is on device!
```

### Anything else?

I have changed the scene to turn the Blue recieve light off while viewing a key, despite the radio being on. This allows the code to be fixed in the future, but makes it right for the user. This could be improved if the code is not ever going to be fixed, the radio could be stopped while viewing a key.

Thats just my temporary work around!

LeeroysHub, 2023-11-17:
I apologize, this is a CFW issue only by the looks.

LeeroysHub, 2023-11-26:
After re-installing OFW, I can confirm this bug is present in latest Dev version of OFW.

Skorpionm, 2023-12-02:
I don’t get the same error; when I look at it, new keys are added to the list. Do you happen to have Hopping turned on during your appointment?

LeeroysHub, 2023-12-02:
I am definitely not Hopping. How can it be adding to the list,the add_to_history_caklback() only gets called if receiverinfo is the active scene? If you are viewing a key to TX or Save, there is no code active to add a key.

I've confirmed with Dev Probe, all the CFW developers are aware of the issue as well.

LeeroysHub, 2023-12-04:
I don't have a link to click that video, where can I find that?

Skorpionm, 2023-12-04:
https://youtu.be/3ax980shNeU

LeeroysHub, 2023-12-04:
I'm so sorry, you guys fixed this a long time ago. Somehow, I've gotten the Flipper firmware confused with another one somehow. I was sure I checked properly this time! But there is the code blame, plain as day. The SubGHZ refactor looks great, Im excited to play with it... 

I'm installing OFW and keeping it, I'm sick of finding and fixing bugs that no longer exist in OFW! 

Thankyou, can close issue, and once again I apologise! This is my 2nd time reporting CFW bugs...