flipperdevices/flipperzero-firmware #3228: Re-open of #1923: Subghz raw sub transmits an unwanted signal at the beginning (OOK 433MHz) #1923

coded-with-claws, 2023-11-18:
### Describe the bug.

Hi,
Following closure of #1923, I retried the same sub file. There is stil an unwanted preamble (but not exactly the same as before).

- unwanted preamble
  - sends ON for duration ~1065 (selected in blue on the screenshot)
  - sends OFF for ~1010 (just after the selected blue)
- then applies the content of the raw sub (1700, -900, etc.).

I'm not sure it's really disturbing the receiving devices generally, but if it's fixable, why not :)
[example.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13400652/example.sub.txt)
[image]

Thanks :)

### Reproduction

Download the attached .sub.txt file, rename to remove the ".txt" (which was added because of github supported file types).
Open it in the Subghz application and hit Send, and capture in urh with a RTL SDR dongle.

skotopes, 2023-11-18:
Yep looks like 2nd transmission starts incorrectly. We'll fix it in next release.

Skorpionm, 2023-11-19:
@coded-with-claws  check fix plz https://github.com/flipperdevices/flipperzero-firmware/pull/3230

coded-with-claws, 2023-11-19:
@Skorpionm Nice your fix works great, I only get the signal configured into sub file now!! :) Thanks!! :+1:

closed 2023-11-22 by doomwastaken