flipperdevices/flipperzero-firmware #3231: NFC  not longer working

eloisasal, 2023-11-19:
### Describe the bug.

Since I upgraded to the last firmware 0.94.1, some nfc just cant be readed anymore and needs to be re-saved.
Not only that, I try to open the work door and just wont work more than one time.
All amiibo that I dumped is no longer recognised as an amiibo.

What happened?
I need to downgrade to keep using my (for now) most used features?

Im dont have logs of this because I still dont know how to do most of the things but. I think this is a big problem, especially because as a door work opener is a large part of their advertising that no longer works.

### Reproduction

Get firmware official 0.94.1
try to emulate some amiibo -> this object is not an amiibo
try to save and emulate a nfc work door -> will work first time, maybe for an hour, 100% next day not working.

Astrrra, 2023-11-21:
Please refer to https://github.com/flipperdevices/flipperzero-firmware/issues/3188 for the NTAG215 emulation.

TLDR: The dumps don't have a password and PACK in them, but the emulation worked because when the password was unknown, Flipper just replied with the default hardcoded PACK. This made the emulation work in some cases, but is incorrect from the technical point of view, and was removed in favor of more accurate emulation of NTAG215 cards. To make your cards work again, use the "Unlock NTAG/Ultralight" option from the "Extra Actions" menu in the NFC app. That will allow you to calculate the correct password and get the PACK from the tag, which you'll then be able to use with your systems. Other solutions can be found in the original issue linked above.

Astrrra, 2023-11-21:
As for the door - please provide more information about the tag you're using. Please include the model (Mifare Classic, Mifare Ultralight, NTAG, etc.), the manufacturer of the reader, and everything else you know. To help us better diagnose your problem, please follow these steps to capture all the necessary data:

0. Make sure you're on the latest firmware release.
1. Read your tag with the Flipper. Save it, name the file tag1.
2. Repeat step 1 two more times, saving files tag2 and tag3.
3. Touch your original tag to the reader, check that the door has opened.
4. Read your tag once again with the Flipper, twice. Save the resulting files as tag4 and tag5.
6. From the "Saved" menu in the NFC app, select tag2 and start the emulation.
7. Touch your Flipper to the reader, check that the door has opened.
8. Go through files tag1-tag5, emulate each, note which ones open the door and which don't.
9. Wait for the next day.
10. From the "Saved" menu in the NFC app, select tag3 and start the emulation.
11. Touch your Flipper to the reader, note whether the door has opened or not.
12. Read your original tag once with the Flipper, save the file as tag6.
13. Touch your tag to the reader, check that the door has opened.
14. Read your original tag once with the Flipper, save the file as tag7.
15. Go through files tag1-tag7, emulate each, note which ones open the door and which don't.

After that, please send us all the files (note that there may be more than 7, if the reader changes the data on the tag, then a .shd file will be saved, and we need all of them), and the two lists of what files opened the door at day 1 and day 2. Example:

- Day 1:
    - Tag1 - open
    - Tag2 - open
    - Tag3 - open
    - Tag4 - open
- Day 2:
    - Tag1 - closed
    - Tag2 - closed
    - Tag3 - closed
    - Tag4 - closed
    - Tag5 - closed
    - Tag6 - open
    - Tag7 - open
    
I understand that this is a lot, but this is required for us to collect enough information about your reader to know what exactly went wrong and how to fix it. Please follow this guide carefully and don't skip any steps, all of them are important for us.

skotopes, 2024-01-10:
Please retest on latest dev, there were bunch of fixes in emulation. Also amiibo cards need to be updated to include correct password. Check ameebo repo for that.

skotopes, 2024-01-10:
Please open new issue if problem persist.

closed 2024-01-10 by skotopes