flipperdevices/flipperzero-firmware #3235: NFC refactor causing emulation problems

Didgitalpunk, 2023-11-21:
### Describe the bug.

post update 0.94.1, the NFC emulation for amiibos is non-functional for me and other people according to the discord
I've also noticed emulation problems with Slix2 cards, as used on DYMO LabelMaker rolls as DRM. flipper is no longer detected, even with a new dump.

### Reproduction

- update to firmware 0.94.1
-- download the amiibo repo from https://github.com/Gioman101/FlipperAmiibo
-- emulate amiibo on flipper and try and read it with either joycon or pro controller
-- console never ends up reading the emulated tag

-- for SLIX2, first read and save the tag from a roll
-- put flipper against reader, inside the printer. Printer is constantly reading, so it should pick up the emulated tag.
-- printer never ends up reading the emulated tag

Astrrra, 2023-11-21:
Please refer to #3188 for the NTAG251 emulation. 

TLDR: The dumps don't have a password and PACK in them, but the emulation worked because when the password was unknown, Flipper just replied with the default hardcoded PACK. This made the emulation work in some cases, but is incorrect from the technical point of view, and was removed in favor of more accurate emulation of NTAG215 cards. To make your cards work again, use the "Unlock NTAG/Ultralight" option from the "Extra Actions" menu in the NFC app. That will allow you to calculate the correct password and get the PACK from the tag, which you'll then be able to use with your systems. Other solutions can be found in the original issue linked above.

Didgitalpunk, 2023-11-21:
Tysm Astra!
(cool name btw, and I like your profile pic on discord, makes me think of
Amity from The Owl House a bit)

Best regards
Freya
She/her

closed 2024-04-16 by gornekich