flipperdevices/flipperzero-firmware #3245: Sub-1GHz transmission frequencies in EU

MarSik, 2023-11-27:
### Describe the enhancement you're suggesting.

Flipper firmware restricts the transmit frequencies for the 868 MHz band to 868.150 MHz to 868.550 MHz. However the allowed SRD frequencies are much wider in the European Union according to the ETSI EN 300 220-2 V3.2.1 (2018-06) document.

https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_60/en_30022002v030201p.pdf

Notice rows K-Q on the page 22. Frequencies between 863 - 870 MHz should be allowed even though other limits might apply (power, duty cycle).

Can you please update the limits in the firmware or document where the strict restriction comes from?

Just FYI: The 433 MHz band is fully supported by FZ and the limits match the EN ETSI 300 200-2 standard.

arkis0, 2023-11-27:
Restrictions on flipper zero were made to make it legal in most of countries. Some countries might have stricter law than UE, that's why they made flipper so limited with Sub-Ghz.

MarSik, 2023-11-27:
The restrictions on Flipper are actually based on the country you are in when you update the firmware. The common frequencies and country specific frequencies are clearly separated in the document I linked.

Sure, I see that EU is currently in the same bucket as non-EU countries like Russia, Ukraine and other Asian countries (I wonder if that is the whole ITU zone 1). However that basically makes most of the legal 868 MHz signals unavailable to Flipper in the EU.

skotopes, 2023-12-04:
This requires new certification, which we are not planning.

closed 2023-12-04 by skotopes