flipperdevices/flipperzero-firmware #3248: Mifare Classic emulation regression after NFC refactor

dogtopus, 2023-11-28:
### Describe the bug.

The NFC refactor (at least partially) reintroduces bugs fixed by #2825. My phone (Pixel 7 Pro) can no longer read emulated Mifare Classic tags after the NFC refactor was merged.

### Reproduction

1. Emulate some Mifare Classic dump
2. Scan the tag
3. See it fail without anything being read

### Target

f7

### Logs

```Text
Nothing useful.
```

### Anything else?

Proxmark3 trace

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2228 |       4596 | Tag |44  00                                                                   |     | 
      12016 |      16784 | Rdr |50  00  57  cd                                                           |  ok | HALT
     873568 |     874560 | Rdr |52(7)                                                                    |     | WUPA
     875796 |     878164 | Tag |44  00                                                                   |     | 
     885584 |     888048 | Rdr |93  20                                                                   |     | ANTICOLL
     889220 |     895108 | Tag |04  e2  26  75  b5                                                       |     | 
     902464 |     912928 | Rdr |93  70  04  e2  26  75  b5  c8  c0                                       |  ok | SELECT_UID
     914164 |     917684 | Tag |08  b6  dd                                                               |  ok | 
    1108384 |    1110592 | Rdr |50! 00!                                                                  |     | HALT
    1113364 |    1114004 | Tag |00(4)                                                                    |     | 
    1197952 |    1200160 | Rdr |50! 00!                                                                  |     | HALT
    1202868 |    1203508 | Tag |00(4)                                                                    |     | 
    1278304 |    1280512 | Rdr |50! 00!                                                                  |     | HALT
    1283236 |    1283876 | Tag |00(4)                                                                    |     | 
    1342432 |    1344640 | Rdr |50! 00!                                                                  |     | HALT
    1347364 |    1348004 | Tag |00(4)                                                                    |     | 
    1763232 |    1764224 | Rdr |52(7)                                                                    |     | WUPA
    1765460 |    1767828 | Tag |44  00                                                                   |     | 
    1775248 |    1780016 | Rdr |50  00  57  cd                                                           |  ok | HALT
    2636864 |    2637856 | Rdr |52(7)                                                                    |     | WUPA
    2639092 |    2641460 | Tag |44  00                                                                   |     | 
    2648880 |    2651344 | Rdr |93  20                                                                   |     | ANTICOLL
    2652516 |    2658404 | Tag |04  e2  26  75  b5                                                       |     | 
    2665760 |    2676224 | Rdr |93  70  04  e2  26  75  b5  c8  c0                                       |  ok | SELECT_UID
    2677460 |    2680980 | Tag |08  b6  dd                                                               |  ok | 
    2795232 |    2797440 | Rdr |50! 00!                                                                  |     | HALT
    2800180 |    2800820 | Tag |00(4)                                                                    |     | 
    2893056 |    2895264 | Rdr |50! 00!                                                                  |     | HALT
    2898052 |    2898692 | Tag |00(4)                                                                    |     | 
    2978192 |    2980400 | Rdr |50! 00!                                                                  |     | HALT
    2983188 |    2983828 | Tag |00(4)                                                                    |     | 
    3036672 |    3038880 | Rdr |50! 00!                                                                  |     | HALT
    3041668 |    3042308 | Tag |00(4)                                                                    |     | 
    3425520 |    3426512 | Rdr |52(7)                                                                    |     | WUPA
    3427748 |    3430116 | Tag |44  00                                                                   |     | 
    3437536 |    3442304 | Rdr |50  00  57  cd                                                           |  ok | HALT
    4299088 |    4300080 | Rdr |52(7)                                                                    |     | WUPA
    4301316 |    4303684 | Tag |44  00                                                                   |     | 
    4311120 |    4313584 | Rdr |93  20                                                                   |     | ANTICOLL
    4314756 |    4320644 | Tag |04  e2  26  75  b5                                                       |     | 
    4328000 |    4338464 | Rdr |93  70  04  e2  26  75  b5  c8  c0                                       |  ok | SELECT_UID
    4339700 |    4343220 | Tag |08  b6  dd                                                               |  ok | 
[... 124 lines cut ...]
   16485252 |   16485892 | Tag |00(4)                                                                    |     | 
   16554944 |   16557152 | Rdr |50! 00!                                                                  |     | HALT
   16559940 |   16560580 | Tag |00(4)                                                                    |     | 
   16932176 |   16933168 | Rdr |52(7)                                                                    |     | WUPA
   16934404 |   16936772 | Tag |44  00                                                                   |     | 
   16944208 |   16948976 | Rdr |50  00  57  cd                                                           |  ok | HALT
   17805808 |   17806800 | Rdr |52(7)                                                                    |     | WUPA
   17808036 |   17810404 | Tag |44  00                                                                   |     | 
   17817824 |   17820288 | Rdr |93  20                                                                   |     | ANTICOLL
   17821460 |   17827348 | Tag |04  e2  26  75  b5                                                       |     | 
   17834704 |   17845168 | Rdr |93  70  04  e2  26  75  b5  c8  c0                                       |  ok | SELECT_UID
   17846404 |   17849924 | Tag |08  b6  dd                                                               |  ok | 
   17982448 |   17984656 | Rdr |50! 00!                                                                  |     | HALT
   17987444 |   17988084 | Tag |00(4)                                                                    |     | 
   18054736 |   18056944 | Rdr |50! 00!                                                                  |     | HALT
   18059732 |   18060372 | Tag |00(4)                                                                    |     | 
   18135696 |   18137904 | Rdr |50! 00!                                                                  |     | HALT
   18140692 |   18141332 | Tag |00(4)                                                                    |     | 
   18222256 |   18224464 | Rdr |50! 00!                                                                  |     | HALT
   18227236 |   18227876 | Tag |00(4)                                                                    |     | 
   18562416 |   18563408 | Rdr |52(7)                                                                    |     | WUPA
   18564660 |   18567028 | Tag |44  00                                                                   |     | 
   18574432 |   18579200 | Rdr |50  00  57  cd                                                           |  ok | HALT
   19438292 |   19440660 | Tag |44  00                                                                   |     | 
```

gornekich, 2023-12-05:
Hello @dogtopus 
We reproduced this issue and a fix arrived in #3264 . For some reason Pixel sends 15bit "halt" command instead of 16 bit 50 00 + 16 bit crc. I added crutch and now emulation should work as expected. Please update firmware to latest RC or dev and let me know if you still have this issue

dogtopus, 2023-12-05:
It does go further and now empty/unlocked tags can be read. Though I'm now getting tag lost when trying to unlock a locked card. More info coming soon...

dogtopus, 2023-12-08:
[Test.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13608567/Test.zip)

Test file that reproduces this.

Tag lost occurs when unlocking the sector 1 (which has the access condition patched to 0x70f878) with Metrodroid.

gornekich, 2023-12-13:
Thanks for information!
#3291 should fix the issue

dogtopus, 2023-12-15:
@gornekich Seems to work well.

closed 2023-12-15 by gornekich