flipperdevices/flipperzero-firmware #3249: False positive in gate_tx

yoavw, 2023-11-29:
### Describe the bug.

Some signals are misidentified as GateTX despite having more than 24 bits. The serialized result cannot be transmitted because the deserialize function fails the 24 bits check.  Below is a sample of such a file.

I suspect the bug is in [gate_tx.c:230](https://github.com/flipperdevices/flipperzero-firmware/blob/f9101d80840b4b7ef7146e41bba99aac881bbfca/lib/subghz/protocols/gate_tx.c#L230), where it checks for `if(instance->decoder.decode_count_bit >= subghz_protocol_gate_tx_const.min_count_bit_for_found)` accepting more than 24 bits, whereas during deserialization the check is `subghz_block_generic_deserialize_check_count_bit&instance->generic, flipper_format, subghz_protocol_gate_tx_const.min_count_bit_for_found)` which requires **exactly** 24 bits.

Sample serialized result that cannot be deserialized:
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 434075000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: GateTX
Bit: 36
Key: 00 00 00 02 8C A3 60 4D

### Reproduction

1. Record a signal that isn't really GateTX but that the gate_tx feed detects as such.
2. View it. The view shows only 24 bits although the serialized file has more.
3. Try Send.  Get "Error in protocol parameters description".

Skorpionm, 2023-12-01:
https://github.com/flipperdevices/flipperzero-firmware/pull/3253

yoavw, 2023-12-01:
I already tested an identical fix, it stops the false positive and the signal gets recognized correctly. But I don't have any gate_tx device so I can't verify that it still works.

Skorpionm, 2024-01-16:
added to dev

skotopes, 2024-04-14:
Should be fixed in latest release.

closed 2024-04-14 by skotopes