flipperdevices/flipperzero-firmware #3257: Menu navigation buttons

Tomi2965, 2023-12-02:
### Description of the feature you're suggesting.

Currently menu navigation works with "Up-Down-OK-Back". It's possible to add a "Left-Right" buttons? Like on a Samsung TV's. Left for back, right for OK.

skotopes, 2023-12-04:
Can you elaborate? Are we talking about main menu, submenu, apps?

Tomi2965, 2023-12-04:
Main menu, submenus, app list, file list. 

Example: I want to open NFC menu. On main screen hit OK, then scroll to NFC, press right, but then I want to go back, and open RFID. So press left (back to main menu), up (from NFC to RFID), right (open RFID). Then I want to open Read - just press right.  

As I see, in app list/file list is left for back working. Much better than press back button in main menu/submenus :)

skotopes, 2023-12-07:
That was made intentionally to simply UX learning.

closed 2023-12-07 by skotopes