flipperdevices/flipperzero-firmware #3258: NFC - Missing function?

Tomi2965, 2023-12-02:
### Describe the bug.

In NFC/Extra actions menu is item "Mifare Classic Keys". On press OK, this item does nothing, only returns to NFC menu.

### Reproduction

Open menu - NFC - Extra Actions - click on Mifare Classic Keys

skotopes, 2023-12-04:
@gornekich I think we should show some error message when we can't open file

Tomi2965, 2023-12-04:
> Check your SD card: ensure that NFC app assets are not damaged. Format SD Card and reinstall firmware.

@skotopes  Checked - problem not fixed. I was try:
- replace card with a new one, then reformat, copy old data back, then reinstall 0.95.0,
- update from 0.95.0 to RC 0.96.0,
- reinstall RC 0.96.0,
- factory reset, installed RC 0.96.0 (just clean install without any data on int/ext mem),
- factory reset, installed 0.95.0 (just clean install without any data on int/ext mem),
- put my FZ to previous state (with new SD card, old data from int/ext mem), reinstall FW.

I was try to enter NFC-Extra Act.-Mifare Classic Keys after each step. Still same result. Any suggestions? Can you please verify this on another FZ? Thanks a lot.

FearnessBG, 2023-12-04:
And did you make any nfc nesting ? Because i was not able to open it from where you want till made first reading with MF1 s50 4 mb

FearnessBG, 2023-12-04:
You don't have the mf_classic_dict_user.nfc .That is why i am asking?

gornekich, 2023-12-04:
Reproduced this issue, will be fixed soon

Tomi2965, 2023-12-04:
> Can you open it on the FZ with the browser to read it?

Which file u mean? Browser - just in "Saved" submenu?

> And did you make any nfc nesting ? Because i was not able to open it from where you want till made first reading with MF1 s50 4 mb

Yep, I have five saved NFC's in /ext/nfc folder

> You don't have the mf_classic_dict_user.nfc .That is why i am asking?

Correct - file "mf_classic_dict_user.nfc" is missing, and it was not created during factory reset/update/reinstall. 

> Reproduced this issue, will be fixed soon

Thanks a lot! <3

FearnessBG, 2023-12-04:
Can you open it on the FZ with the browser to read it?

Which file u mean? Browser - just in "Saved" submenu?
Yes, sorry, i was asking on the Flipper in the browser can you open it from there, but then came the answer Reproduced this issue, will be fixed soon , so that is good. Was trying to help.

Tomi2965, 2023-12-04:
> Can you open it on the FZ with the browser to read it?
>
> Which file u mean? Browser - just in "Saved" submenu? Yes, sorry, i was asking on the Flipper in the browser can you open it from there, but [...]

Many thanks for try. Any help is always better :)

gornekich, 2023-12-05:
@Tomi2965 , we fix this issue in #3264. Please, test in on latest RC or dev and let me know if you still have any problems.

Tomi2965, 2023-12-05:
@gornekich , verified via RC 0.96.1 - it **works**. Thanks for help.

closed 2023-12-05 by Tomi2965