flipperdevices/flipperzero-firmware #3261: Booting with OK button - silent recovery mode?

Tomi2965, 2023-12-04:
### Describe the bug.

When you try to boot (after Soft/Hard Reboot or normal Power On) with OK button pressed, FZ go to recovery mode without anything on screen, no backlight, no LED, and no response to anything. Then a Soft or Hard Reboot must be applied. 

### Reproduction

1.1) hold Left+OK+Back, wait for Soft Reboot, then release only Left+Back (remain holding OK button)
**-OR-**
1.2) hold OK+Back until FZ do a Hard Reboot, then release only Back button (remain holding OK button)
**-OR-**
1.3) do normal Shutdown, then press and hold OK, then press and hold Back for two-three seconds, then release both. 
2.) open qFlipper
3.) connect to USB
4.) you will see Recovery Mode
5.) to exit do Soft/Hard Reboot

hedger, 2023-12-04:
It's documented as hardware-based DFU recovery mode.
https://docs.flipper.net/basics/firmware-update/firmware-recovery#tSOGv

Tomi2965, 2023-12-04:
> It's documented as hardware-based DFU recovery mode. https://docs.flipper.net/basics/firmware-update/firmware-recovery#tSOGv

Thanks for help/link and sorry for mistake.