flipperdevices/flipperzero-firmware #3268: Duplicate scanned Felica card to NFC card

kai0310, 2023-12-05:
### Description of the feature you're suggesting.

Thank you for this wonderful community!

Thank you for supporting Felica emulation.
I immediately tried to use the function and this feature is indeed working!

Now, I would like to duplicate a Felica card to one NFC card and copy it to another copied NFC card without having to carry Flipper.

First of all, as a prerequisite, is this functionality technically possible using Flipper?
And if possible, I request this functionality.

Thanks.

choryuidentify, 2024-01-01:
No, You can't.

1. Flipper can read 'IDm' and can emulate it.
    but no 'IDm Cloneable Felica Card' on the market. (At least as far as I know)
2. Flipper can't read 'data' of Felica Card, so can't emulate or clone it.

skotopes, 2024-01-02:
@kai0310 yep. There are no known cards like MiFare magic cards for felica. At this point you need flipper.

closed 2024-01-02 by skotopes