flipperdevices/flipperzero-firmware #3270: Unable to unlock via BT-RC when asking for PIN

Tomi2965, 2023-12-05:
### Describe the bug.

I'm not sure, if this is a "bug", or it's normal behavior (then sorry for useless issue report).

FZ cannot be unlocked via bluetooth remote control, when unlock screen is active on FZ (asking for PIN). Command is sent, but nothing happen (without any visible response). After unlock screen disappear (timeout or back button press), FZ can be normally unlocked via BT remote control.

Tested on 0.95.0 and RC 0.96.1, Flipper app for Android, ver. 1.6.7.1326.

Thanks for any help!

### Reproduction

1.) connect FZ to phone via BT
2.) open Flipper app - open remote control
3.) lock FZ (timeout or lock function)
4.) on FZ press any key to show screen "Enter PIN"
5.) when unlock screen ("Enter PIN") is present, on phone press "unlock" button (nothing will happen, no visible response)
6.) wait until unlock screen disappear or press back button
7.) when unlock screen is NOT present, on phone press "unlock" button (FZ will unlock)

skotopes, 2024-01-02:
@doomwastaken can you check please?

doomwastaken, 2024-01-03:
Verified, app and flipper work as described, you cannot unlock on the pin enter scene, only when there is no prompt. Always assumed it to be correct behavior

closed 2024-01-03 by hedger

Tomi2965, 2024-01-05:
> Verified, app and flipper work as described, you cannot unlock on the pin enter scene, only when there is no prompt. Always assumed it to be correct behavior

Dear @doomwastaken, thanks for reply. As I see, and as I tested latest dev release, it's fixed. Thanks for fix!

If I can ask (just only for info), me and my colleague are not sure, what you mean by "always assumed it to be correct behavior".

Again, thanks.

doomwastaken, 2024-01-05:
Greetings, @Tomi2965, there is an use case scenario, where flipper is being used in backpack or pocket, for opening gates or garrage doors. So if prompt is open by accident, let's say by pressing the button on device itself(on accident), then it would prevent it from interfering with the app use, until user fixes it. Maybe that is a stretch, hard to know without data

Tomi2965, 2024-01-07:
@doomwastaken , thank you for reply. Yep, we (me and my colleague) can fully understand you'r arguments - when FZ is in pocket/backpack and any button is pressed (FZ is on PIN screen). 

Constructively: there are two ways how to handle this situation.

1.) Fix - use remote control as a remote control without limitations (again, thank for fix)
2.) Decline - just decline to unlock FZ. In this case, app need to show a msg/warning, maybe allow to unlock (with notice to pocket/backpack warning and option to ignore in future [+option to change/revert somewhere in settings]) 

For me, is "fix" the right way. When you have a remote control, it's fine to have a RC without limitations. When there is a limitation, then it's fine, to notify user, and allow to choose preffered way.

Again, seriously thanks for help & you'r work.

doomwastaken, 2024-01-07:
I also considered that from the view of pure code my point was incorrect.  API for unlock should not depend on other conditions.