flipperdevices/flipperzero-firmware #3276: cli nfc command does not work after NFC refactor

u0d7i, 2023-12-07:
### Describe the bug.

After NFC refactor, cli "nfc" command shows empty sub-command list and does not work anymore:
```
+ device: /dev/ttyACM0
starting cli, Ctrl+C to terminate

              _.-------.._                    -,
          .-"```"--..,,_/ /`-,               -,  \ 
       .:"          /:/  /'\  \     ,_...,  `. |  |
      /       ,----/:/  /`\ _\~`_-"`     _;
     '      / /`"""'\ \ \.~`_-'      ,-"'/ 
    |      | |  0    | | .-'      ,/`  /
   |    ,..\ \     ,.-"`       ,/`    /
  ;    :    `/`""\`           ,/--==,/-----,
  |    `-...|        -.___-Z:_______J...---;
  :         `                           _-'
 _L_  _     ___  ___  ___  ___  ____--"`___  _     ___
| __|| |   |_ _|| _ \| _ \| __|| _ \   / __|| |   |_ _|
| _| | |__  | | |  _/|  _/| _| |   /  | (__ | |__  | |
|_|  |____||___||_|  |_|  |___||_|_\   \___||____||___|

Welcome to Flipper Zero Command Line Interface!
Read Manual https://docs.flipperzero.one

Firmware version: dev unknown (82baf1e9 built on 05-12-2023)

>: nfc
nfc
Usage:
nfc <cmd>
Cmd list:

>: 
```

### Reproduction

1. Connect to FZ serial port
2. Execute 'nfc' command via cli

### Target

7

### Anything else?

tested on latest devbuild ATM (latest releases and rc's have the same issues)

Tomi2965, 2023-12-07:
I can confirm - same here on latest versions. Same issue

u0d7i, 2023-12-08:
> I don't think it is a bug, after that refactoring the only command available (if the debug flag is active) is "field"

Before refactor it had:
```
Firmware version: 0.89.0 0.89.0 (5a58004b built on 16-08-2023)

>: nfc
nfc
Usage:
nfc <cmd>
Cmd list:
        detect   - detect nfc device
        emulate  - emulate predefined nfca card
        apdu     - Send APDU and print response 

>: 

```
And I was relying on `apdu` one. If this is official and final now, and it 's not going to be re-implemented, would be nice to have a clear statement from the development team on that. I've seen changes in official docs here: https://docs.flipper.net/development/cli , which only refers to: "nfc - NFC factory test app: controls field when the device is in Debug mode." now.

gornekich, 2023-12-11:
Hello @u0d7i . These commands were removed after refactoring indeed. I am planning to add some cli commands in next release

weiss-schatten, 2025-01-28:
Any update on this? It would really help being able to use the previously available commands, if nothing else than to make Flipper communicate with otherwise-unsupported devices. If adding new equivalent commands in future releases is harder than expected / not a current priority, is it at least possible to get the previous commands at least? Or did the refactor change something that allowed them to work?