flipperdevices/flipperzero-firmware #3281: All NTAG215 reads of a blank card appear with 2 pages locked

johng3587, 2023-12-09:
### Describe the bug.

I noticed that recently my flipper started saying all my NTAG215 cards are password protected, and can only read 133/135 pages. This didn't happen before whatever changed. The 
NFC tools app on my phone says the card is not password protected. I even tried a fresh card, never used, still says password protected. Recently, I had been messing around with password protecting NTAG215 cards on my phone, and then unlocking them with the flipper, this could have been the start of the problem, but I'm not sure.

[image]
[image]

### Reproduction

1. Get blank NTAG215 card
2. Read card with regular NFC app on flipper
3. Flipper should say there are some password-protected pages at this point.
4. Unlock with password "FFFFFFFF"
5. Read card again
6. Flipper should say all pages unlocked.

### Logs

```Text
53511 [I][Loader] Loading /ext/apps/NFC/nfc.fap
53648 [I][Elf] Total size of loaded sections: 53013
53650 [I][Loader] Loaded in 139ms
53654 [I][AnimationManager] Unload animation 'L2_Soldering_128x64'
54736 [D][NfcScanner] Found 5 base protocols
54744 [D][DolphinState] icounter 427, butthurt 0
54796 [D][Nfc] FWT Timeout
54821 [D][Nfc] FWT Timeout
54873 [D][Nfc] FWT Timeout
54901 [D][Nfc] FWT Timeout
54916 [D][NfcScanner] Found 4 children
54971 [D][Nfc] FWT Timeout
55009 [D][Nfc] FWT Timeout
55043 [E][Iso14443_4aPoller] Failed to parse ATS response
55045 [D][Iso14443_4aPoller] Failed to read ATS
55047 [D][Nfc] FWT Timeout
55063 [I][NfcScanner] Detected 1 protocols
55193 [D][MfUltralightPoller] Read version success
55195 [D][MfUltralightPoller] NTAG215 detected. Total pages: 135
55197 [D][MfUltralightPoller] Reading signature
55205 [D][MfUltralightPoller] Read page 0 success
55207 [D][MfUltralightPoller] Read page 1 success
55209 [D][MfUltralightPoller] Read page 2 success
55211 [D][MfUltralightPoller] Read page 3 success
55215 [D][MfUltralightPoller] Read page 4 success
55217 [D][MfUltralightPoller] Read page 5 success
55219 [D][MfUltralightPoller] Read page 6 success
55221 [D][MfUltralightPoller] Read page 7 success
55226 [D][MfUltralightPoller] Read page 8 success
55228 [D][MfUltralightPoller] Read page 9 success
55231 [D][MfUltralightPoller] Read page 10 success
55233 [D][MfUltralightPoller] Read page 11 success
55239 [D][MfUltralightPoller] Read page 12 success
55241 [D][MfUltralightPoller] Read page 13 success
55243 [D][MfUltralightPoller] Read page 14 success
55246 [D][MfUltralightPoller] Read page 15 success
55251 [D][MfUltralightPoller] Read page 16 success
55253 [D][MfUltralightPoller] Read page 17 success
55255 [D][MfUltralightPoller] Read page 18 success
55257 [D][MfUltralightPoller] Read page 19 success
55262 [D][MfUltralightPoller] Read page 20 success
55264 [D][MfUltralightPoller] Read page 21 success
55266 [D][MfUltralightPoller] Read page 22 success
55268 [D][MfUltralightPoller] Read page 23 success
55272 [D][MfUltralightPoller] Read page 24 success
55274 [D][MfUltralightPoller] Read page 25 success
55276 [D][MfUltralightPoller] Read page 26 success
55278 [D][MfUltralightPoller] Read page 27 success
55282 [D][MfUltralightPoller] Read page 28 success
55284 [D][MfUltralightPoller] Read page 29 success
[... 276 lines cut ...]
65549 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
65552 [I][AllInOne] Pass not verified
65599 [I][Elf] Total size of loaded sections: 1244
65603 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
65644 [I][Elf] Total size of loaded sections: 1324
65647 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
65699 [I][Elf] Total size of loaded sections: 1724
65703 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
65746 [I][Elf] Total size of loaded sections: 1768
65749 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
65795 [I][Elf] Total size of loaded sections: 1464
65797 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
65803 [D][DolphinState] icounter 427, butthurt 0
74730 [D][ViewDispatcher] View changed while key press 2000B838 -> 2000B158. Sending key: Back, type: Release, sequence: 00000100 to previous view port
75840 [I][Loader] App returned: 0
75848 [I][Loader] Application stopped. Free heap: 143272
76200 [I][AnimationManager] Select 'L2_Soldering_128x64' animation
76204 [I][AnimationManager] Load animation 'L2_Soldering_128x64'
76208 [I][SavedStruct] Loading "/int/.desktop.settings"
76233 [D][ViewDispatcher] View changed while key press 20009EA0 -> 00000000. Sending key: Back, type: Release, sequence: 0000010C to previous view port
```

### Anything else?

I notice in the debug logs, it says "Read page 134 success" on line 55566, and "Read success" on line 55582, this was before I tried the password. Flipper says 133/135 pages read, but the logs say read successful. If page 134 was successful, why does flipper say only 133/135 could be read? Logs also say it tries the default password, before saying read success.

johng3587, 2023-12-12:
[image]
```
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
26340679 [D][WMFU] Card searching...
26340684 [D][DolphinState] icounter 451, butthurt 0
26340699 [D][Nfc] FWT Timeout
26340801 [D][Nfc] FWT Timeout
26340911 [D][MfUltralightPoller] Read version success
26340913 [D][MfUltralightPoller] NTAG215 detected. Total pages: 135
26340916 [D][MfUltralightPoller] Reading signature
26340923 [D][MfUltralightPoller] Read page 0 success
26340926 [D][MfUltralightPoller] Read page 1 success
26340930 [D][MfUltralightPoller] Read page 2 success
[...]
26341284 [D][MfUltralightPoller] Read page 133 success
26341287 [D][MfUltralightPoller] Read page 134 success
26341289 [D][MfUltralightPoller] Skip reading counters
26341291 [D][MfUltralightPoller] Reading tearing flag 2
26341294 [D][MfUltralightPoller] No limits in authentication. Trying default password
26341302 [D][Nfc] FWT Timeout
26341304 [D][MfUltralightPoller] Read success
26341305 [D][MfUltralightPoller] Check writing capability
26341308 [D][MfUltralightPoller] Unknown password
26341310 [D][Nfc] FWT Timeout
26348114 [D][ViewDispatcher] View changed while key press 2000BBF8 -> 2000E218. Sending key: Back, type: Release, sequence: 00000180 to previous view port
```
Can't write to the card either.

gornekich, 2023-12-12:
Hello! We reproduces the issue. The fix will arive with #3285 
It would be very helpful if you confirm that mentioned PR fix your issue

johng3587, 2023-12-12:
Built and tested, the PR does fix the issue. Nice job!

closed 2023-12-12 by johng3587