flipperdevices/flipperzero-firmware #3282: NTAG216 Unlocking Issue

droidXrobot, 2023-12-10:
### Describe the bug.

I tried to unlock an NTAG216 by inputting a password manually. I got a success messaging saying all pages were unlocked successfully but when I tried to read the tag again it said "Password-protected pages!"

### Reproduction

1.  Navigate to NFC
2. Read a password protected NTAG216
3.  More
4. Unlock
5. Enter password manually
6. Save
7. Continue
8.  See message "All pages unlocked!"
9. Exit and read card again. It says  "Password-protected pages!"

### Target

 0.96.1

### Anything else?

 Flipper firmware 0.96.1

doomwastaken, 2023-12-11:
Greetings! If it is possible, could you take take a debug log, also dump of  the card would be helpful

droidXrobot, 2023-12-11:
Yes, I can read all the pages when I follow the steps you provided. However, I can not write to the card afterwards.

gornekich, 2023-12-12:
I see. Now we don't write to locked tags, because it makes writing state machine complicated. We need to work on UI/UX first.

I believe that this issue is not a bug, but a feature request. We will try to deliver these feature in next releases

nekolab, 2024-01-19:
Hi @gornekich I've implemented the write support of the NTAG21x card in this [PR](https://github.com/flipperdevices/flipperzero-firmware/pull/3364), it works on my card now. Could you please take a look?