flipperdevices/flipperzero-firmware #3310: Support for Kedsum-TH weather stations

s1shed, 2023-12-22:
### Describe the enhancement you're suggesting.

Add support for Kedsum-TH weather stations. This is supported by rtl_433 (https://github.com/merbanan/rtl_433/blob/master/src/devices/kedsum.c)

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```
time      : 2023-12-22 22:44:40
model     : Kedsum-TH    ID        : 158
Channel   : 3            Battery level: 0.000      Flags2    : 0             Temperature: -6.00 C      Humidity  : 89 %          Integrity : CRC
*** Saving signal to file g001_433.92M_250k.cu8 (296332 samples, 655360 bytes)
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```
time      : 2023-12-22 22:44:42
model     : Kedsum-TH    ID        : 26
Channel   : 1            Battery level: 0.000      Flags2    : 0             Temperature: 19.61 C      Humidity  : 33 %          Integrity : CRC
*** Saving signal to file g002_433.92M_250k.cu8 (503792 samples, 1048576 bytes)
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```
time      : 2023-12-22 22:45:09
model     : Kedsum-TH    ID        : 250
Channel   : 2            Battery level: 0.000      Flags2    : 0             Temperature: 20.22 C      Humidity  : 37 %          Integrity : CRC
*** Saving signal to file g003_433.92M_250k.cu8 (299724 samples, 655360 bytes)
```

The file [kedsum-th.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13756701/kedsum-th.zip) contains the .cu8 files. 

I converted these to sub files using EvilPete's `subghz_ook_to_sub.py`.  Only one ([g002_433.92M_250k.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13756710/g002_433.92M_250k.zip)) will be detected by rtl_433 when transmitted with the FlipperZero, so I did not include the others.

Please let me know if I can provide anything else to help get this device supported.

Skorpionm, 2023-12-23:
Well, at least make normal records, in readRaw format.  positive temperatures so that tens and units of degrees and tenths change.  several records of negative temperatures, at the same time attach to each file what is shown on the receiver display.  , that's at least

s1shed, 2023-12-23:
Thanks for your response. ☺ 

> Well, at least make normal records, in readRaw format. positive temperatures so that tens and units of degrees and tenths change. several records of negative temperatures, at the same time attach to each file what is shown on the receiver display. , that's at least

ACK. Over the next few days I'll work on getting these sorts of readings.