flipperdevices/flipperzero-firmware #3311: Bug in MIFARE Classic 1K emulation

garathX, 2023-12-22:
### Describe the bug.

Since the 0.94 firmware version : It’s not possible anymore to emulate Mifare Classic cards. It was working fine in 0.93 (so i’m stuck in this version and can’t update).

Perhaps it could be resolved in future releases as it was working in 0.93 ? And if VIGIK Mifare emulation is different from other emulation, perhaps have an option to select it ?

Thank you and don’t hesitate to ask m

### Reproduction

ACR122U reader.
With version > 0.93 :
If I I write the file emulated on an empty magic card, the card will work perfectly and I can read it on my ACR122U with all sectors and no errors.
If I try to emulate the same file with the flipper, the ACR122U can only read ATQA/UID/SAK but not the sectors, it said : “Error : tag disappeared”.
With 0.93 version : the ACR122U can read all the sectors of the emulated card.

### Target

MIFARE Classic 1K emulation read by ACR122U

doomwastaken, 2023-12-25:
We have this reader available. Once we have information, we will give an update

gornekich, 2023-12-28:
Thanks! Fix will arrive with #3324 . Please, test it and let me know if you still have any problems

garathX, 2023-12-28:
Just tested #3324 : AWESOME 🥇 NFC emulation is back and it's working great ! Thank you very much👍

closed 2023-12-29 by gornekich