flipperdevices/flipperzero-firmware #3326: FSK RFID

Aidan-McNay, 2023-12-28:
### Description of the feature you're suggesting.

Hi! I saw that development for FSK support for the 125kHz RFID [looked like it begun](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/lib/lfrfid/tools/fsk_demod.c), but never got finished/integrated. I didn't know if there were plans to finish this (currently new, so figuring out how to best contribute), but it seems like it would be a useful feature. (This would be extending the supported RFID protocols)

(Also, not 100% whether this would be better for firmware or an app, but leaning towards the former)

skotopes, 2024-01-02:
It's literally used in 8 protocols supported by flipper.

closed 2024-01-02 by skotopes

Aidan-McNay, 2024-01-02:
Sorry for the confusion - I saw ASK & PSK on mine, I'm guessing it's under ASK? Any reason for the name change?

skotopes, 2024-01-03:
I got feeling that you mixing different things together: probably because ask/psk that you see in app is a little bit misleading way of naming 2 different groups of data coding schemes.

Long story short:
- ASK - group of protocols with codings that doesn't require special hacks to decode them. For example `EM4100.
- PSK - group of protocols with codings that does require special hacks to decode them. For example `Indala`.