flipperdevices/flipperzero-firmware #3328: Slix-L unlocking support is gone on firmware after NFC refactoring

d4g, 2023-12-29:
### Describe the bug.

As mentioned, the Slix-L unlocing support from #2316 is gone. It probably needs to be refactored or reimplemented @g3gg0 .
After watching your talk at 37c3 I found out that the merged pull request mentioned now is gone again.

### Reproduction

1. Try to find the Slix-L unlocking function in NFC Menug
2. Don't find it ;)

gornekich, 2023-12-29:
Working on it

bluenazgul, 2024-01-24:
Anyone Progress? This is an Feature for me that keeps me away from updating tona newer Release as the october 2023 FW iam running atm

gornekich, 2024-01-25:
Sorry for delay. Should be ready on next week.

nortakales, 2024-02-04:
I stumbled across this issue after getting my Flipper Zero a few weeks ago, and trying to read/emulate my kid's Tonies figures. I'm hoping once this issue is resolved it will be working in the next firmware release. I didn't try the earlier release that had this feature as I didn't have my Flipper back then. If it does work I'll get a repo started with scanned Tonies figures, and hopefully others can contribute too for the benefit of all our kids.

kurazli, 2024-02-05:
> I stumbled across this issue after getting my Flipper Zero a few weeks ago, and trying to read/emulate my kid's Tonies figures. I'm hoping once this issue is resolved it will be working in the next [...]

you can just use an older firmware version, like unleashed 65, until it will be back in a newer version.

gornekich, 2024-02-05:
Could you please the PR #3425 and let me know if you have any problems with unlock function

nortakales, 2024-02-05:
I was able to unlock and read some Tonies figures, and emulate them to the box successfully!

nortakales, 2024-02-05:
Just encountered a crash that required a hard reboot while reading one of the figures. Let me see if I can reproduce...

nortakales, 2024-02-05:
Could not reproduce the crash after reading 9 figures that I have. 🤷‍♂️

I am not very familiar with how these figures or SLIX-L work, but it seems like in order for the Tonies box to work with the Flipper's emulation, you need to place the figure back on the box again (after unlocking/reading with the Flipper) and only then will the Flipper's emulation work. Would be interesting to share some of these .nfc files and see if they work for anyone who doesn't own the corresponding figure. I'll put them in a git repo shortly.

nortakales, 2024-02-05:
The files for the 9 figures I have are here: https://github.com/nortakales/flipper-zero-tonies

They all emulate correctly on my box. Yet to see if a real figure is necessary at first, or if these files will work for anyone.

bluenazgul, 2024-02-06:
> Edit: I turned the box off and back on again, and the Flipper Zero's emulation does NOT work until the authentic figure was placed on the box at least once while it was on.

That sounds that the emulation isnt working as it should, the emulation should work like an authentic figure - thats how it worked before the NFC Refactoring

nortakales, 2024-02-06:
> That sounds that the emulation isnt working as it should, the emulation should work like an authentic figure - thats how it worked before the NFC Refactoring

I just gave firmware 93.0 a try, and you are right, it works right away (no need to put a figure on the box first). Seems like there is still a regression here.

I uploaded a couple files generated from firmware 0.93 [here](https://github.com/nortakales/flipper-zero-tonies/tree/master/Old%20Firmware), you can compare to files generated by this PR if it helps debug at all.

gornekich, 2024-02-06:
@nortakales @bluenazgul made some emulation fixes in #3425 . Please, retest emulation with the fixes and let me know if it works.

nortakales, 2024-02-06:
Works like a charm now. Turned the box on for the first time this morning and the Flipper worked right away.

bluenazgul, 2024-02-07:
confimed, SLIX-L is working again with Tonies like it was before the NFC refactoring

nortakales, 2024-02-07:
@bluenazgul if you have some Tonies I haven't uploaded yet, send a PR! https://github.com/nortakales/flipper-zero-tonies

CybGITHB, 2024-02-10:
All Tonies files are working @nortakales ! @bluenazgul Can you upload some german nfc tonies ?

bluenazgul, 2024-02-10:
i just tried latest dev build and reading/unlocking is working, but emulation did not work

CybGITHB, 2024-02-10:
@bluenazgul use https://github.com/RogueMaster/flipperzero-firmware-wPlugins/releases/tag/RM0208-2318-0.97.2-f563329 firmware
 
See: https://www.g3gg0.de/rf/flipper-zero-for-toniebox-figurines/ and use Unlock Slix-L !

bluenazgul, 2024-02-10:
> @bluenazgul use https://github.com/RogueMaster/flipperzero-firmware-wPlugins/releases/tag/RM0208-2318-0.97.2-f563329 firmware
> 
> See: https://www.g3gg0.de/rf/flipper-zero-for-toniebox-figurines/ and use Unlock Slix-L !

emulation with this RM FW is also not working, even with old NFC files and new NFC Files

bofax, 2024-02-10:
Some Tonies can not be read - Biene Maja (Flip) works, Benjamin Blümchen is not readable.

bluenazgul, 2024-02-10:
> Some Tonies can not be read - Biene Maja (Flip) works, Benjamin Blümchen is not readable.

Reading "Benjamin" is a Bit Tricky but working, only have Problems with emulation, also all old Files are Not working with emulation

CybGITHB, 2024-02-10:
@bluenazgul can you upload .nfc to try ? I habe tried 5 tonies and all are working no problem reading and emulating!

CybGITHB, 2024-02-10:
@bluenazgul what error do you get when trying to read ?

jbouecke, 2024-02-10:
Same for me. Reading seems to work very well, but emulation seems to do
nothing to the box.

bofax, 2024-02-10:
> @bluenazgul can you upload .nfc to try ? I habe tried 5 tonies and all are working no problem reading and emulating!

Try this:
`Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB, EMV
Device type: SLIX
# UID is common for all formats
UID: E0 04 03 50 16 EA B6 29
# ISO15693-3 specific data
# Data Storage Format Identifier
DSFID: 00
# Application Family Identifier
AFI: 00
# IC Reference - Vendor specific meaning
IC Reference: 03
# Lock Bits
Lock DSFID: false
Lock AFI: false
# Number of memory blocks, valid range = 1..256
Block Count: 8
# Size of a single memory block, valid range = 01...20 (hex)
Block Size: 04
Data Content: CE 51 03 52 C1 30 B6 2A FD E1 BA 21 35 2E B4 C0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# Block Security Status: 01 = locked, 00 = not locked
Security Status: 
# SLIX specific data
# Passwords are optional. If a password is omitted, a default value will be used
Password Privacy: 7F FD 6E 5B
Password Destroy: FF FF FF FF
Password EAS: 00 00 00 00
Privacy Mode: false
# SLIX Lock Bits
Lock EAS: false
`

nortakales, 2024-02-10:
Emulation was working perfectly on the [gornek/3676_slix_fixes](https://github.com/flipperdevices/flipperzero-firmware/tree/gornek/3676_slix_fixes) branch, but is not working once merged into dev

bluenazgul, 2024-02-10:
> Emulation was working perfectly on the [gornek/3676_slix_fixes](https://github.com/flipperdevices/flipperzero-firmware/tree/gornek/3676_slix_fixes) branch, but is not working once merged into dev

Good to know, so there are some Code conficts - got IT working sporadic in RM, but found another Feature Change, Emulation only Runs for 5min, Wish me an Option inside the NFC File to Change it from "normal" (5min) to "Long" (use as long as needed)

CybGITHB, 2024-02-10:
Has anyone tried the roquemaster fw for flipper ? It seems to work just fine

nortakales, 2024-02-10:
@CybGITHB this is the repo for the stock firmware. They had this feature working fine last week on a branch, there just seems to be an issue with the merge.

nortakales, 2024-02-12:
@CybGITHB I don't. I live in the states and only have access to English Tonies. If anyone else has access to other languages or Tonies I haven't uploaded yet they can send a PR.

nortakales, 2024-02-14:
@gornekich I pulled down dev, built it, flashed it, and gave reading/emulating a try after you merged https://github.com/flipperdevices/flipperzero-firmware/pull/3444

At first I thought emulating was not working well, but then I started emulating my existing files (instead of reading new ones) and all my existing known-good files emulate perfectly. I think there may be a problem with reading now. Some files end up with `Password Privacy: FF FF FF FF` and no `Data Content`. Other files end up with some `Data Content` missing like this: `Data Content: 40 2E 79 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`.

Other files ended up with `Security Status: ` empty, but otherwise have the necessary data, so those files can easily be modified to have `Security Status: 00 00 00 00 00 00 00 00` and will then work.

I am not sure how much of it to attribute to the Tonies figures being kind of thick, and I have the silicon case on my Flipper. Maybe that is enough to make reading these a bit difficult, and there is no issue with the code?

Please let me know if I can do anything else to help out.

bluenazgul, 2024-02-14:
Maybe it could be possible to add an lock Feature for the "TommyBox" function too and save the unlocked tags with Status locked and the PW that was used to unlock them

nortakales, 2024-02-14:
I tried reading the same figure today 10 times (5 without silicone case, 5 with). I guess the silicone case really has nothing to do with the issue. All 10 reads had all of the `Data Content` and `Security Status` as expected this time. The first read had the correct `Password Privacy` but all 9 reads after that had `Password Privacy: FF FF FF FF`. Maybe I will play around with it some more later to see if I notice anything else. I'm not sure why my reads yesterday were coming out so bad.

moqua, 2024-03-02:
Hi there,

any update on this?

Im new to flipper zero. If i understand correctly, release [RM0217-1146-0.97.2-cc1ffca] was the one with slix-l support. Can anyone provide a link to this or any other release where its working? Seems not available anymore on https://github.com/RogueMaster/flipperzero-firmware-wPlugins/releases. 

Kind regards,
moqua

gornekich, 2024-03-07:
#3504 should fix the remaining issue. If the card is already unlocked, flipper tries to find privacy password after reading slix data.

mamasn, 2024-03-12:
I am trying to write on SLIX-L tags but there is no option for write, only emulate which works, currently using XFW-DEV_@F50D9F5

Fabunator, 2024-04-02:
I have a bunch of old NFC Tonie files read with Version 0.93.1 with Device type ISO15693.
All these files are not compatible with the new version 0.99.1. Here it should be Device type SLIX and some other extra stuff.

deathduck666, 2024-04-02:
> I have a bunch of old NFC Tonie files read with Version 0.93.1 with Device type ISO15693. All these files are not compatible with the new version 0.99.1. Here it should be Device type SLIX and some other extra stuff.

If you have the UID and Data Bytes, you can easily copy them into a current Version4 file. You can find many of them in [nortakales/flipper-zero-tonies](https://github.com/nortakales/flipper-zero-tonies).
Just take one existing file, make a local copy and overwrite UID and Data Bytes. If this works, feel free to add them to the rep.

closed 2024-04-15 by gornekich