flipperdevices/flipperzero-firmware #3330: System crash when selecting saved NFC file (Mifare DESFire)

AndreasLeeb, 2023-12-30:
### Describe the bug.

I saved a Mifare DESFire card and when I select this NFC file, the whole system crashes and reboots.

### Reproduction

1. Unzip attached archive ([crash.nfc](https://github.com/flipperdevices/flipperzero-firmware/files/13798778/crash.nfc.zip)) and put the "crash.nfc" file onto SD card into the "nfc" folder
2. Open "NFC" application on the F0
3. Select the menu option "Saved"
4. Open the file "crash"
5. [image]

### Target

Official firmware release 0.97.1

### Logs

```Text
398968 [D][BrowserWorker] End
398980 [T][StorageApi] File/Dir 0001E280 alloc
398986 [T][StorageApi] Dir 0001E280 - 0001E3AC open (/data/plugins)
398989 [T][StorageApi] File/Dir 0001E390 alloc
398993 [T][StorageApi] File 0001E390 - 0001E72C open (/data/plugins/aime_parser.fal)
399030 [I][Elf] Total size of loaded sections: 888
399033 [T][StorageApi] File 0001E390 - 0001E72C closed
399035 [T][StorageApi] File/Dir 0001E390 free
399037 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399041 [T][StorageApi] File/Dir 0001E390 alloc
399048 [T][StorageApi] File 0001E390 - 0001E85C open (/data/plugins/all_in_one_parser.fal)
399084 [I][Elf] Total size of loaded sections: 420
399087 [T][StorageApi] File 0001E390 - 0001E85C closed
399089 [T][StorageApi] File/Dir 0001E390 free
399091 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399094 [T][StorageApi] File/Dir 0001E390 alloc
399101 [T][StorageApi] File 0001E390 - 0001E8EC open (/data/plugins/mykey_parser.fal)
399139 [I][Elf] Total size of loaded sections: 872
399141 [T][StorageApi] File 0001E390 - 0001E8EC closed
399143 [T][StorageApi] File/Dir 0001E390 free
399145 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399148 [T][StorageApi] File/Dir 0001E390 alloc
399155 [T][StorageApi] File 0001E390 - 0001E974 open (/data/plugins/myki_parser.fal)
399200 [I][Elf] Total size of loaded sections: 1244
399202 [T][StorageApi] File 0001E390 - 0001E974 closed
399206 [T][StorageApi] File/Dir 0001E390 free
399208 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399212 [T][StorageApi] File/Dir 0001E390 alloc
399221 [T][StorageApi] File 0001E390 - 0001E9FC open (/data/plugins/opal_parser.fal)
399259 [I][Elf] Total size of loaded sections: 1324
399262 [T][StorageApi] File 0001E390 - 0001E9FC closed
399264 [T][StorageApi] File/Dir 0001E390 free
399267 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399270 [T][StorageApi] File/Dir 0001E390 alloc
399278 [T][StorageApi] File 0001E390 - 0001EA8C open (/data/plugins/plantain_parser.fal)
399317 [I][Elf] Total size of loaded sections: 1724
399319 [T][StorageApi] File 0001E390 - 0001EA8C closed
399323 [T][StorageApi] File/Dir 0001E390 free
399325 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399328 [T][StorageApi] File/Dir 0001E390 alloc
399336 [T][StorageApi] File 0001E390 - 0001EB1C open (/data/plugins/troika_parser.fal)
399374 [I][Elf] Total size of loaded sections: 1768
399376 [T][StorageApi] File 0001E390 - 0001EB1C closed
399379 [T][StorageApi] File/Dir 0001E390 free
399381 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399384 [T][StorageApi] File/Dir 0001E390 alloc
399392 [T][StorageApi] File 0001E390 - 0001EBAC open (/data/plugins/two_cities_parser.fal)
399430 [I][Elf] Total size of loaded sections: 1464
399433 [T][StorageApi] File 0001E390 - 0001EBAC closed
399435 [T][StorageApi] File/Dir 0001E390 free
399438 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399442 [T][StorageApi] File/Dir 0001E390 alloc
399450 [T][StorageApi] File 0001E390 - 0001EC3C open (/data/plugins/umarsh_parser.fal)
399498 [I][Elf] Total size of loaded sections: 636
399502 [T][StorageApi] File 0001E390 - 0001EC3C closed
399504 [T][StorageApi] File/Dir 0001E390 free
399507 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
399510 [T][StorageApi] Dir 0001E280 - 0001E3AC closed
399512 [T][StorageApi] File/Dir 0001E280 free
399514 [D][NfcSupportedCards] Loaded 9 plugins
399522 [T][StorageApi] File/Dir 0001E230 alloc
399529 [T][StorageApi] File 0001E230 - 0001ED4C open (/any/nfc/crash.nfc)

[CRASH][NFC] furi_check failed
        r0 : 0
        r1 : 20000ef8
        r2 : 20000284
        r3 : 0
        r4 : 0
        r5 : 2000a558
        r6 : 0
        r7 : 200
        r8 : 0
        r9 : 20000ecc
        r10 : 0
        r11 : 20031330
        lr : 8011127
        stack watermark: 3684
             heap total: 186024
              heap free: 62160
         heap watermark: 61064
        core2: not faulted
Rebooting system.
```

skotopes, 2024-01-02:
@gornekich @doomwastaken please take a look

skotopes, 2024-01-02:
```(gdb) bt
#0  0x080104f6 in __furi_crash_implementation () at furi/core/check.c:164
#1  0x0801117e in pvPortMalloc (xWantedSize=<optimized out>) at furi/core/memmgr_heap.c:485
#2  0x08010b2c in malloc (size=<optimized out>) at furi/core/memmgr.c:13
#3  0x0803153a in simple_array_init (instance=0x2000d7a0, count=count@entry=0) at lib/toolbox/simple_array.c:30
#4  0x080355d0 in mf_desfire_application_load (data=data@entry=0x2000d690, prefix=0x2000df70 "Application 7087f8", ff=ff@entry=0x2000d3d0) at lib/nfc/protocols/mf_desfire/mf_desfire_i.c:482
#5  0x08034c08 in mf_desfire_load (data=0x2000d540, ff=0x2000d3d0, version=4) at lib/nfc/protocols/mf_desfire/mf_desfire.c:130
#6  0x08038562 in nfc_device_load_unified (version=4, ff=0x2000d3d0, instance=0x2001f010) at lib/nfc/nfc_device.c:265
#7  nfc_device_load (instance=instance@entry=0x2001f010, path=<optimized out>) at lib/nfc/nfc_device.c:343
#8  0x20019126 in nfc_load_file (instance=instance@entry=0x2000e8b8, path=<optimized out>, show_dialog=show_dialog@entry=true) at applications/main/nfc/nfc_app.c:356
#9  0x200192fa in nfc_load_from_file_select (instance=instance@entry=0x2000e8b8) at applications/main/nfc/nfc_app.c:412
#10 0x200144b4 in nfc_scene_file_select_on_enter (context=0x2000e8b8) at applications/main/nfc/scenes/nfc_scene_file_select.c:6
```

mf_desfire_application_load is trying to allocate empty array which causes crash on allocator assert

oskarboer, 2024-01-08:
I have the exact same problem.

I was however still able to successfully emulate the card without saving it.

gornekich, 2024-04-16:
The fix arrived in dev. Please, try to reproduce the issue with latest dev.

closed 2024-06-19 by gornekich