flipperdevices/flipperzero-firmware #3333: SubGhz ProtocolKeeloq can custom user key and data file

hsSam, 2024-01-03:
### Describe the enhancement you're suggesting.

1.   I update my keeloq file "keeloq_mfcodes_user" with my own key, then i can use "Read" func read out the Keeloq msg, but I cant save. (cause not have the save button show). I buy flipper zero only two day, so I am not sure that,  Is only the keeloq key by system define can save?

2.  I check the "Add Manually", I add the RF "DoorHan_433", and I edit the file , use my keeloq data, then try to "Emulate", i show protocol params error, so I clone the firmware , I found this in Keeloq.c file
`        if(strcmp(instance->manufacture_name, "DoorHan") != 0) {
            FURI_LOG_E(TAG, "Wrong manufacturer name");
            ret = SubGhzProtocolStatusErrorParserOthers;
            break;
}`
this is the first problem I found, Only DoorHan can use keeloq file.

3. then i modify the source code , remove the code
` if(strcmp(instance->manufacture_name, "DoorHan") != 0) {`
then i can open the keeloq rf file, 
`Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: KeeLoq
Bit: 64
Key: AE EC FB 91 00 FE B4 F4
`
then I use the "Emulate" try to send my own key rf,  after i try 10 times,  the manufacture_name change the another name.
the keeloq data decrypt not right , cause the wrong key.   I thinking why it change my manufacture name use the wrong key, 
I check the func "subghz_protocol_keeloq_check_decrypt", so that's another problem. 
`    if((decrypt >> 28 == btn) && (((((uint16_t)(decrypt >> 16)) & 0xFF) == end_serial) ||
                                  ((((uint16_t)(decrypt >> 16)) & 0xFF) == 0))) {`
use this condition to check keeloq data is not enough, need to  check "Cnt", i update the code
`    if((decrypt >> 28 == btn) && ((((uint16_t)(decrypt >> 16)) & 0x3FF) == end_serial) &&
       ((instance->cnt == (decrypt & 0x0000FFFF)) ||
        ((instance->cnt + 1) == (decrypt & 0x0000FFFF)))) {`
but I found the "instance->cnt" is zero , cause not save the data, then i update the code to save Cnt data like this:
`Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: KeeLoq
Bit: 64
Key: E9 92 BC 0A 25 64 8D 74
Cnt: 4
`

that's the problem and the code i update, so I suggest that ,   support to save keeloq user owner key data, 
also add Cnt data  to compare the decrypt data

hsSam, 2024-01-03:
there is another question , why my log show
```
25403 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 1811
25407 [D][SubGhzProtocolKeeloq] name: KEY, decrypt: C4294C86
25410 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 19590
25416 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 34065
25419 [D][SubGhzProtocolKeeloq] name: Gibidi, decrypt: 55D4BE6F
25422 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 48751
25425 [D][SubGhzProtocolKeeloq] name: JCM_Tech, decrypt: BF492623
25428 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 9763
25431 [D][SubGhzProtocolKeeloq] name: Kingates_Stylo4k, decrypt: A5582409
25436 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 9225
25440 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 12989
25443 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 46051
25448 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 53843
25451 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 12009
25455 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 63396
25459 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 34711
25462 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 47095
25467 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 780
25472 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 14100
25477 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 30563
25483 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 46145
25487 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 32241
25491 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 57096
25494 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 26148
25500 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 58772
25504 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 46787
25508 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 63091
25513 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 54008
25517 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 39565
25521 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 61495
25524 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 30399
25526 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 40904
25529 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 5014
25536 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 57877
25541 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 50762
.................more.............
--------below is my own key
25703 [D][SubGhzProtocolKeeloq] name: TEST1, decrypt: 5CCCB5F5
25706 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 46581
25711 [D][SubGhzProtocolKeeloq] name: TEST2, decrypt: A0ED327E
25714 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 12926
25717 [D][SubGhzProtocolKeeloq] name: TEST3, decrypt: 2F0065AE
25720 [D][SubGhzProtocolKeeloq] instance->btn: 0, instance->serial: 00, instance->cnt: 0, decrypt_cnt: 26030
25724 [D][SubGhzProtocolKeeloq] subghz_protocol_keeloq_check_decrypt ok...
```

hsSam, 2024-01-03:
[image]

Also like this, can see the Btn value if decrypt ok can check the btn value fast

Skorpionm, 2024-04-16:
This is 1 of Flipper's protections to prevent it from being used for illegal purposes with rolling code emulation

closed 2024-04-16 by Skorpionm