flipperdevices/flipperzero-firmware #3336: Cannot Read or Write Liftmaster Passport Passcard.

sfazzina, 2024-01-04:
### Describe the enhancement you're suggesting.

I have a Liftmaster Passport Passcard I am trying to read,  I cant read it.  It's part of the LiftMaster passport suite of products.    I just got my device today and I'm playing around with it.  This is the card I'm trying to read.   https://www.liftmaster.com/ppcsc-passport-30-bit-proximity-card/p/PPCSC

Some community members on Discord recommended I come here and share this. Hopefully we can add support for this card type. 

Thanks and have a great day

### Anything else?

Hi there, 
I have a Liftmaster Passport Passcard I am trying to read,  I cant read it.  It's part of the LiftMaster passport suite of products.   

 This is the card I'm trying to read.   https://www.liftmaster.com/ppcsc-passport-30-bit-proximity-card/p/PPCSC.  

Adding RAW captures as well - I had to rename to .txt otherwise, it wouldn't let me upload.  

Verified its an LF card. See photos 

[image]

[image]
[image]
[RfidRecord.ask.raw.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13824591/RfidRecord.ask.raw.txt)
[RfidRecord.psk.raw.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13824592/RfidRecord.psk.raw.txt)

sfazzina, 2024-01-30:
Hi Team, Just wondering how long until I can read/test duplicate this card ?
I think your project is amazing. I love Flippers zero!!!!

zinongli, 2024-06-09:
Some info that might be helpful:
Using [flipper-raw-rifl](https://github.com/hnesk/flipper-raw-rfid), I read this from Manchester demodulating the .ask.raw:

`100000000011001110111111111111111111111111010111010100011000111010111100011101111111111111111111`

96-bit long. If we flip every bit we get this sequence:

`0111111111 0 01100010 0 00000000 0 00000000 0 0000101|0 0 01010111 0 0111000|1 0 10000111 0 00100000 0 00000000 0 0000`

I formatted the sequence assuming spacers made of 0s between bytes. It appears to be that the 16 bits in between the vertical dashes (without spacers) encode card number.
`0 0 01010111 0 0111000`
remove spacers 
`00101011 10111000`
which is equal to 11192 in decimal. 

The long pulses are around 349 mu sec according to Flipper Lab's pulse plotter. I assume this is RF/40 which would yield 320 mu sec, the closest possible clock (otherwise it will be RF/50 corresponding to 400 mu sec). 

Securakey uses the last 6 bits in the second byte to signify the bit length of encoded data. This might sound confusing but OP's link to the tag's official page is titled "PPCSC Passport 30-bit Proximity Card" but it's actually 34-bit in the product title. In any case, if we look at the digit where Securakey encode its bit length, we will get 100010 from
                          |  here |
0111111111 0 01100010 0 00000000 0 00000000.....
which equals to 34. 

The preamble, spacer, and bit length denotation format are all very similar to what is used in Securakey which is also RF/40 with ASK&Manchester modulation. The only difference being the signal is reverted. But it could also be due to some analysis mistake on my part.

OP: if you are still interested in this project, just let me know. To reverse engineer this card we might need some more back-and-forth: more raw readings (preferably of a different card) and a photo of the reader will be very helpful. I have a Securakey protocol script at hand so adding compatibility for this shouldn't take more than an hour. But we can't do that before knowing what exactly each bit means.

sfazzina, 2024-06-09:
Don't worry, worry about it. I was very interested in being able to duplicate this card but it's been a while now and we left the building and no longer need access. I have access to the facility in order to do any testing.

skotopes, 2024-07-07:
Fixed in https://github.com/flipperdevices/flipperzero-firmware/pull/3697, thanks to @zinongli

closed 2024-07-07 by skotopes