flipperdevices/flipperzero-firmware #3338: Saving Mifare DESFire NFC crashes Flipper

scapermoya, 2024-01-04:
### Describe the bug.

I am trying to save a fob. I can read it with the NFC function, which identifies it as a Mifare DESFire chip. It reads the UID, the bytes saved and available, the number of applications, and number of files. When I go to save the info, after choosing a name, the screen goes blank and the Flipper reboots. It then tells me "Flipper crashed and was rebooted" and "NULL pointer dereference". Interestingly, the name I chose appears as a saved item within NFC, but appears to be blank. 

I have re-formatted the SD card and reinstalled the latest firmware (0.97.1) and the problem persists.

I am able to scan and save other NFC devices without issue.

### Reproduction

1. Switch on
2. Go to NFC app
3. Scan fob
4. Attempt to save NFC data

### Logs

```Text
Weirdly, even in debug mode, I do not see any activity in the logs when I do these actions
```

simonkeng, 2024-01-20:
I'm seeing the same behavior on both release and release-candidate versions

VictorCalderon, 2024-01-23:
> I'm seeing the same behavior on both release and release-candidate firmware versions. Crashes with `NULL pointer dereference` message after choosing a name and pressing "save".

Same is happening to me.

gornekich, 2024-01-25:
Hello! Thanks for reporting the bug.

Unfortunately I can't reproduce this issue. I kindly ask for some help in debugging.

If you have debugger - st-link or wifi-board, please follow these steps:

1. Build firmware with debug flag: `./fbt LIB_DEBUG=1`
2. Flash firmware `./fbt LIB_DEBUG=1 flash`
3. Reproduce crash
4. Launch debug session: `./fbt  LIB_DEBUG=1 debug`
5. Run `bt` command to get stack trace and share the result here

If you don't have debugger - build and flash firmware form #3395, collect logs during issue reproduction and share logs here.

scapermoya, 2024-01-25:
Hello,

Sorry, I am a bit of a novice with github code and building patched firmware from commits like this. Do you have a resource or guide I can use to build this custom firmware, because I would be glad to build it and flash my flipper with it and generate these logs.

Thanks

gornekich, 2024-01-26:
@scapermoya  thanks a lot for testing!

> Now after I successfully scan the fob and see all the data, when I go to save it and choose a filename (or use one of the quirky default names), it no longer crashes with the previous error. Now it just says "Cannot save key file" with an SD card logo with a ? inside it.

This looks like we are on the right way. However it is strange that you can't see any logs. Could you please share how you collect logs?

You can find the recommended way to read flipper's logs here https://docs.flipper.net/development/cli in section **Reading Flipper Zero logs via the CLI**. Necessary log level is Debug

scapermoya, 2024-01-30:
Hello, this helped a lot. Here is the log of my attempt to save this fob data:

91266 [I][Loader] Loading /ext/apps/NFC/nfc.fap
91377 [I][Elf] Total size of loaded sections: 60625
91377 [I][Loader] Loaded in 111ms
91379 [I][AnimationManager] Unload animation 'L1_Recording_128x51'
97517 [D][NfcScanner] Found 5 base protocols
97518 [D][DolphinState] icounter 62, butthurt 0
97530 [D][Nfc] FWT Timeout
[...]
99065 [D][Nfc] FWT Timeout
99080 [D][NfcScanner] Found 4 children
99135 [D][Nfc] FWT Timeout
99136 [D][Nfc] FWT Timeout
99216 [D][Iso14443_4aPoller] Read ATS success
99237 [I][NfcScanner] Detected 1 protocols
99404 [I][Elf] Total size of loaded sections: 980
99405 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99447 [I][Elf] Total size of loaded sections: 496
99447 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99490 [I][Elf] Total size of loaded sections: 964
99490 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99539 [I][Elf] Total size of loaded sections: 1172
99540 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99592 [I][Elf] Total size of loaded sections: 1312
99592 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99636 [I][Elf] Total size of loaded sections: 1428
99637 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99681 [I][Elf] Total size of loaded sections: 1904
99681 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99726 [I][Elf] Total size of loaded sections: 1880
99727 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99772 [I][Elf] Total size of loaded sections: 1576
99772 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99813 [I][Elf] Total size of loaded sections: 652
99814 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99864 [I][Elf] Total size of loaded sections: 1100
99864 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
99866 [D][NfcSupportedCards] Loaded 11 plugins
99880 [D][Iso14443_4aPoller] Read ATS success
99888 [D][MfDesfirePoller] Read version success
99891 [D][MfDesfirePoller] Read free memory success
99893 [D][MfDesfirePoller] Read master key settings success
99895 [D][MfDesfirePoller] Read master key version success
99899 [D][MfDesfirePoller] Read application ids success
100076 [D][MfDesfirePoller] Read applications success
100076 [D][MfDesfirePoller] Read success.
100077 [D][Nfc] FWT Timeout
100154 [I][Elf] Total size of loaded sections: 1312
100155 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
100213 [I][Elf] Total size of loaded sections: 1428
100214 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
100310 [D][DolphinState] icounter 62, butthurt 0
118759 [I][MfDesfire] Saving iso-4 data
118759 [I][MfDesfire] Saving comment
118759 [I][MfDesfire] Saving version
118761 [I][MfDesfire] Saving free memory
118763 [I][MfDesfire] Saving key settings
118765 [I][MfDesfire] Saving key version
118765 [I][MfDesfire] Saving key version 0
118765 [I][MfDesfire] Saving app count
118767 [I][MfDesfire] Saving app ids
118767 [I][MfDesfire] Saving app 0
118781 [I][MfDesfire] Saving app 1
118787 [I][MfDesfire] Saving app 2
118795 [E][MfDesfire] file_data 0 is null
130310 [I][Dolphin] Flush stats
130310 [I][SavedStruct] Saving "/int/.dolphin.state"
130321 [D][StorageInt] Device erase: page 2, translated page: a9
130324 [D][StorageInt] Device sync: skipping
130324 [I][DolphinState] State saved
137443 [I][FuriThread] NFC allocation balance: 0
137455 [I][Loader] App returned: 0
137462 [I][Loader] Application stopped. Free heap: 142240
137636 [I][AnimationManager] Select 'L1_Recording_128x51' animation
137636 [I][AnimationManager] Load animation 'L1_Recording_128x51'
137636 [I][SavedStruct] Loading "/int/.desktop.settings"

gornekich, 2024-01-30:
@scapermoya thanks a lot for testing. Could you please try again firmware from #3395 ? Try to save and load file. Latest updates in that PR should solve the issue.

scapermoya, 2024-01-30:
Hello,

We are making some progress I think. Loaded the new version of the firmware from #3395.

Log below shows my scanning of the fob and then attempted saving. After choosing a file name to save, the screen goes to an hourglass and hangs there. The end of the log represents this hang, no apparent specific error associated with that, but there are some errors above during the scanning of the fob. thanks for continuing to work on this.

Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
183031 [I][Loader] Loading /ext/apps/NFC/nfc.fap
183142 [I][Elf] Total size of loaded sections: 60625
183143 [I][Loader] Loaded in 112ms
183145 [I][AnimationManager] Unload animation 'L1_Waves_128x50'
187523 [D][NfcScanner] Found 5 base protocols
187524 [D][DolphinState] icounter 62, butthurt 0
187564 [D][Nfc] FWT Timeout
187589 [D][Nfc] FWT Timeout
187643 [D][Nfc] FWT Timeout
187673 [D][Nfc] FWT Timeout
187688 [D][NfcScanner] Found 4 children
187743 [D][Nfc] FWT Timeout
187744 [D][Nfc] FWT Timeout
187779 [D][Nfc] FWT Timeout
187812 [D][Iso14443_4aPoller] Read ATS success
187812 [I][MfDesfirePoller] CMD: 60
187833 [I][NfcScanner] Detected 1 protocols
188003 [I][Elf] Total size of loaded sections: 980
188003 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188045 [I][Elf] Total size of loaded sections: 496
188046 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188089 [I][Elf] Total size of loaded sections: 964
188089 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188131 [I][Elf] Total size of loaded sections: 1172
188132 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188185 [I][Elf] Total size of loaded sections: 1312
188185 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188229 [I][Elf] Total size of loaded sections: 1428
188230 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188274 [I][Elf] Total size of loaded sections: 1904
188275 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188327 [I][Elf] Total size of loaded sections: 1880
188328 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188373 [I][Elf] Total size of loaded sections: 1576
188374 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188415 [I][Elf] Total size of loaded sections: 652
188416 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188471 [I][Elf] Total size of loaded sections: 1100
188472 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188473 [D][NfcSupportedCards] Loaded 11 plugins
188488 [D][Iso14443_4aPoller] Read ATS success
188489 [I][MfDesfirePoller] CMD: 60
188496 [D][MfDesfirePoller] Read version success
188497 [I][MfDesfirePoller] CMD: 6E
188499 [D][MfDesfirePoller] Read free memory success
188499 [I][MfDesfirePoller] CMD: 45
188501 [D][MfDesfirePoller] Read master key settings success
188501 [I][MfDesfirePoller] CMD: 64
188503 [D][MfDesfirePoller] Read master key version success
188503 [I][MfDesfirePoller] CMD: 6A
188508 [D][MfDesfirePoller] Read application ids success
188508 [I][MfDesfirePoller] CMD: 5A
188511 [I][MfDesfirePoller] CMD: 45
188513 [I][MfDesfirePoller] CMD: 64
188515 [I][MfDesfirePoller] CMD: 64
188517 [I][MfDesfirePoller] CMD: 64
188519 [I][MfDesfirePoller] CMD: 64
188522 [I][MfDesfirePoller] CMD: 64
188524 [I][MfDesfirePoller] CMD: 64
188526 [I][MfDesfirePoller] CMD: 64
188528 [I][MfDesfirePoller] CMD: 64
[... 67 lines cut ...]
188681 [I][MfDesfirePoller] CMD: F5
188684 [I][MfDesfirePoller] CMD: F5
188687 [I][MfDesfirePoller] CMD: BD
188693 [I][MfDesfirePoller] CMD: BD
188696 [E][MfDesfirePoller] Error code: AE
188696 [D][MfDesfirePoller] Read applications success
188697 [D][MfDesfirePoller] Read success.
188697 [D][Nfc] FWT Timeout
188775 [I][Elf] Total size of loaded sections: 1312
188776 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188820 [I][Elf] Total size of loaded sections: 1428
188820 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
188930 [D][DolphinState] icounter 62, butthurt 0
203837 [I][MfDesfire] Saving iso-4 data
203837 [I][MfDesfire] Saving comment
203837 [I][MfDesfire] Saving version
203839 [I][MfDesfire] Saving free memory
203841 [I][MfDesfire] Saving key settings
203843 [I][MfDesfire] Saving key version
203843 [I][MfDesfire] Saving key version 0
203843 [I][MfDesfire] Saving app count
203845 [I][MfDesfire] Saving app ids
203845 [I][MfDesfire] Saving app 0
203855 [I][MfDesfire] Saving app 1
203861 [I][MfDesfire] Saving app 2

scapermoya, 2024-02-07:
Hey @gornekich, any new thoughts about this? thanks for all of your help so far!

gornekich, 2024-02-13:
@scapermoya sorry for delay, I am back working on this issue. Could you please update firmware from #3395 and repeat procedure? Also, does your flipper freeze if you open info and scroll to 2nd application after reading your card?

scapermoya, 2024-02-14:
@gornekich thanks for continuing to work on this.

I updated the firmware and the same behavior occurs as before. Hangs at an hourglass after trying to save. Have to reset with the back+left button to reboot the device.

I am not sure what you mean by opening info and scrolling to the second application. I can scan the fob and then go into info and I see a lot of data about the fob on the first page. I can also go to "more" and see "card info" and then a bunch of "app" with 6 digit hex (I think) codes. Going into them does indeed cause freezing.

log of scan and save:

>: log
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
13577 [I][Loader] Loading /ext/apps/NFC/nfc.fap
13691 [I][Elf] Total size of loaded sections: 65521
13692 [I][Loader] Loaded in 115ms
13693 [I][AnimationManager] Unload animation 'L1_Waves_128x50'
14954 [D][NfcScanner] Found 5 base protocols
14955 [D][DolphinState] icounter 62, butthurt 0
14967 [D][Nfc] FWT Timeout
...
17618 [D][Nfc] FWT Timeout
17633 [D][NfcScanner] Found 4 children
17688 [D][Nfc] FWT Timeout
17689 [D][Nfc] FWT Timeout
17727 [D][Nfc] FWT Timeout
17761 [D][Iso14443_4aPoller] Read ATS success
17761 [I][MfDesfirePoller] CMD: 60
17782 [I][NfcScanner] Detected 1 protocols
17949 [I][Elf] Total size of loaded sections: 980
17950 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
17991 [I][Elf] Total size of loaded sections: 496
17992 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18039 [I][Elf] Total size of loaded sections: 3428
18039 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18098 [I][Elf] Total size of loaded sections: 320
18099 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18147 [I][Elf] Total size of loaded sections: 1584
18147 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18190 [I][Elf] Total size of loaded sections: 940
18191 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18233 [I][Elf] Total size of loaded sections: 627
18234 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18282 [I][Elf] Total size of loaded sections: 1636
18282 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18330 [I][Elf] Total size of loaded sections: 1852
18330 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18374 [I][Elf] Total size of loaded sections: 1176
18374 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18424 [I][Elf] Total size of loaded sections: 699
18425 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18469 [I][Elf] Total size of loaded sections: 1419
18470 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18514 [I][Elf] Total size of loaded sections: 1908
18515 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18558 [I][Elf] Total size of loaded sections: 2376
18559 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18605 [I][Elf] Total size of loaded sections: 1588
18605 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18647 [I][Elf] Total size of loaded sections: 652
18648 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18691 [I][Elf] Total size of loaded sections: 1104
18692 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
18693 [D][NfcSupportedCards] Loaded 17 plugins
18708 [D][Iso14443_4aPoller] Read ATS success
18708 [I][MfDesfirePoller] CMD: 60
18716 [D][MfDesfirePoller] Read version success
18716 [I][MfDesfirePoller] CMD: 6E
18718 [D][MfDesfirePoller] Read free memory success
18718 [I][MfDesfirePoller] CMD: 45
18720 [D][MfDesfirePoller] Read master key settings success
[... 98 lines cut ...]
23626 [I][MfDesfire] Saving iso-4 data
23626 [I][MfDesfire] Saving comment
23626 [I][MfDesfire] Saving version
23628 [I][MfDesfire] Saving free memory
23630 [I][MfDesfire] Saving key settings
23632 [I][MfDesfire] Saving key version
23632 [I][MfDesfire] Saving key version 0
23632 [I][MfDesfire] Saving app count
23634 [I][MfDesfire] Saving app ids
23634 [I][MfDesfire] Saving app 0
23635 [I][MfDesfire] Saving key settings
23647 [I][MfDesfire] Saving file ids. File count: 1
23648 [I][MfDesfire] Saving file settings for file 0
23649 [I][MfDesfire] Saving file data for file 0
23649 [I][MfDesfire] Saving data size 22
23650 [I][MfDesfire] Saving app 1
23650 [I][MfDesfire] Saving key settings
23655 [I][MfDesfire] Saving file ids. File count: 1
23655 [I][MfDesfire] Saving file settings for file 0
23656 [I][MfDesfire] Saving file data for file 0
23656 [I][MfDesfire] Saving data size 22
23657 [I][MfDesfire] Saving app 2
23658 [I][MfDesfire] Saving key settings
23663 [I][MfDesfire] Saving file ids. File count: 2
23663 [I][MfDesfire] Saving file settings for file 0

gornekich, 2024-02-14:
@scapermoya  thanks a lot. Still I see lack of info from logs to resolve the issue. Please, update firmware from #3395 and repeat procedure. Hopefully for last time, since there should be all necessary logs.

scapermoya, 2024-02-14:
@gornekich 

new log:

>: log
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
48462 [D][StorageInt] Device sync: skipping
54686 [I][Loader] Loading /ext/apps/NFC/nfc.fap
54800 [I][Elf] Total size of loaded sections: 65521
54801 [I][Loader] Loaded in 115ms
54802 [I][AnimationManager] Unload animation 'L1_Sleep_128x64'
58160 [D][NfcScanner] Found 5 base protocols
58161 [D][DolphinState] icounter 62, butthurt 0
58201 [D][Nfc] FWT Timeout
58226 [D][Nfc] FWT Timeout
58278 [D][Nfc] FWT Timeout
58306 [D][Nfc] FWT Timeout
58321 [D][NfcScanner] Found 4 children
58376 [D][Nfc] FWT Timeout
58377 [D][Nfc] FWT Timeout
58412 [D][Nfc] FWT Timeout
58445 [D][Iso14443_4aPoller] Read ATS success
58445 [I][MfDesfirePoller] Poller: 60
58448 [I][MfDesfirePoller] Tag: AF 04 01 02 12 00 1A 05
58449 [I][MfDesfirePoller] Poller: AF 04 01 02 12 00 1A 05
58451 [I][MfDesfirePoller] Tag: AF 04 01 01 02 01 1A 05
58452 [I][MfDesfirePoller] Poller: AF 04 01 01 02 01 1A 05
58455 [I][MfDesfirePoller] Tag: 00 04 22 2F 4A 89 0F 90 CF 5D 14 55 40 40 21
58477 [I][NfcScanner] Detected 1 protocols
58657 [I][Elf] Total size of loaded sections: 980
58657 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
58705 [I][Elf] Total size of loaded sections: 496
58706 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
58752 [I][Elf] Total size of loaded sections: 3428
58753 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
58794 [I][Elf] Total size of loaded sections: 320
58795 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
58843 [I][Elf] Total size of loaded sections: 1584
58843 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
58886 [I][Elf] Total size of loaded sections: 940
58887 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
58929 [I][Elf] Total size of loaded sections: 627
58930 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
58978 [I][Elf] Total size of loaded sections: 1636
58979 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59032 [I][Elf] Total size of loaded sections: 1852
59033 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59075 [I][Elf] Total size of loaded sections: 1176
59076 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59119 [I][Elf] Total size of loaded sections: 699
59120 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59164 [I][Elf] Total size of loaded sections: 1419
59165 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59209 [I][Elf] Total size of loaded sections: 1908
59210 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59253 [I][Elf] Total size of loaded sections: 2376
59254 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59300 [I][Elf] Total size of loaded sections: 1588
59300 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59342 [I][Elf] Total size of loaded sections: 652
59343 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59392 [I][Elf] Total size of loaded sections: 1104
59393 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
59394 [D][NfcSupportedCards] Loaded 17 plugins
59409 [D][Iso14443_4aPoller] Read ATS success
59409 [I][MfDesfirePoller] Poller: 60
59412 [I][MfDesfirePoller] Tag: AF 04 01 02 12 00 1A 05
59413 [I][MfDesfirePoller] Poller: AF 04 01 02 12 00 1A 05
[... 159 lines cut ...]
73189 [I][MfDesfire] Saving key settings
73196 [I][MfDesfire] Saving file ids. File count: 1
73196 [I][MfDesfire] Saving file settings for file 0
73196 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_TYPE_KEY
73197 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_COMM_SETTINGS_KEY
73197 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_ACCESS_RIGHTS_KEY
73198 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_SIZE_KEY
73198 [I][MfDesfire] Saving file data for file 0
73198 [I][MfDesfire] Saving data size 22
73199 [I][MfDesfire] Saving app 1
73199 [I][MfDesfire] Saving key settings
73204 [I][MfDesfire] Saving file ids. File count: 1
73205 [I][MfDesfire] Saving file settings for file 0
73205 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_TYPE_KEY
73205 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_COMM_SETTINGS_KEY
73206 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_ACCESS_RIGHTS_KEY
73206 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_SIZE_KEY
73207 [I][MfDesfire] Saving file data for file 0
73207 [I][MfDesfire] Saving data size 22
73208 [I][MfDesfire] Saving app 2
73208 [I][MfDesfire] Saving key settings
73213 [I][MfDesfire] Saving file ids. File count: 2
73214 [I][MfDesfire] Saving file settings for file 0
73214 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_TYPE_KEY
73214 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE

gornekich, 2024-02-14:
> 59619 [I][MfDesfirePoller] Poller: BD 00 00 00 73183 [I][MfDesfire] Saving iso-4 data

This line looks abnormal, and it's might be critical for resolving this issue? Can you confirm that you can see this log every time you read the card?
There flipper sends command to read data from 1st file of 2nd application. And it looks like there's no response. However if there is no response, the error should be printed as well. Also the absence of newline character also is very wierd

scapermoya, 2024-02-14:
@gornekich There is definitely some weird behavior.

This is what happens when I "Read" the fob:

71704 [D][NfcScanner] Found 5 base protocols
71708 [D][DolphinState] icounter 62, butthurt 0
71727 [D][ViewDispatcher] View changed while key press 200143E0 -> 2000CAF0. Sending key: Right, type: Release, sequence: 00000060 to previous view port
71745 [D][Nfc] FWT Timeout
71770 [D][Nfc] FWT Timeout
71833 [D][Nfc] FWT Timeout
71861 [D][Nfc] FWT Timeout
71876 [D][NfcScanner] Found 4 children
71931 [D][Nfc] FWT Timeout
71932 [D][Nfc] FWT Timeout
71967 [D][Nfc] FWT Timeout
72000 [D][Iso14443_4aPoller] Read ATS success
72000 [I][MfDesfirePoller] Poller: 60
72003 [I][MfDesfirePoller] Tag: AF 04 01 02 12 00 1A 05
72004 [I][MfDesfirePoller] Poller: AF 04 01 02 12 00 1A 05
72006 [I][MfDesfirePoller] Tag: AF 04 01 01 02 01 1A 05
72007 [I][MfDesfirePoller] Poller: AF 04 01 01 02 01 1A 05
72010 [I][MfDesfirePoller] Tag: 00 04 22 2F 4A 89 0F 90 CF 5D 14 55 40 40 21
72032 [I][NfcScanner] Detected 1 protocols
72199 [I][Elf] Total size of loaded sections: 980
72200 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72242 [I][Elf] Total size of loaded sections: 496
72243 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72289 [I][Elf] Total size of loaded sections: 3428
72290 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72331 [I][Elf] Total size of loaded sections: 320
72332 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72379 [I][Elf] Total size of loaded sections: 1584
72380 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72423 [I][Elf] Total size of loaded sections: 940
72424 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72472 [I][Elf] Total size of loaded sections: 627
72473 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72520 [I][Elf] Total size of loaded sections: 1636
72521 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72568 [I][Elf] Total size of loaded sections: 1852
72569 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72612 [I][Elf] Total size of loaded sections: 1176
72612 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72656 [I][Elf] Total size of loaded sections: 699
72657 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72701 [I][Elf] Total size of loaded sections: 1419
72701 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72746 [I][Elf] Total size of loaded sections: 1908
72746 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72796 [I][Elf] Total size of loaded sections: 2376
72797 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72855 [I][Elf] Total size of loaded sections: 1588
72855 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72897 [I][Elf] Total size of loaded sections: 652
72898 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72941 [I][Elf] Total size of loaded sections: 1104
72942 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
72943 [D][NfcSupportedCards] Loaded 17 plugins
72958 [D][Iso14443_4aPoller] Read ATS success
72958 [I][MfDesfirePoller] Poller: 60
72961 [I][MfDesfirePoller] Tag: AF 04 01 02 12 00 1A 05
72962 [I][MfDesfirePoller] Poller: AF 04 01 02 12 00 1A 05
72964 [I][MfDesfirePoller] Tag: AF 04 01 01 02 01 1A 05
72965 [I][MfDesfirePoller] Poller: AF 04 01 01 02 01 1A 05
72968 [I][MfDesfirePoller] Tag: 00 04 22 2F 4A 89 0F 90 CF 5D 14 55 40 40 21
72969 [D][MfDesfirePoller] Read version success
72969 [I][MfDesfirePoller] Poller: 6E
72971 [I][MfDesfirePoller] Tag: 00 00 0E 00
72971 [D][MfDesfirePoller] Read free memory success
72972 [I][MfDesfirePoller] Poller: 45
[... 168 lines cut ...]
209812 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_COMM_SETTINGS_KEY
209812 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_ACCESS_RIGHTS_KEY
209813 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_SIZE_KEY
209813 [I][MfDesfire] Saving file data for file 0
209813 [I][MfDesfire] Saving data size 22
209814 [I][MfDesfire] Saving app 1
209815 [I][MfDesfire] Saving key settings
209819 [I][MfDesfire] Saving file ids. File count: 1
209820 [I][MfDesfire] Saving file settings for file 0
209820 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_TYPE_KEY
209821 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_COMM_SETTINGS_KEY
209821 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_ACCESS_RIGHTS_KEY
209821 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_SIZE_KEY
209822 [I][MfDesfire] Saving file data for file 0
209822 [I][MfDesfire] Saving data size 22
209823 [I][MfDesfire] Saving app 2
209823 [I][MfDesfire] Saving key settings
209828 [I][MfDesfire] Saving file ids. File count: 2
209829 [I][MfDesfire] Saving file settings for file 0
209829 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_TYPE_KEY
209830 [I][MfDesfire] Saving MF_DESFIRE_FFF_

and it hangs at the hourglass

so I think the reason it double printed the lines previously as you pointed out is that i navigated and attempted to save the data quickly the last time

gornekich, 2024-03-04:
@scapermoya could you please repeat procedure with firmware from #3395 

Btw, is it possible to get one of this card? Where is it used?

scapermoya, 2024-03-14:
@gornekich repeated, logs are below.

It still hangs mid-read (second line that starts with 61180) for multiple seconds, and then prints new information on that same line starting with the timestamp 91540. This happens before I even attempt to save. Then when I go to save, it endlessly hangs midline at the bottom of the log.

Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
54883 [I][Loader] Loading /ext/apps/NFC/nfc.fap
54999 [I][Elf] Total size of loaded sections: 65521
55000 [I][Loader] Loaded in 117ms
55003 [I][AnimationManager] Unload animation 'L1_Recording_128x51'
59715 [D][NfcScanner] Found 5 base protocols
59716 [D][DolphinState] icounter 63, butthurt 0
59756 [D][Nfc] FWT Timeout
59781 [D][Nfc] FWT Timeout
59844 [D][Nfc] FWT Timeout
59872 [D][Nfc] FWT Timeout
59887 [D][NfcScanner] Found 4 children
59942 [D][Nfc] FWT Timeout
59943 [D][Nfc] FWT Timeout
59978 [D][Nfc] FWT Timeout
60011 [D][Iso14443_4aPoller] Read ATS success
60011 [W][MfDesfirePoller] Free size: 44680
60011 [I][MfDesfirePoller] Poller: 60
60014 [I][MfDesfirePoller] Tag: AF 04 01 02 12 00 1A 05
60015 [I][MfDesfirePoller] Poller: AF
60017 [I][MfDesfirePoller] Tag: AF 04 01 01 02 01 1A 05
60018 [I][MfDesfirePoller] Poller: AF
60021 [I][MfDesfirePoller] Tag: 00 04 22 2F 4A 89 0F 90 CF 5D 14 55 40 40 21
60043 [I][NfcScanner] Detected 1 protocols
60211 [I][Elf] Total size of loaded sections: 980
60212 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60254 [I][Elf] Total size of loaded sections: 496
60255 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60300 [I][Elf] Total size of loaded sections: 2264
60301 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60348 [I][Elf] Total size of loaded sections: 3428
60348 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60391 [I][Elf] Total size of loaded sections: 320
60391 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60441 [I][Elf] Total size of loaded sections: 1584
60442 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60492 [I][Elf] Total size of loaded sections: 940
60493 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60536 [I][Elf] Total size of loaded sections: 627
60537 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60585 [I][Elf] Total size of loaded sections: 1636
60586 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60635 [I][Elf] Total size of loaded sections: 1852
60636 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60691 [I][Elf] Total size of loaded sections: 1176
60692 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60736 [I][Elf] Total size of loaded sections: 699
60737 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60782 [I][Elf] Total size of loaded sections: 1419
60782 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60833 [I][Elf] Total size of loaded sections: 1908
60834 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60879 [I][Elf] Total size of loaded sections: 2376
60880 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60926 [I][Elf] Total size of loaded sections: 1588
60927 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
60969 [I][Elf] Total size of loaded sections: 652
60970 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
61015 [I][Elf] Total size of loaded sections: 1104
61016 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
61017 [D][NfcSupportedCards] Loaded 18 plugins
61032 [D][Iso14443_4aPoller] Read ATS success
61032 [W][MfDesfirePoller] Free size: 44320
61032 [I][MfDesfirePoller] Poller: 60
[... 159 lines cut ...]
107905 [I][MfDesfire] Saving file settings for file 0
107906 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_TYPE_KEY
107906 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_COMM_SETTINGS_KEY
107907 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_ACCESS_RIGHTS_KEY
107907 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_SIZE_KEY
107908 [I][MfDesfire] Saving file data for file 0
107908 [I][MfDesfire] Saving data size 22
107909 [I][MfDesfire] Saving app 1
107909 [I][MfDesfire] Saving key settings
107914 [I][MfDesfire] Saving file ids. File count: 1
107915 [I][MfDesfire] Saving file settings for file 0
107915 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_TYPE_KEY
107916 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_COMM_SETTINGS_KEY
107916 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_ACCESS_RIGHTS_KEY
107917 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_SIZE_KEY
107917 [I][MfDesfire] Saving file data for file 0
107918 [I][MfDesfire] Saving data size 22
107918 [I][MfDesfire] Saving app 2
107919 [I][MfDesfire] Saving key settings
107924 [I][MfDesfire] Saving file ids. File count: 2
107925 [I][MfDesfire] Saving file settings for file 0
107925 [I][MfDesfire] Saving MF_DESFIRE_FFF_FILE_TYPE_KEY
107926 [I][MfDesfire] Saving MF_DESFIRE_FFF_

in terms of the fob itself, it is a security fob for the apartment that I live in. I cannot send it.

gornekich, 2024-04-16:
The fix arrived in dev. Please, try to reproduce the issue with latest dev.

closed 2024-07-07 by skotopes