flipperdevices/flipperzero-firmware #3340: BusFault in furi hal doing picopass emulation

nvx, 2024-01-04:
### Describe the bug.

I'm one of the devs working on the picopass fap. While emulating I intermittently would notice the flipper crash. I threw the dev board on and captured a stack trace but surprisingly my code wasn't in the back trace at all.

Running 0.97.1 Release and picopass from this branch, although I suspect the current release version has the same issue - https://github.com/nvx/flipperzero-good-faps/tree/feature/picopass_non_secure/picopass

### Reproduction

Emulate a picopass card against a proxmark3 and run `hf ic dump --ns --ki 0` a bunch of times and it'll eventually crash. Note there seems to be another issue where emulation just stops randomly (looks fine on the flipper but isn't responding on the NFC interface) - if this happens just hit back then hit emulate again and dumping until you run into a crash.

### Logs

```Text
(gdb) bt full
#0  0x080104ea in __furi_crash_implementation () at furi/core/check.c:164
        r0 = 33
        isr = true
        debug = 184
#1  0x080067c6 in BusFault_Handler () at targets/f7/furi_hal/furi_hal_interrupt.c:282
        r12 = 0x20031334 <__furi_check_registers>
#2  <signal handler called>
No symbol table info available.
#3  0x0803f088 in furi_hal_sw_digital_pin_dma_rx_isr (context=0x20019d78) at lib/signal_reader/signal_reader.c:150
        bit = <optimized out>
        i = 0
        instance = 0x20019d78
        gpio_buff_start = <optimized out>
        bitstream_buff_start = 0x20019db8 ""
#4  <signal handler called>
No symbol table info available.
#5  vPortSuppressTicksAndSleep (expected_idle_ticks=35) at targets/f7/furi_hal/furi_hal_os.c:173
        unexpected_idle_ticks = <optimized out>
#6  0x08014d04 in prvIdleTask (pvParameters=<optimized out>) at lib/FreeRTOS-Kernel/tasks.c:3524
        xExpectedIdleTime = <optimized out>
#7  0x08012a1a in furi_thread_catch () at furi/core/thread.c:63
No locals.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

nvx, 2024-01-04:
I'd noticed sometimes the crashes had different messages before I had gotten the debug probe out, so I thought I'd try it again to see if I could get a different stack trace:

```
Program received signal SIGTRAP, Trace/breakpoint trap.
0x080104ea in __furi_crash_implementation () at furi/core/check.c:164
164     furi/core/check.c: No such file or directory.
New application loaded. Adding debug info
Loading debug information from /home/nvx/.ufbt/build/picopass_d.elf
add symbol table from file "/home/nvx/.ufbt/build/picopass_d.elf" at
        .text_addr = 0x2000e06c
        .bss_addr = 0x2000a324
        .data_addr = 0x2000a2cc
        .rodata_addr = 0x20016a04
(gdb) bt full
#0  0x080104ea in __furi_crash_implementation () at furi/core/check.c:164
        r0 = 64
        isr = true
        debug = false
#1  0x08006796 in MemManage_Handler () at targets/f7/furi_hal/furi_hal_interrupt.c:271
        r12 = 0x20031334 <__furi_check_registers>
        memfault_address = <optimized out>
#2  <signal handler called>
No symbol table info available.
#3  0x0802fd90 in bit_buffer_append_byte (buf=0x2000c158, byte=228 '\344') at lib/toolbox/bit_buffer.c:308
        data_size_bytes = 1088
        new_data_size_bytes = 1089
#4  0x0803ed1c in iso15693_parser_parse_1_out_of_256 (instance=0x2000c100) at lib/signal_reader/parsers/iso15693/iso15693_parser.c:242
        j = 0
        i = <optimized out>
        command = Iso15693ParserCommandWaitData
        eof = 4 '\004'
#5  0x0803ef48 in iso15693_parser_run (instance=0x2000c100) at lib/signal_reader/parsers/iso15693/iso15693_parser.c:270
        command = Iso15693ParserCommandProcessed
#6  0x08007a8a in furi_hal_nfc_iso15693_wait_event (timeout_ms=4294967295) at targets/f7/furi_hal/furi_hal_nfc_iso15693.c:378
        flag = 16
        event = 0
        thread_id = <optimized out>
#7  0x08037760 in nfc_worker_listener (context=0x2000a568) at lib/nfc/nfc.c:121
        event = <optimized out>
        instance = 0x2000a568
        event_data = {
          buffer = 0x2000cb40
        }
        nfc_event = {
          type = NfcEventTypeRxEnd,
          data = {
            buffer = 0x2000cb40
          }
        }
        command = <optimized out>
#8  0x08012a5c in furi_thread_body (context=0x2000a180) at furi/core/thread.c:92
        thread = 0x2000a180
        task_handle = 0x2000d820
#9  0x08012a1a in furi_thread_catch () at furi/core/thread.c:63
No locals.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

Might be a multiple issues?

hedger, 2024-01-04:
This looks like a generic memory corruption.