flipperdevices/flipperzero-firmware #3341: NFC-B Support

lemonchu, 2024-01-05:
### Description of the feature you're suggesting.

NFC-B Simulate Request

The 0.97.1 release can read NFC-B card now, but it cannot stimulate NFC-B. If trigger a simulation, the simulation will not work and showed an error of “NULL pointer dereference”

skotopes, 2024-01-05:
NFC-B emulation is not yet here, how did you manage to emulate it?

lemonchu, 2024-01-05:
> NFC-B emulation is not yet here, how did you manage to emulate it?

In the app for iOS.

Click it and it appears a button to emulate. 
[image]

skotopes, 2024-01-05:
That is a bug then. Firmware must tell you that emulation of NFC-B is not supported

lemonchu, 2024-01-05:
Well, I am still wondering whether the NFC-B emulation will be supported in the future. Thanks~

skotopes, 2024-01-05:
It will be

gornekich, 2024-04-15:
@lemonchu on latest releases you will see card info instead of emulation attempt when opening your card. 

Unfortunately it's impossible to emulate NFC-B on flipper zero.

closed 2024-04-15 by gornekich

skotopes, 2024-04-23:
@yagueto hardware limitations: nfc-b emulation is not support on hardware level by the NFC chip, software emulation is not possible because of lack of computing resources to do it in real time.

AntoineZoghbi, 2024-10-02:
@skotopes, will there ever be an NFC-B emulation support on the current Flipper Zero, or does there need to be a hardware update to do so?