flipperdevices/flipperzero-firmware #3345: subGhz: Keeloq: DTM System - remote protocol request

AABatteries, 2024-01-06:
### Description of the feature you're suggesting.

I'm hoping that somebody can add support for DTM System (Keeloq based) protocol - I know DTM Neo protocol is already implemented but this seems to be a different protocol - maybe very similar so easy to add?
The transmission frequency is 433,868 MHz with AM modulation

# Subfiles
Below you will find 4 files with 10 keypresses from my 4-button remote, recorded as RAW files
[RAW_20231224-pola1.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13851597/RAW_20231224-pola1.sub.txt)
[RAW_20231224-pola2.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13851598/RAW_20231224-pola2.sub.txt)
[RAW_20231224-pola3.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13851599/RAW_20231224-pola3.sub.txt)
[RAW_20231224-pola4.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13851600/RAW_20231224-pola4.sub.txt)

# Photos of the PCB
[image]
[image]

Thanks a lot!

skotopes, 2024-01-09:
@AABatteries it's a little bit more complicated, we need this system in our office to figure out what kind of keeloq learning scheme used and if it got static manufacture key.

@Skorpionm can you guide @AABatteries throw the process?

Skorpionm, 2024-01-09:
https://github.com/flipperdevices/flipperzero-firmware/issues/3085
This is where a similar system is used. To try to subtract a manufactory key from a key fob, I physically need an HCS301 microscope on my desk. you need to remove 1000+ very clean traces during nitring and use them to restore the factory key. BUT this does NOT guarantee that it will fit the entire system as in the topic above

Skorpionm, 2024-04-16:
I can’t help but encrypt by records

closed 2024-04-16 by Skorpionm

carlosD2884MK, 2025-12-17:
> analyzing the HCS301 power traces

Hello, How can I perform the power trace analysis of the HC301?