flipperdevices/flipperzero-firmware #3348: NFC: unable to read DESFire fob

Tomi2965, 2024-01-07:
### Describe the bug.

I am unable to read one DESFire fob. Same result when I select read DESFire. When I select read MFC, it works. 
[image]
[image]

Sorry for short info. If any info is needed, ask please. 

Fob info: https://www.rys.sk/html_sk/500cl3a04sx.php

### Reproduction

Scan that token... :) If any help needed, do not hestitate to ask please. 

### Logs

```Text
See image (sorry, long story)
```

skotopes, 2024-01-10:
Known issues. Fix coming soon.

Tomi2965, 2024-01-11:
@skotopes thank you for info. I'll wait for fix.

philicious, 2024-01-25:
FYI: I still get that same error with current RC `0.98.0-rc`
> 655685 [E][MfDesfirePoller] Failed to read master key settings

gornekich, 2024-01-29:
Hello @Tomi2965 , @philicious 
Can't reproduce this issue with my tags. Could you please install firmware from #3395 , collect logs while reading card and share logs here?

philicious, 2024-01-29:
Can and will do in next few days.

Tomi2965, 2024-01-30:
@gornekich tryed, see log. If needed anything else, please ask. 

>: log trace
Current log level: trace
Use <log ?> to list available log levels
Press CTRL+C to stop...
1686461 [T][StorageApi] File/Dir 0000C520 alloc
1686463 [I][Loader] Loading /ext/apps/NFC/nfc.fap
1686467 [T][StorageApi] File 0000C520 - 0000C6CC open (/ext/apps/NFC/nfc.fap)
1686617 [I][Elf] Total size of loaded sections: 60625
1686620 [T][StorageApi] File 0000C520 - 0000C6CC closed
1686623 [T][StorageApi] File/Dir 0000C520 free
1686625 [I][Loader] Loaded in 162ms
1686630 [I][AnimationManager] Unload animation 'L1_Sleep_128x64'
1694378 [D][NfcScanner] Found 5 base protocols
1694386 [D][DolphinState] icounter 3444, butthurt 0
1694397 [D][Nfc] FWT Timeout
1694399 [T][Nfc] Failed TRX state machine
1694435 [D][Nfc] FWT Timeout
1694437 [T][Nfc] Failed TRX state machine
1694471 [D][Nfc] FWT Timeout
1694472 [T][Nfc] Failed TRX state machine
1694535 [D][Nfc] FWT Timeout
1694537 [T][Nfc] Failed TRX state machine
1694539 [T][FelicaPoller] Activation failed error: 8
1694563 [D][Nfc] FWT Timeout
1694564 [T][Nfc] Failed TRX state machine
1694567 [T][ST25TBPoller] error during trx: 2
1694599 [D][Nfc] FWT Timeout
1694601 [T][Nfc] Failed TRX state machine
1694625 [D][Nfc] FWT Timeout
1694627 [T][Nfc] Failed TRX state machine
1694661 [D][Nfc] FWT Timeout
1694662 [T][Nfc] Failed TRX state machine
1694725 [D][Nfc] FWT Timeout
1694727 [T][Nfc] Failed TRX state machine
1694729 [T][FelicaPoller] Activation failed error: 8
1694764 [D][Nfc] FWT Timeout
1694765 [T][Nfc] Failed TRX state machine
1694768 [T][ST25TBPoller] error during trx: 2
1694800 [D][Nfc] FWT Timeout
1694802 [T][Nfc] Failed TRX state machine
1694826 [D][Nfc] FWT Timeout
1694828 [T][Nfc] Failed TRX state machine
1694862 [D][Nfc] FWT Timeout
1694863 [T][Nfc] Failed TRX state machine
1694926 [D][Nfc] FWT Timeout
1694928 [T][Nfc] Failed TRX state machine
1694930 [T][FelicaPoller] Activation failed error: 8
1694954 [D][Nfc] FWT Timeout
1694955 [T][Nfc] Failed TRX state machine
1694958 [T][ST25TBPoller] error during trx: 2
1694993 [T][ISO14443_3A] Sel resp: 24
1694997 [T][ISO14443_3A] Sel resp: 20
1694999 [T][ISO14443_3A] Col resolution complete
1695027 [D][Nfc] FWT Timeout
1695029 [T][Nfc] Failed TRX state machine
1695063 [D][Nfc] FWT Timeout
1695064 [T][Nfc] Failed TRX state machine
1695127 [D][Nfc] FWT Timeout
1695129 [T][Nfc] Failed TRX state machine
1695131 [T][FelicaPoller] Activation failed error: 8
1695155 [D][Nfc] FWT Timeout
1695156 [T][Nfc] Failed TRX state machine
1695159 [T][ST25TBPoller] error during trx: 2
1695183 [D][NfcScanner] Found 4 children
1695196 [T][ISO14443_3A] Sel resp: 24
1695200 [T][ISO14443_3A] Sel resp: 20
1695202 [T][ISO14443_3A] Col resolution complete
1695232 [T][ISO14443_3A] Sel resp: 24
[... 142 lines cut ...]
1698566 [T][Nfc] Failed TRX state machine
1698669 [D][Nfc] FWT Timeout
1698671 [T][Nfc] Failed TRX state machine
1698774 [D][Nfc] FWT Timeout
1698776 [T][Nfc] Failed TRX state machine
1698879 [D][Nfc] FWT Timeout
1698881 [T][Nfc] Failed TRX state machine
1698984 [D][Nfc] FWT Timeout
1698986 [T][Nfc] Failed TRX state machine
1699089 [D][Nfc] FWT Timeout
1699091 [T][Nfc] Failed TRX state machine
1699194 [D][Nfc] FWT Timeout
1699196 [T][Nfc] Failed TRX state machine
1699299 [D][Nfc] FWT Timeout
1699301 [T][Nfc] Failed TRX state machine
1699404 [D][Nfc] FWT Timeout
1699406 [T][Nfc] Failed TRX state machine
1699509 [D][Nfc] FWT Timeout
1699511 [T][Nfc] Failed TRX state machine
1699614 [D][Nfc] FWT Timeout
1699616 [T][Nfc] Failed TRX state machine
1699719 [D][Nfc] FWT Timeout
1699721 [T][Nfc] Failed TRX state machine
1699824 [D][Nfc] FWT Timeout
1699826 [T][Nfc] Failed TRX state machine

philicious, 2024-01-30:
@gornekich excerpt from my log. looks pretty much the same as for @Tomi2965 

```bash
nfc-list

1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 03  44
       UID (NFCID1): 04  xx  xx  62  6a  13  90
      SAK (SEL_RES): 20
                ATS: 75  77  81  02  80
```

PS: unfortunately I am loosing access to this fob tomorrow. so wont be able to further help

```
95999 [T][ISO14443_3A] Sel resp: 24
96003 [T][ISO14443_3A] Sel resp: 20
96005 [T][ISO14443_3A] Col resolution complete
96012 [D][Nfc] FWT Timeout
96014 [T][Nfc] Failed TRX state machine
96050 [T][ISO14443_3A] Sel resp: 24
96054 [T][ISO14443_3A] Sel resp: 20
96056 [T][ISO14443_3A] Col resolution complete
96060 [D][Iso14443_4aPoller] Read ATS success
96086 [I][NfcScanner] Detected 1 protocols
96216 [T][ISO14443_3A] Sel resp: 24
96220 [T][ISO14443_3A] Sel resp: 20
96222 [T][ISO14443_3A] Col resolution complete
96226 [D][Iso14443_4aPoller] Read ATS success
96234 [D][MfDesfirePoller] Read version success
96238 [D][MfDesfirePoller] Read free memory success
96242 [E][MfDesfire] Key setting incorrect length: 0
96244 [E][MfDesfirePoller] Failed to parse read key settings cmd
96248 [E][MfDesfirePoller] Failed to read master key settings
96252 [D][Nfc] FWT Timeout
96253 [T][Nfc] Failed TRX state machine
96256 [D][Nfc] FWT Timeout
96257 [T][Nfc] Failed TRX state machine
96360 [D][Nfc] FWT Timeout
96361 [T][Nfc] Failed TRX state machine
```

gornekich, 2024-01-30:
@Tomi2965 @philicious thanks a lot for testing. I think that we get authentication error while trying to read master key. To be sure, could you please update firmware from #3395 once again, read the tag and share logs? Log level Debug will be enough

Tomi2965, 2024-01-30:
@gornekich, if it help with diag, I am able to attach that fob to FZ's back and give you full access to remote desktop/other service of virtual machine, with FZ connected to USB. If you agree, just tell me, including type/version of OS (email, WA, Discord?).

Then you will be able to test anything directly, without any delay. It's much faster and cheaper to order/deliver that fob's :)

gornekich, 2024-01-30:
@Tomi2965 it's a great idea. Let's start with the logs, may be I fix the issue in next iteration. If not, we can proceed fixing issue like you suggested

Tomi2965, 2024-01-30:
@gornekich, reinstalled from #3395, then again readed fob with "log debug". Attaching log. 

: log debug
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
32194 [D][GattChar] Updating Power State char
32199 [D][GattChar] Updating Battery Level char
32225 [I][Loader] Loading /ext/apps/NFC/nfc.fap
32360 [I][Elf] Total size of loaded sections: 60625
32363 [I][Loader] Loaded in 138ms
32367 [I][AnimationManager] Unload animation 'L1_My_dude_128x64'
38916 [D][NfcScanner] Found 5 base protocols
38922 [D][DolphinState] icounter 3445, butthurt 0
38935 [D][Nfc] FWT Timeout
38962 [D][Nfc] FWT Timeout
38987 [D][Nfc] FWT Timeout
39039 [D][Nfc] FWT Timeout
39067 [D][Nfc] FWT Timeout
39090 [D][Nfc] FWT Timeout
39115 [D][Nfc] FWT Timeout
39140 [D][Nfc] FWT Timeout
39192 [D][Nfc] FWT Timeout
39220 [D][Nfc] FWT Timeout
39243 [D][Nfc] FWT Timeout
39268 [D][Nfc] FWT Timeout
39304 [D][Nfc] FWT Timeout
39356 [D][Nfc] FWT Timeout
39384 [D][Nfc] FWT Timeout
39407 [D][Nfc] FWT Timeout
39432 [D][Nfc] FWT Timeout
39457 [D][Nfc] FWT Timeout
39509 [D][Nfc] FWT Timeout
39537 [D][Nfc] FWT Timeout
39596 [D][Nfc] FWT Timeout
39621 [D][Nfc] FWT Timeout
39673 [D][Nfc] FWT Timeout
39701 [D][Nfc] FWT Timeout
39716 [D][NfcScanner] Found 4 children
39762 [D][Nfc] FWT Timeout
39764 [D][Nfc] FWT Timeout
39798 [D][Nfc] FWT Timeout
39831 [D][Iso14443_4aPoller] Read ATS success
39833 [I][MfDesfirePoller] CMD: 60
39863 [I][NfcScanner] Detected 1 protocols
40051 [I][Elf] Total size of loaded sections: 980
40054 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40107 [I][Elf] Total size of loaded sections: 496
40110 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40166 [I][Elf] Total size of loaded sections: 964
40169 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40227 [I][Elf] Total size of loaded sections: 1172
40230 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40308 [I][Elf] Total size of loaded sections: 1312
40311 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40366 [I][Elf] Total size of loaded sections: 1428
40369 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40425 [I][Elf] Total size of loaded sections: 1904
40428 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40483 [I][Elf] Total size of loaded sections: 1880
40486 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40548 [I][Elf] Total size of loaded sections: 1576
40551 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40602 [I][Elf] Total size of loaded sections: 652
40605 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40660 [I][Elf] Total size of loaded sections: 1100
40663 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
40667 [D][NfcSupportedCards] Loaded 11 plugins
40684 [D][Iso14443_4aPoller] Read ATS success
40686 [I][MfDesfirePoller] CMD: 60
40694 [D][MfDesfirePoller] Read version success
40696 [I][MfDesfirePoller] CMD: 6E
40700 [D][MfDesfirePoller] Read free memory success
40702 [I][MfDesfirePoller] CMD: 45
40706 [E][MfDesfirePoller] Error code: AE
40708 [E][MfDesfire] Key setting incorrect length: 3
00 01 00
40711 [E][MfDesfirePoller] Failed to parse read key settings cmd
40714 [E][MfDesfirePoller] Failed to read master key settings
40718 [D][Nfc] FWT Timeout
40720 [D][Nfc] FWT Timeout
40822 [D][Nfc] FWT Timeout
40924 [D][Nfc] FWT Timeout
41026 [D][Nfc] FWT Timeout
41128 [D][Nfc] FWT Timeout
41230 [D][Nfc] FWT Timeout
41332 [D][Nfc] FWT Timeout
41434 [D][Nfc] FWT Timeout
41536 [D][Nfc] FWT Timeout
41638 [D][Nfc] FWT Timeout
41740 [D][Nfc] FWT Timeout

gornekich, 2024-01-31:
@Tomi2965 thanks a lot for logs. I configured one of my DESFire cards and reproduced the issue. The fix will be ready soon.

philicious, 2024-01-31:
@gornekich great to hear! .. as a very curious by nature person, I am looking forward to the commit and seeing the reason

Tomi2965, 2024-02-16:
@gornekich have you any news? I know/see, that you have a lot of work, I only want a small ongoing info. This issue is not important for me. Thanks a lot!

gornekich, 2024-04-15:
#3576 should fix the issue

Tomi2965, 2024-04-16:
@gornekich many thanks for fix! It works now :) 
[image]

closed 2024-04-16 by gornekich