flipperdevices/flipperzero-firmware #3351: Cloud-Based Badge Synchronization for Entry System

anonymous10download, 2024-01-09:
### Description of the feature you're suggesting.

In response to issue #1345, it would be beneficial to introduce a cloud-based badge synchronization system. Consider the following use case: We have a badge-protected entry system (e.g., Badge Ori), where a counter is incremented on each scan of the NFC tag. Users can create copies (e.g., X1, X2, X3), and it's essential for users to synchronize their copies before and after each entry into a building.

### Sync scenario
Proposed Synchronization Scenario:
Scan the badge and check against a known hash list of previous scans in database:
 -1: hash of the badge is known
       -1.1 and identified as an old version: Download and write the latest version from the cloud.
       -1.2 and identified as the latest version: No action needed; this badge is the most recent one that passed.
  -2: If the hash of the badge is unknown: -> Read the badge and upload it to the cloud, defining it as the latest version.

Cloud services like Google Drive or other shared services can be utilized for this purpose.

Synchronization Scenario Before and After Entering a Building.

### Entering building using X1:
Scenario for entering a building for the first time after copying:
- Pre-scan: Hash of the badge is not known (first scan), upload a copy to the cloud, and save the hash in the known list.
- Portal scan: The badge is desynchronized, and the counter is incremented.
- Post-scan: Hash of the badge is not known, upload a copy to the cloud, and save the hash in the known list.

### Entering Building using any X:
- Pre-scan: Perform the synchronization scenario, usually downloading and writing the latest version.
- Portal scan: Badge is incremented.
- Post-scan: Perform the synchronization scenario, uploading the badge to the cloud.

### System limitations
In the event that a user forgets to skip either the pre-scan or post-scan process even once, it may result in the detection of a copied badge.

skotopes, 2024-01-09:
Hi. I believe you've already answered your question in last sentence.
Impossible to implement in a reliable way. Also too niche to be included into FZ firmware/companions.

closed 2024-01-09 by skotopes

quantum-x, 2024-01-15:
@anonymous10download 
Head's up, this specific mechanism is patented in the AU/EU/US
https://data.epo.org/gpi/EP3797383A4-WIRELESS-ACCESS-TAG-SYSTEM-AND-METHOD