flipperdevices/flipperzero-firmware #3370: SubGHz: Missing FDI/URMET remote (2 channel 433MHz + 125kHz fob, yellow)

Tomi2965, 2024-01-15:
### Describe the enhancement you're suggesting.

Hi. I want to report a currently unsupported remote. Can you add it please? Many thanks!

Manufacturer: FDI Matelec FR
Model nr.: unknown
Channels: 2
Freq.: 433.92MHz
Battery CR1620
Chip: TM02NL (unknown)
Rolling code: no
Tested RAW files: yes, functional
RAW signal code (ch.1): 99356478
Note: inside this RC is also a 125kHz fob, identified as "Viking", rewriteable, same ID as RF. Sucessfully rewrited and tested from Viking to HID. 

RAW files: [2b125.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13941541/2b125.zip)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Skorpionm, 2024-01-18:
So. The protocol is invert usart 8N1 at baunt rate 2000.
sends in portions of unequal length, the messages are not identical to each other, but there are repetitions. Let's figure out the protocol in place and I'll add it.
in the files, a sketch for Arduino is needed to decode the RAW file (you need to output data to an external pin on the flipper and connect it to the D10 arduino)
[image]
[image]

[2b125_decode.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13978015/2b125_decode.txt)

[decode_arduino.ino.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13978069/decode_arduino.ino.txt)

Skorpionm, 2024-01-18:
Have you tried to play back a recorded RAW signal with a single button press recorded? Does the receiver always work on it? It seems to me that the signal is still not all rolling or RAW recorded

Tomi2965, 2024-01-18:
Hi @Skorpionm! Thanks for reply. All four types what I posted was recorded with 10-11x button holding down, until RF stop blinking. Sometimes when I was much fast, RC sends a 1-1.5 second siagnal. Others are appx 3.5 seconds. 

For now, I can send you only this file. Short single shot, saved as RAW, tested - fully working. It's from 4-button version with code 90033044, button 4.

Sorry for sub file with another code, I am now at work, and previous RC is at home. But, I can test anything at work - I have full access to entry system management. If you need a single shot sub from previous RC, just tell me and I will upload it here. 

[433_btn4_short.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13979874/433_btn4_short.zip)

Skorpionm, 2024-01-18:
Ideally, it would be if you connected an Arduino to the remote control.  with my sketch.  to record 100+ clicks on 1 button without errors and personal noise from the air.  and try to figure out the algorithm.  and ideally it would be to have such a remote control.  but if raw works, what's the problem?  all remote controls have the same type of encoding, but the 868 also adds a Manchester package, and it also changes when you hold the button

Tomi2965, 2024-01-18:
For now, me and my colleague don't have a Arduino. But I can ask my friends if needed.

If you need, I can capture all buttons again as RAW in concrete room four floors underground, to avoid interferences. 

"if raw works, what's the problem?" - there is no problem of course. I added these issues/enhancement request, because FZ don't know these RC's. I was ask official support for best way, how to report unknown RC's/RFID's/... They tell me, to report it here.

Sadly, this type of RC (2 button) is not oficially manufactured anymore :( but maybe eBay?

How I can help you to research for now, without Arduino?

Skorpionm, 2024-01-19:
protocol scrolling, encoded by sets of bytes (0xFF, 0x5F, 0xF7, 0x77, 0x75, 0x75, 0xFD, 0xD5, 0xD7, 0xDD, 0x5D) or (0xF, 0xD, 0x7, 0x5). and only such patterns are found in different consoles. How can I help you? figure out how the data is encoded

Skorpionm, 2024-04-16:
I can’t help but encrypt by records