flipperdevices/flipperzero-firmware #3371: SubGHz: Missing FDI/URMET remote (4 channel 433MHz + 125kHz fob, yellow)

Tomi2965, 2024-01-15:
### Describe the enhancement you're suggesting.

Hi. I want to report a currently unsupported remote. Can you add it please? Many thanks!

Manufacturer: FDI Matelec FR
Model nr.: FD-010-169
Channels: 4
Freq.: 433.92MHz
Battery CR2032
Chip: 10268A
Rolling code: no
Tested RAW files: yes, functional
RAW signal code (ch.1): 90339320
Note 1: number in circle on case is year of production
Note 2: colour of buttons identify this exact type (yellow = 433MHz + 125kHz fob)
Note 3: inside this RC is also a 125kHz fob, identified as "Viking", rewriteable, same ID as RF. Sucessfully rewrited and tested from Viking to HID. 

RAW files: [4b125.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13941598/4b125.zip)

[image]
[image]
[image]
[image]
[image]

Skorpionm, 2024-01-18:
https://github.com/flipperdevices/flipperzero-firmware/issues/3370#issuecomment-1898627391

[4b125_key1.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13978472/4b125_key1.txt)
[4b125_key2.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13978473/4b125_key2.txt)
[4b125_key3.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13978474/4b125_key3.txt)
[4b125_key4.txt](https://github.com/flipperdevices/flipperzero-firmware/files/13978475/4b125_key4.txt)

Skorpionm, 2024-04-16:
I can’t help but encrypt by records

closed 2024-04-16 by Skorpionm