flipperdevices/flipperzero-firmware #3373: SubGHz: Missing FDI/URMET remote (4 channel 868MHz + Mifare fob, blue)

Tomi2965, 2024-01-15:
### Describe the enhancement you're suggesting.

Hi. I want to report a currently unsupported remote. Can you add it please? Many thanks!

Manufacturer: FDI Matelec FR
Model nr.: FD-010-172
Channels: 4
Freq.: 868.35MHz
Battery CR2032
Chip: 10268A
Rolling code: probably yes
Tested RAW files: yes, entry system says "Access denies (rolling code)" - sometimes show code, sometimes not
RAW signal code (ch.1): 0A1734EA
Note 1: number in circle on case is year of production
Note 2: colour of buttons identify this exact type (blue= 868MHz + Mifare fob)
Note 3: inside this RC is also a Mifare Classic 1k fob, with same ID as RF. 

RAW files: 
[4b868.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13942159/4b868.zip)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Skorpionm, 2024-04-16:
I can’t help but encrypt by records

closed 2024-04-16 by Skorpionm

franciscomfcmaia, 2025-03-28:
Did you ever figure out how to clone the RF on this? 🤔