flipperdevices/flipperzero-firmware #3375: Emulation destablizes NFC read

dogtopus, 2024-01-16:
### Describe the bug.

NFC read becomes very unstable (frequently showing tag lost, freezing during reads, failing to detect tag type, etc.) after tag emulation or Detect Reader is used.

### Reproduction

1. Do a fresh reboot
2. Emulate a tag or trigger Detect Reader once (with or without actually using the emulated tag)
3. Use the read function on any tags that have a supported high level protocol (NTAG21x or Mifare Classic)

### Target

f7

### Logs

```Text
# Below are the logs after Flipper failed to detect a Mifare Classic tag
306384 [D][NfcScanner] Found 5 base protocols                                   
306392 [D][DolphinState] icounter 3630, butthurt 0                              
306444 [D][Nfc] FWT Timeout                                                     
306469 [D][Nfc] FWT Timeout                                                     
306521 [D][Nfc] FWT Timeout                                                     
306549 [D][Nfc] FWT Timeout                                                     
306564 [D][NfcScanner] Found 4 children                                         
306577 [D][Nfc] Failed in poller RX                                             
306580 [E][ISO14443_3A] Sel request failed: 1                                   
306715 [E][ISO14443_3A] Sel request failed: 6                                   
306849 [E][ISO14443_3A] Sdd response wrong length                               
306977 [E][Iso14443_4aPoller] ATS request failed                                
306979 [D][Iso14443_4aPoller] Failed to read ATS                                
306983 [D][Nfc] FWT Timeout                                                     
307005 [I][NfcScanner] Detected 1 protocols                                     
307367 [D][DolphinState] icounter 3630, butthurt 0
```

skotopes, 2024-01-23:
@doomwastaken can you retest it?

doomwastaken, 2024-01-30:
Greetings, do you have any more details for us? 
I've tried to reproduce it on latest RC, did following steps:

1. Reboot the device
2. Do a full detect reader
3. Do a read of MFC card / NTAG

And repeat same with partial detect reader/emulation on step 2

dogtopus, 2024-01-30:
I can no longer reproduce this on 0.98.2 RC so I assume this has been fixed.

dogtopus, 2024-02-07:
Spoke too soon. Seems like something changed from RC to release and the problem came back...

doomwastaken, 2024-02-07:
Try on latest dev, one of nfc fixes was overwritten and re-introduced

dogtopus, 2024-02-07:
@doomwastaken No I didn't change log level. It's reproducible on default log level for me.

I'll retest on dev and let you know.

dogtopus, 2024-02-10:
@doomwastaken I'm on bcdb9cb and I can still reproduce it.

dogtopus, 2024-02-10:
I also rolled back to 0.98.{0,1,2}-rc and I can't seem to find the version where it was fixed. This suggesting that either it may be intermittent or something went wrong when I first tested it.

dogtopus, 2024-02-10:
OK I was able to reproduce the "fix" on 0.98.2 release. It might be a hardware defect/hardware issue triggered by software since after I charge the battery to 4.1V the problem disappeared.

It may be possible to reproduce this using a bench power supply. I don't have one so I can't test further easily.

skotopes, 2024-03-05:
@dogtopus sounds as strange as it can be. NFC chip got its own power supply that provides NFC frontend with 5V.

You may want to check NFC antenna connection pads: make sure that there is no water damage. Also feel free to ping me on discord if you want to

dogtopus, 2024-03-21:
@skotopes I cleaned up the NFC antenna contacts and it didn't seem to change anything. The contacts also don't seem damaged or corroded.

Though it still feels like a hardware issue.

Unfortunately I'm not on Flipper Discord (I don't like the phone verification there).

skotopes, 2024-07-07:
@dogtopus sorry for late reply, for hardware issues please ping support: https://support.flipper.net/hc/en-us

closed 2024-07-07 by skotopes