flipperdevices/flipperzero-firmware #3385: NFC: regression: Detect Reader seems to be broken

philicious, 2024-01-18:
### Describe the bug.

I am on FW `0.97.1` and tried to get the nonces from a Reader via `[Saved Tag] -> Detect Reader`
Flipper is successfully emulating the Tag and the Reader even accepts it. However the initial "Touch the reader" screen doesnt show any collected nonces or ask me to "touch reader again" as usual.

However, on one of the readers I performed the attack successfully couple weeks ago when I was on an earlier FW. Probably `0.95.0` or `0.94.1`. Hence I suspect a regression

### Reproduction

Try to sniff nonces from a reader

skotopes, 2024-01-20:
@doomwastaken can you reproduce it with our readers?

doomwastaken, 2024-01-20:
It worked fine with most our readers on autotests,  will check on Monday with all of them.

philicious, 2024-01-20:
Hm. Thanks for investigating. So doesn't sound like a regression then. 

What else could be the reason?

I have one sniffed key from there already added to user dict. 
But that can't be a reason to my understanding, right?

nekolab, 2024-01-22:
Encountered the same issue too, but I found if I attached a CLI and ran `log debug` (must be `log debug` but not `log`), the reader starts to react to the emulation. If I exit the `log debug`  (just press Ctrl-C), the emulation fails again. Hope this can help

philicious, 2024-01-28:
Would you have any idea what could cause this issue or how it could be swarmed?

My only idea so far is to downgrade flipper to one of the versions that I probably was on when it was working last time and try again. 

But I'd rather get it working again with latest versions.

philicious, 2024-01-31:
@olg857 thats a different issue then described in here. pls create a separate issue

philicious, 2024-02-02:
seems to be fixed with `0.98.2`

closed 2024-02-02 by philicious