flipperdevices/flipperzero-firmware #3387: NFC Mifare Emulate doesn't work after update to 0.97.1

vincent-bidaux, 2024-01-23:
### Describe the bug.

I'm using my Flipper Zero for 18 months and successfully emulate a Mifare NFC that I had read using the nonces capture method. I updated on Jan 22 2024 from the iOS app, and emulating then produced zero response from the reader. Updated again on Jan 23 2024 (today) to 0.97.1 and same symptoms.

Usually this reader produces beeps, when emulation is successful, and even more beeps when it's not. And now, there's nothing, as if the reader didn't get anything from Flipper.

- I checked that an official badge still works on the reader: it works
- I made a new read of that badges and got a complete read, 32/32 and 16/16
- I ordered a magic card to test but haven't received it yet (will update the ticket).

I'm attaching a picture of the reader.

[image]

Tell me if you need any more details that can be helpful to tackle this bug.

### Reproduction

Load the save
Place Zero on reader
Emulate
...
nothing happens

### Target

0.97.1 and the one before I guess

philicious, 2024-02-02:
for what its worth, I successfully emulated old saved dumps with `0.98.2` 

@vincent-bidaux could you retry with latest firmware? there were some issues around NFC with last firmwares, so maybe its fixed by now

skotopes, 2024-03-05:
@vincent-bidaux any updates?

gornekich, 2024-04-15:
Please, reopen if the issue persists

closed 2024-04-15 by gornekich