flipperdevices/flipperzero-firmware #3389: Infrared Beamer not working/ Only in diffrent ways

Killer74-hub, 2024-01-23:
### Describe the bug.

I have a problem. So I recorded a signal of a remote. When I wanted to send it, nothing happened. In the firmware of flipper zero the signal is included, But it needs to be pressed 2 times. Under: "Projectors" and the on and off signal. When you want to turn it off , you first need to wait for like: 47 percent or 50 percent. Then exit, and need to do it again for the "Ok turn off" signal. It is on 27 percent or so. The projector is a EPSON Projector. Maybe fix it, so that it will send first the turn off signal, and then the ok signal 

(i wrote this in discord in the IR channel)

### Reproduction

I wrote this also in the describe the Bug section.

### Target

EPSON Projector

hedger, 2024-01-23:
If you want to control a specific device, just create a custom IR remote only with the signals you need. Universal remote is universal for a reason.

closed 2024-01-23 by hedger

Killer74-hub, 2024-01-23:
> If you want to control a specific device, just create a custom IR remote only with the signals you need. Universal remote is universal for a reason.

I can’t. That’s the problem. Could Zhou maybe just fix this and not ignore? Would be really nice

Killer74-hub, 2024-01-23:
> If you want to control a specific device, just create a custom IR remote only with the signals you need. Universal remote is universal for a reason.

For that device it doesn’t work