flipperdevices/flipperzero-firmware #3400: Better handling of NFC 14 b

rct31, 2024-01-28:
### Description of the feature you're suggesting.

Hi,

Is it possible to have a better handling of the NFC 14B norm, (shown as NFC-B on flipper ? ).

This is used in public transportation ticket in France. For the moment the flipper is only able to read the UID of the ticket, not the content of it. 

It's a different type of NFC with 2 block defined as counter. There is some fun to have around it. 

Let me show you the work of GentilKiwi on this topic : 
[ST25TB_transport.pdf](https://github.com/flipperdevices/flipperzero-firmware/files/14077290/ST25TB_transport.pdf)

An implentation this feature could make a huge difference her in France. 

Tell me if I can help by any mean,

Thanks for reading,

mxcdoam, 2024-01-28:
Hi, my city used to utilize same chips in it's transportation tickets, as in mentioned PDF. Flipper has no issues with reading memory contents, counters and otp on 0.97.1 FW. Your card may be utilizing some kind of Calypso chip? They are are usually iso14443B too, and the exact chip name may be difficult to find.

rct31, 2024-01-28:
Oh, my bad ! I wasn't on the last official firmware ! (I was on an unofficial). 

With 0.97.1, I can, like you, read everything, which is a big progress. 

So sad emulation of all the card is still unavailable.

skotopes, 2024-01-29:
NFC-B is not supported on hardware level. Software emulation is not possible because it requires much faster CPU than we have.

closed 2024-01-29 by skotopes

mxcdoam, 2024-01-30:
Will writing be supported at least for SRI512/SRT512?