flipperdevices/flipperzero-firmware #3413: F0 freezes after turning off Bluetooth from the mobile app

Hari-B6, 2024-01-31:
### Describe the bug.

Flipper zero freezes after turning off Bluetooth from the Remote Control in the Flipper mobile app. It resumes normal operation after 5~10 seconds.

Reproduction:
https://github.com/flipperdevices/flipperzero-firmware/assets/149514426/d48f27f4-6026-45a2-b648-d834dbf23545

Logs:
[16:05:20 31-01-2024.txt](https://github.com/flipperdevices/flipperzero-firmware/files/14112856/16.05.20.31-01-2024.txt.txt)

### Reproduction

1. Open the Flipper mobile app
2. Navigate to the **Hub** tab
3. Open Remote control
4. Open Settings>Bluetooth
5. Turn off Bluetooth _using the app_

### Logs

```Text
*attached in the description*
```

skotopes, 2024-03-05:
@Hari-B6 try latest RC or Dev

Hari-B6, 2024-03-05:
I can confirm the issue has been fixed in Firmware 0.99.0 RC1.

skotopes, 2024-03-05:
Awesome. I'm closing this issue.

closed 2024-03-05 by skotopes

Hari-B6, 2024-03-05:
Thank you, have a nice day.