flipperdevices/flipperzero-firmware #3419: Emulation of ISO 15693 tags crashes Flipper

mxcdoam, 2024-02-01:
### Describe the bug.

I've tried to emulate different types of ISO 15693 tags on 0.98.2 FW and each attempt resulted with a crash. Screen says: "Flipper crashed and was rebooted. NULL pointer dereference"
Tags I checked:
1) NXP ICODE SLI (SL2ICS20)
2) NXP ICODE SLIX (SL2S2002)
3) NXP ICODE SLIX2 (SL2S2602)
4) TI Tag-it HF-I Plus (TMS37112)
All tags are not password-protected, can be read & saved. 
Reader devices are:
1) an Android phone with NXP TagInfo running
2) HID Cardman 5321 (PSCS-compatible reader).

Sometimes flipper crashes right after being put into reader's field, sometimes - it starts communication and even can emulate tag type, size etc, but never the memory contents (even if it's just zeros). 

### Reproduction

1) Read any (SLI, SLIX, SLIX, TagIT HF-I) tag on Flipper zero
2) Open menu and choose "Emulate"
2) Put Flipper Zero into reader's field

### Target

7

mxcdoam, 2024-02-01:
I've tried different NFC Android app, and everything seems to work fine. So it's probably NXP TagInfo/Android issue, not Flipper Zero. Closing issue as I can't reproduce it anymore.

closed 2024-02-01 by mxcdoam

rygerek, 2024-12-06:
It's been a while, but if you haven't found a solution yet, try turning off debugging mode. It worked for me and the flipper doesn't hang.

TriplEight, 2025-01-26:
I'm facing the same issue on the 1.20 firmware. Trying to emulate SLIX2 cards.

"Flipper crashed and was rebooted." But no details, no even null pointer ref.