flipperdevices/flipperzero-firmware #3420: Card type recognition issue

mxcdoam, 2024-02-01:
### Describe the enhancement you're suggesting.

This issue affects Plaintain parser plugin. Upon reading the card flipper zero recognizes Mifare Plus EV1 card as a multiprotocol card (Mifare Classic + DESfire) and asks to choose how it should handle card. After selecting "Read as Mifare Classic" card is read and parsed. If "Read as MIFARE DESFire" selected, flipper will attempt to read card indefinitly.
Previous generations of Plaintains were issued with MIFARE Plus X chip, they are automatically recognized, read, and parsed.
Again, this issue does not prevent parser from working, just a weird inconvinience

gornekich, 2024-03-07:
Thanks for the report. #3504 should fix the issue.

mxcdoam, 2024-03-13:
On 0.99.1-rc this issue still appears while reading Mifare plus EV1 plantain. NFC app suggests two ways to read it: as MFC and DESfire. Also, now Mifare plus X plantains have issues too: while reading MPX plantain the "don't move" screen appears and stays for about 10-15 seconds (which is already weird, because used keys are known and read should be fast like in previous fw versions), after that NFC app opens dictionary attack screen and starts checking user/system dicts! Parsed data will be shown at the end. This feature was useful to quickly check balance, and now it's just takes too much time.

gornekich, 2024-04-15:
@mxcdoam this fix came in release 0.100.3. Could you please try again?

mxcdoam, 2024-04-16:
EV1 recognition seems to be fixed on 0.100.3. Thanks!