flipperdevices/flipperzero-firmware #3422: NFC: regression in reading MFClassic since 0.98.0

philicious, 2024-02-02:
### Describe the bug.

Today, on `0.98.2` I re-read a MFClassic 1K Tag and to my surprise it only found 14/32 keys. However it used to be 30/32 !
I also noticed the screen is not showing the progress bar where it tries all keys of user-dict and system-dict. It only shows "Dont move" for a little longer than usual.

Comparing the new dump with my old, shows that it doesnt even have KeyA for alot sectors, but the key is default key `FF FF FF FF FF FF`
Furthermore, one section, which I have the sniffed key in user-dict, is also missing.

Now I went back Firmware versions and re-tried:
- 0.98.2  bad
- 0.98.1-rc bad
- 0.98.0-rc bad
- 0.97.1 good, gets all 30/32 keys again

I wonder if it could be related to https://github.com/flipperdevices/flipperzero-firmware/pull/3356 which got introduced in 0.98.0-rc and mangles 

### Reproduction

see description

philicious, 2024-02-02:
cc @gornekich : tell me what logs you'd like to have *G

gornekich, 2024-02-04:
@philicious please attach here logs with debug log level set

philicious, 2024-02-04:
so, with the old `0.97.1` FW I also see these lines of loading dictionary. this line is missing in the latest FW log

```
342599 [32m[I][KeysDict] [0mLoaded dictionary with 2095 keys
342626 [34m[D][Nfc] [0mFWT Timeout
342738 [34m[D][MfClassicPoller] [0m1K detected
```

gornekich, 2024-02-04:
@philicious it looks like recently added NFC plugin partially read your card and thinks that it read enough. This behavior will be changed in next release. Thanks a lot for providing logs!

gornekich, 2024-02-04:
To be clear how MFC read works:

1. Flipper try to load cache keys and read card with previously saved keys.
2. Flipper load plugins and try to read supported card. If plugins function read() returns true - it means that no further reading is necessary. Otherwise we go to the next step.
3. Flipper tries to find keys from User and System dictionaries.

philicious, 2024-02-04:
alright 👍  .. if there is any branch I can test with and verify, let me know :)

philicious, 2024-02-17:
@gornekich its been a while with no update, so I checked the source of the problem and understand it myself now in [`washcity_verify` ](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/main/nfc/plugins/supported_cards/washcity.c#L56C13-L56C28)

I have two thoughts:
- the washcity plugin probably breaks quite alot MFC-reading as it only checks if reading sector 0 and block 0 works with key `0xA0A1A2A3A4A5`. unfortunately thats one of the very commonly used keys. I'll try doing a PR

- running the plugins before using the dicts and allowing the verify and parse functions to prevent the dicts being used at all  is prone for more trouble in the future. especially if the verify func does is checking single key only, that might very well happen to work on alot cards. especially if its a common key.
then the parse func might also coincidently return true and prevent failing of the plugin.

as preventing this by plugin-PR-reviews might not be failsafe either, a long-term solution might be to only optionally run plugins by asking the user if he wants to try dicts or plugins.

a good measure would also be to at least do checks with at least two different keys on different sectors and avoid using commonly used keys.

PR #3467 (2024-03-05): NFC: Fix washcity plugin verify function being to greedy
washcity plugin verify func only checks authing a single block with a single key. and that key being a very commonly used one. thus potentially breaking reading of alot MFC cards by dicts

**I have built this with `./fbt && ./fbt flash_usb_full` and verified it working**

fixes #3422 
obsoletes #3457 with proper code

# What's new

- use two blocks and keys for verification
- avoid using a commonly used key at all

# Verification 

- read a MFC with 0xA0A1A2A3A4A5 A-key for sector0 thats not a WashCity card: should now use user/system di [... cut]
closed 2024-03-05 by skotopes