flipperdevices/flipperzero-firmware #3432: Saving NFC key crashes Flipper Zero

RoBoneHead22, 2024-02-08:
### Describe the bug.

I am attempting to save a Mifare DESFire NFC key fob and it crashes my flipper. After the reboot the key cannot be emulated
I have attached Screenshots from the Flipper iOS app as well as the debug log.
[image]
[image]

### Reproduction

Main Menu --> NFC --> Read --> Save

### Logs

```Text
_.-------.._                    -,
          .-""--..,,_/ /`-,               -,  \ 
       .:"          /:/  /'\  \     ,_...,  `. |  |
      /       ,----/:/  /`\ _\~`_-"`     _;
     '      / /`"""'\ \ \.~`_-'      ,-"'/ 
    |      | |  0    | | .-'      ,/`  /
   |    ,..\ \     ,.-"`       ,/`    /
  ;    :    `/`""\`           ,/--==,/-----,
  |    `-...|        -.___-Z:_______J...---;
  :         `                           _-'
 _L_  _     ___  ___  ___  ___  ____--"`___  _     ___
| __|| |   |_ _|| _ \| _ \| __|| _ \   / __|| |   |_ _|
| _| | |__  | | |  _/|  _/| _| |   /  | (__ | |__  | |
|_|  |____||___||_|  |_|  |___||_|_\   \___||____||___|

Welcome to Flipper Zero Command Line Interface!
Read the manual: https://docs.flipper.net/development/cli
Run `help` or `?` to list available commands

Firmware version: 0.98.2 0.98.2 (9744fd8f built on 01-02-2024)

>: log debug
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
214180 [32m[I][Loader] [0mLoading /ext/apps/NFC/nfc.fap
214313 [32m[I][Elf] [0mTotal size of loaded sections: 55725
214316 [32m[I][Loader] [0mLoaded in 136ms
214320 [32m[I][AnimationManager] [0mUnload animation 'L1_Cry_128x64'
216134 [34m[D][NfcScanner] [0mFound 5 base protocols
216140 [34m[D][DolphinState] [0micounter 183, butthurt 11
216154 [34m[D][Nfc] [0mFWT Timeout
216191 [34m[D][Nfc] [0mFWT Timeout
216216 [34m[D][Nfc] [0mFWT Timeout
216268 [34m[D][Nfc] [0mFWT Timeout
216296 [34m[D][Nfc] [0mFWT Timeout
216319 [34m[D][Nfc] [0mFWT Timeout
216344 [34m[D][Nfc] [0mFWT Timeout
216369 [34m[D][Nfc] [0mFWT Timeout
216421 [34m[D][Nfc] [0mFWT Timeout
216449 [34m[D][Nfc] [0mFWT Timeout
216472 [34m[D][Nfc] [0mFWT Timeout
216497 [34m[D][Nfc] [0mFWT Timeout
216533 [34m[D][Nfc] [0mFWT Timeout
216585 [34m[D][Nfc] [0mFWT Timeout
216613 [34m[D][Nfc] [0mFWT Timeout
216636 [34m[D][Nfc] [0mFWT Timeout
216661 [34m[D][Nfc] [0mFWT Timeout
216686 [34m[D][Nfc] [0mFWT Timeout
216738 [34m[D][Nfc] [0mFWT Timeout
216766 [34m[D][Nfc] [0mFWT Timeout
216826 [34m[D][Nfc] [0mFWT Timeout
216862 [34m[D][Nfc] [0mFWT Timeout
216914 [34m[D][Nfc] [0mFWT Timeout
216942 [34m[D][Nfc] [0mFWT Timeout
216957 [34m[D][NfcScanner] [0mFound 4 children
217003 [34m[D][Nfc] [0mFWT Timeout
217005 [34m[D][Nfc] [0mFWT Timeout
217039 [34m[D][Nfc] [0mFWT Timeout
217072 [34m[D][Iso14443_4aPoller] [0mRead ATS success
217104 [32m[I][NfcScanner] [0mDetected 1 protocols
217266 [32m[I][Elf] [0mTotal size of loaded sections: 888
217269 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217308 [32m[I][Elf] [0mTotal size of loaded sections: 420
217311 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217348 [32m[I][Elf] [0mTotal size of loaded sections: 836
217351 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217391 [32m[I][Elf] [0mTotal size of loaded sections: 924
217394 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217442 [32m[I][Elf] [0mTotal size of loaded sections: 1244
217445 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217487 [32m[I][Elf] [0mTotal size of loaded sections: 1324
217490 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217536 [32m[I][Elf] [0mTotal size of loaded sections: 1724
217539 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217581 [32m[I][Elf] [0mTotal size of loaded sections: 1768
217584 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217626 [32m[I][Elf] [0mTotal size of loaded sections: 1464
217629 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217669 [32m[I][Elf] [0mTotal size of loaded sections: 636
217672 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217714 [32m[I][Elf] [0mTotal size of loaded sections: 1028
217717 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
217721 [34m[D][NfcSupportedCards] [0mLoaded 11 plugins
217738 [34m[D][Iso14443_4aPoller] [0mRead ATS success
217748 [34m[D][MfDesfirePoller] [0mRead version success
217753 [34m[D][MfDesfirePoller] [0mRead free memory success
217757 [34m[D][MfDesfirePoller] [0mRead master key settings success
217762 [34m[D][MfDesfirePoller] [0mRead master key version success
217769 [34m[D][MfDesfirePoller] [0mRead application ids success
217916 [34m[D][MfDesfirePoller] [0mRead applications success
217918 [34m[D][MfDesfirePoller] [0mRead success.
217921 [34m[D][Nfc] [0mFWT Timeout
217992 [32m[I][Elf] [0mTotal size of loaded sections: 1244
217995 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
218036 [32m[I][Elf] [0mTotal size of loaded sections: 1324
218039 [34m[D][Fap] [0mLibrary for NfcSupportedCardPlugin, API v. 1 loaded
218155 [34m[D][DolphinState] [0micounter 183, butthurt 11
225908 [34m[D][GattChar] [0mUpdating Battery Level char
```

skotopes, 2024-02-12:
@RoBoneHead22 can you try dev? we have couple desfire fixes in it.

ArkBrj, 2024-02-21:
Got the same problem with the same type of NFC.
Per suggestion, installed the latest dev firmware and tried again.
The situation is worse now. When I press Save, Flipper displays an hourglass and hangs in this state forever.
The log is attached.
[cli-dump.txt](https://github.com/flipperdevices/flipperzero-firmware/files/14352446/cli-dump.txt)

RoBoneHead22, 2024-02-29:
@skotopes Hello, I tried release, RC, and dev Channels. Same result. Currently updated to Release Channel 0.98.3 and same issue

ArkBrj, 2024-03-12:
@skotopes 
Tried 0.99.1-RC and dev 022fccf0. Same as before. RC build crashes with null pointer dereference, the dev build hangs forever.

RoBoneHead22, 2024-03-29:
I am starting to believe that this issue is deeper than just the firmware. I even tried the XFW firmware and had the same results. Reverted back to the release firmware now, still no change.

skotopes, 2024-03-29:
@RoBoneHead22 @ArkBrj we need someone to provide us backtrace. Anyone of you have debugger(wifi board/st-link/j-link/etc)?

ArkBrj, 2024-03-29:
@RoBoneHead22
I have a cheap ST-Link clone bought on AliExpress few years back. I also have Flipper's wi-fi board, but I have not tried it yet and have no idea what software to use with it. I can try to collect the info you need, but I need detailed instructions how to do this. I would prefer to use Windows software on the host if it is an option.

ArkBrj, 2024-04-01:
@skotopes Just realized that I tagged a wrong person in my previous post. Correcting the mistake....
> @RoBoneHead22 I have a cheap ST-Link clone bought on AliExpress few years back. I also have Flipper's wi-fi board, but I have not tried it yet and have no idea what software to use with it. I can [...]

gornekich, 2024-04-15:
#3576 should fix the issue

gornekich, 2024-04-16:
Please try latest dev and reopen if issue persists.

closed 2024-04-16 by gornekich

ArkBrj, 2024-04-17:
@gornekich
I confirm that flipper does not crash anymore. Thanks for the fix!
However, there might be some other problem with such cards - when I replay the recording, the door does not open.
I recorded the same NFC twice and both recording do not work as the original NFC. Compared them - they are identical, so there might be a problem with how flipper emulates (or how it records for that matter).
I have a friend who also has a flipper, I will try to use that one to record what mine produces and compare them.

gornekich, 2024-04-17:
Hello @ArkBrj 
Flipper doesn'r support full Mifare DESFire emulation. When you emulate your card, you can see only "Emulate UID" option. That's why when you read emulation with the other flipper, it shows ISO14443-4A data.

Basically, Mifare DESFire is very secure protocol. If you don't know keys from your card, it's almost impossible to find them. And without these keys you can't emulate your card

ArkBrj, 2024-04-17:
@gornekich 
Thank you very much for the detailed explanation. I missed this part in the doc.