flipperdevices/flipperzero-firmware #3433: Duckyscript preprocessor?

mantacid, 2024-02-08:
### Description of the feature you're suggesting.

After looking over the duckyscript 1.0 spec linked to in your docs, I wanted to write a payload that can modify itself based on the target OS. This is unfortunately infeasible due to inconsistencies in how operating systems send CAPSLOCK and NUMLOCK states to the keyboard, meaning I can’t get information from the target system back to the flipper.

As I was looking for a solution, I realized that the flipper zero is more than just a USB: it has a display, buttons, Bluetooth, and heaps more features than a typical USB RubberDucky. So I had the idea of a duckyscript preprocessor that could leverage the capabilities of the flipper zero to their full potential.

Payloads would be more dynamic, able to utilize the display and buttons of the flipper zero to fine-tune the payload on-the-fly. You’d need less infrastructure to set up the attack, as the payload can simply send data over Bluetooth for exfiltration, or maybe even to a web server on a GPIO board. Users can define payloads separately from the deployment system, such that different ones can be used for different architectures, and deployed in tandem with the scripts they need to set up the system for exploitation (example: a payload that requires root privileges can be called by a script that _gets_ those privileges).

With the flipper’s status as a do-it-all multitool, this kind of addition just makes sense: why restrict your all-in-one device to only emulating the functionality of one tool at any given moment, when you can utilize every sensor and protocol at your disposal to empower the penTester wielding it?

skotopes, 2024-02-12:
we are not planning to continue ducky script support, next release will include Javascript for that task

closed 2024-02-12 by skotopes

mantacid, 2024-02-16:
Oh gosh no please anything but JavaScript.