flipperdevices/flipperzero-firmware #3436: Flipper Zero Crashes when I try to use it. (Reading NFC, Saving IR Remote Sequences)

I-Am-Skoot, 2024-02-09:
### Describe the bug.

I had to replace a lost device.   Since then the new device has been consistently crashing.  This has been over the last 3 firmware versions, so I am inclined to believe it is a hardware issue, however the support team insisted I open a ticket here.

### Reproduction

1. Switch on the device
2.1. Open the NFC app
2.1 Read Card
2.2 Place flipper against card
2.3 Crash
2.2 Open cross remote app
2.3 Create a new sequence
2.4 start adding entries
2.5 Crash

### Logs

```Text
_.-------.._                    -,
          .-""--..,,_/ /`-,               -,  \
       .:"          /:/  /'\  \     ,_...,  `. |  |
      /       ,----/:/  /`\ _\~`_-"`     _;
     '      / /`"""'\ \ \.~`_-'      ,-"'/
    |      | |  0    | | .-'      ,/`  /
   |    ,..\ \     ,.-"`       ,/`    /
  ;    :    `/`""\`           ,/--==,/-----,
  |    `-...|        -.___-Z:_______J...---;
  :         `                           _-'
 _L_  _     ___  ___  ___  ___  ____--"`___  _     ___
| __|| |   |_ _|| _ \| _ \| __|| _ \   / __|| |   |_ _|
| _| | |__  | | |  _/|  _/| _| |   /  | (__ | |__  | |
|_|  |____||___||_|  |_|  |___||_|_\   \___||____||___|

Welcome to Flipper Zero Command Line Interface!
Read Manual https://docs.flipperzero.one

Firmware version: 0.96.1 0.96.1 (5dbe1cf0 built on 08-12-2023)

>: log debug
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
340848 [I][Loader] Loading /ext/apps/NFC/nfc.fap
340980 [I][Elf] Total size of loaded sections: 53013
340982 [I][Loader] Loaded in 134ms
340985 [I][AnimationManager] Unload animation 'L1_Waves_128x50'
342355 [D][NfcScanner] Found 5 base protocols
342363 [D][DolphinState] icounter 1, butthurt 0
342377 [D][Nfc] FWT Timeout
342404 [D][Nfc] FWT Timeout
342429 [D][Nfc] FWT Timeout
342481 [D][Nfc] FWT Timeout
342509 [D][Nfc] FWT Timeout
342532 [D][Nfc] FWT Timeout
342579 [D][Nfc] FWT Timeout
342604 [D][Nfc] FWT Timeout
342667 [D][Nfc] FWT Timeout
342695 [D][Nfc] FWT Timeout
342718 [D][Nfc] FWT Timeout
342743 [D][Nfc] FWT Timeout
342768 [D][Nfc] FWT Timeout
342820 [D][Nfc] FWT Timeout
342848 [D][Nfc] FWT Timeout
342871 [D][Nfc] FWT Timeout
342907 [D][Nfc] FWT Timeout
342932 [D][Nfc] FWT Timeout
342984 [D][Nfc] FWT Timeout
343012 [D][Nfc] FWT Timeout
343035 [D][Nfc] FWT Timeout
343060 [D][Nfc] FWT Timeout
343085 [D][Nfc] FWT Timeout
343137 [D][Nfc] FWT Timeout
343165 [D][Nfc] FWT Timeout
343188 [D][Nfc] FWT Timeout
343213 [D][Nfc] FWT Timeout
343238 [D][Nfc] FWT Timeout
343291 [D][Nfc] FWT Timeout
343319 [D][Nfc] FWT Timeout
343342 [D][Nfc] FWT Timeout
343367 [D][Nfc] FWT Timeout
343392 [D][Nfc] FWT Timeout
343444 [D][Nfc] FWT Timeout
343472 [D][Nfc] FWT Timeout
343495 [D][Nfc] FWT Timeout
343520 [D][Nfc] FWT Timeout
343545 [D][Nfc] FWT Timeout
343608 [D][Nfc] FWT Timeout
343638 [D][Nfc] FWT Timeout
343663 [D][Nfc] FWT Timeout
343688 [D][Nfc] FWT Timeout
343713 [D][Nfc] FWT Timeout
343765 [D][Nfc] FWT Timeout
343793 [D][Nfc] FWT Timeout
343816 [D][Nfc] FWT Timeout
343841 [D][Nfc] FWT Timeout
343866 [D][Nfc] FWT Timeout
343940 [D][Nfc] FWT Timeout
343968 [D][Nfc] FWT Timeout
343991 [D][Nfc] FWT Timeout
344016 [D][Nfc] FWT Timeout
344041 [D][Nfc] FWT Timeout
344093 [D][Nfc] FWT Timeout
344121 [D][Nfc] FWT Timeout
344180 [D][Nfc] FWT Timeout
344206 [D][Nfc] FWT Timeout
344269 [D][Nfc] FWT Timeout
344297 [D][Nfc] FWT Timeout
344312 [D][NfcScanner] Found 4 children
344369 [D][Nfc] FWT Timeout
344371 [D][Nfc] FWT Timeout
344405 [D][Nfc] FWT Timeout
344438 [D][Iso14443_4aPoller] Read ATS success
344470 [I][NfcScanner] Detected 1 protocols
344602 [D][Iso14443_4aPoller] Read ATS success
344612 [D][MfDesfirePoller] Read version success
344616 [D][MfDesfirePoller] Read free memory success
344620 [D][MfDesfirePoller] Read master key settings success
344625 [D][MfDesfirePoller] Read master key version success
344630 [D][MfDesfirePoller] Read application ids success
```

### Anything else?

I made a video of the issues

[image]

hedger, 2024-02-09:
Reboots with furi_check are 99.9% software issues.
Please try installing latest dev firmware and see if the issues is still reproducible.

I-Am-Skoot, 2024-02-09:
> Reboots with furi_check are 99.9% software issues. Please try installing latest dev firmware and see if the issue is still reproducible.

As stated previously there have been server firmware versions

To date this has been the same on the following firmware versions 
- 0.95.0
- 0.96.1
- 0.97.1
- 0.98.2  (Updated again today and re-tested)

I-Am-Skoot, 2024-02-09:
Serial log

```

              _.-------.._                    -,
          .-"```"--..,,_/ /`-,               -,  \
       .:"          /:/  /'\  \     ,_...,  `. |  |
      /       ,----/:/  /`\ _\~`_-"`     _;
     '      / /`"""'\ \ \.~`_-'      ,-"'/
    |      | |  0    | | .-'      ,/`  /
   |    ,..\ \     ,.-"`       ,/`    /
  ;    :    `/`""\`           ,/--==,/-----,
  |    `-...|        -.___-Z:_______J...---;
  :         `                           _-'
 _L_  _     ___  ___  ___  ___  ____--"`___  _     ___
| __|| |   |_ _|| _ \| _ \| __|| _ \   / __|| |   |_ _|
| _| | |__  | | |  _/|  _/| _| |   /  | (__ | |__  | |
|_|  |____||___||_|  |_|  |___||_|_\   \___||____||___|

Welcome to Flipper Zero Command Line Interface!
Read the manual: https://docs.flipper.net/development/cli
Run `help` or `?` to list available commands

Firmware version: dev unknown (bcdb9cb1 built on 09-02-2024)

>: log trace
Current log level: trace
Use <log ?> to list available log levels
Press CTRL+C to stop...
37861 [I][SavedStruct] Loading "/int/.desktop.settings"
37864 [T][StorageApi] File/Dir 00008750 alloc
37867 [T][StorageInt] Device read: block 0, off 0, buffer: 20007538, size 8, translated address: 080A7000
37872 [T][StorageInt] Device read: block 1, off 0, buffer: 20007538, size 8, translated address: 080A8000
37877 [T][StorageInt] Device read: block 1, off 8, buffer: 20007538, size 16, translated address: 080A8008
37882 [T][StorageInt] Device read: block 1, off 24, buffer: 20007538, size 16, translated address: 080A8018
37887 [T][StorageInt] Device read: block 1, off 40, buffer: 20007538, size 16, translated address: 080A8028
37892 [T][StorageInt] Device read: block 1, off 56, buffer: 20007538, size 16, translated address: 080A8038
37898 [T][StorageInt] Device read: block 1, off 72, buffer: 20007538, size 16, translated address: 080A8048
37903 [T][StorageInt] Device read: block 1, off 88, buffer: 20007538, size 16, translated address: 080A8058
37908 [T][StorageInt] Device read: block 1, off 104, buffer: 20007538, size 16, translated address: 080A8068
37913 [T][StorageInt] Device read: block 1, off 120, buffer: 20007538, size 16, translated address: 080A8078
37918 [T][StorageInt] Device read: block 1, off 136, buffer: 20007538, size 16, translated address: 080A8088
37924 [T][StorageInt] Device read: block 1, off 152, buffer: 20007538, size 16, translated address: 080A8098
37929 [T][StorageInt] Device read: block 1, off 144, buffer: 20007538, size 16, translated address: 080A8090
37934 [T][StorageInt] Device read: block 1, off 160, buffer: 20007538, size 16, translated address: 080A80A0
37939 [T][StorageInt] Device read: block 1, off 176, buffer: 20007538, size 16, translated address: 080A80B0
37945 [T][StorageInt] Device read: block 1, off 192, buffer: 20007538, size 16, translated address: 080A80C0
37950 [T][StorageInt] Device read: block 1, off 184, buffer: 20007538, size 16, translated address: 080A80B8
37955 [T][StorageInt] Device read: block 1, off 200, buffer: 20007538, size 16, translated address: 080A80C8
37960 [T][StorageInt] Device read: block 1, off 216, buffer: 20007538, size 16, translated address: 080A80D8
37966 [T][StorageInt] Device read: block 1, off 232, buffer: 20007538, size 16, translated address: 080A80E8
37971 [T][StorageInt] Device read: block 1, off 248, buffer: 20007538, size 16, translated address: 080A80F8
37976 [T][StorageInt] Device read: block 1, off 240, buffer: 20007538, size 16, translated address: 080A80F0
37982 [T][StorageInt] Device read: block 1, off 256, buffer: 20007538, size 16, translated address: 080A8100
37987 [T][StorageInt] Device read: block 1, off 272, buffer: 20007538, size 16, translated address: 080A8110
37992 [T][StorageInt] Device read: block 1, off 288, buffer: 20007538, size 16, translated address: 080A8120
37997 [T][StorageInt] Device read: block 1, off 280, buffer: 20007538, size 16, translated address: 080A8118
38003 [T][StorageInt] Device read: block 1, off 296, buffer: 20007538, size 16, translated address: 080A8128
38008 [T][StorageInt] Device read: block 1, off 312, buffer: 20007538, size 16, translated address: 080A8138
38013 [T][StorageInt] Device read: block 1, off 328, buffer: 20007538, size 16, translated address: 080A8148
38018 [T][StorageInt] Device read: block 1, off 344, buffer: 20007538, size 16, translated address: 080A8158
38024 [T][StorageInt] Device read: block 1, off 360, buffer: 20007538, size 16, translated address: 080A8168
38029 [T][StorageInt] Device read: block 1, off 376, buffer: 20007538, size 16, translated address: 080A8178
38034 [T][StorageInt] Device read: block 1, off 392, buffer: 20007538, size 16, translated address: 080A8188
38039 [T][StorageInt] Device read: block 1, off 408, buffer: 20007538, size 16, translated address: 080A8198
38045 [T][StorageInt] Device read: block 1, off 424, buffer: 20007538, size 16, translated address: 080A81A8
38050 [T][StorageInt] Device read: block 1, off 440, buffer: 20007538, size 16, translated address: 080A81B8
38055 [T][StorageInt] Device read: block 1, off 456, buffer: 20007538, size 16, translated address: 080A81C8
38060 [T][StorageInt] Device read: block 1, off 472, buffer: 20007538, size 16, translated address: 080A81D8
38066 [T][StorageInt] Device read: block 1, off 488, buffer: 20007538, size 16, translated address: 080A81E8
[... 985 lines cut ...]
54974 [T][ApiHashtable] Can't find symbol with hash dda0fada @ 2001C954!
[... cut ...]
55019 [T][StorageApi] File/Dir 0000F7E0 free
55021 [D][NfcSupportedCards] Loaded 17 plugins
55034 [T][ISO14443_3A] Sel resp: 24
55038 [T][ISO14443_3A] Sel resp: 20
55040 [T][ISO14443_3A] Col resolution complete
55044 [D][Iso14443_4aPoller] Read ATS success
55054 [D][MfDesfirePoller] Read version success
55059 [D][MfDesfirePoller] Read free memory success
55063 [D][MfDesfirePoller] Read master key settings success
55068 [D][MfDesfirePoller] Read master key version success
55073 [D][MfDesfirePoller] Read application ids success
```

KaliStudio, 2024-02-11:
Maybe wrong information on the SD card?

I-Am-Skoot, 2024-02-12:
> Maybe wrong information on the SD card?

I have wiped the SD card a couple times, and re-initialized it.
I have also replaced the SD card with another one as well.

Same results.

I-Am-Skoot, 2024-02-12:
> 26695 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded

Just updated to 389AFFD9
Hard lockup with Don't Move on the screen

```

              _.-------.._                    -,
          .-"```"--..,,_/ /`-,               -,  \
       .:"          /:/  /'\  \     ,_...,  `. |  |
      /       ,----/:/  /`\ _\~`_-"`     _;
     '      / /`"""'\ \ \.~`_-'      ,-"'/
    |      | |  0    | | .-'      ,/`  /
   |    ,..\ \     ,.-"`       ,/`    /
  ;    :    `/`""\`           ,/--==,/-----,
  |    `-...|        -.___-Z:_______J...---;
  :         `                           _-'
 _L_  _     ___  ___  ___  ___  ____--"`___  _     ___
| __|| |   |_ _|| _ \| _ \| __|| _ \   / __|| |   |_ _|
| _| | |__  | | |  _/|  _/| _| |   /  | (__ | |__  | |
|_|  |____||___||_|  |_|  |___||_|_\   \___||____||___|

Welcome to Flipper Zero Command Line Interface!
Read the manual: https://docs.flipper.net/development/cli
Run `help` or `?` to list available commands

Firmware version: dev unknown (389affd9 built on 12-02-2024)

>: log debug
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
152288 [I][Dolphin] Flush stats
152289 [I][SavedStruct] Saving "/int/.dolphin.state"
152301 [D][StorageInt] Device erase: page 2, translated page: a8
152867 [D][StorageInt] Device sync: skipping
152873 [I][Loader] Loading /ext/apps/NFC/nfc.fap
152877 [I][DolphinState] State saved
153108 [I][Elf] Total size of loaded sections: 65521
153111 [I][Loader] Loaded in 242ms
153116 [I][AnimationManager] Unload animation 'L1_Tv_128x47'
154151 [D][NfcScanner] Found 5 base protocols
154157 [D][DolphinState] icounter 31, butthurt 13
154170 [D][Nfc] FWT Timeout
154197 [D][Nfc] FWT Timeout
154222 [D][Nfc] FWT Timeout
154274 [D][Nfc] FWT Timeout
154302 [D][Nfc] FWT Timeout
154325 [D][Nfc] FWT Timeout
154350 [D][Nfc] FWT Timeout
154386 [D][Nfc] FWT Timeout
154439 [D][Nfc] FWT Timeout
154467 [D][Nfc] FWT Timeout
154490 [D][Nfc] FWT Timeout
154515 [D][Nfc] FWT Timeout
154540 [D][Nfc] FWT Timeout
154592 [D][Nfc] FWT Timeout
154620 [D][Nfc] FWT Timeout
154643 [D][Nfc] FWT Timeout
154668 [D][Nfc] FWT Timeout
154693 [D][Nfc] FWT Timeout
154756 [D][Nfc] FWT Timeout
154784 [D][Nfc] FWT Timeout
154807 [D][Nfc] FWT Timeout
154832 [D][Nfc] FWT Timeout
154857 [D][Nfc] FWT Timeout
154909 [D][Nfc] FWT Timeout
154937 [D][Nfc] FWT Timeout
154960 [D][Nfc] FWT Timeout
154985 [D][Nfc] FWT Timeout
[... 61 lines cut ...]
156971 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157020 [I][Elf] Total size of loaded sections: 1176
157023 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157078 [I][Elf] Total size of loaded sections: 699
157081 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157132 [I][Elf] Total size of loaded sections: 1419
157135 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157186 [I][Elf] Total size of loaded sections: 1908
157189 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157241 [I][Elf] Total size of loaded sections: 2376
157244 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157296 [I][Elf] Total size of loaded sections: 1588
157302 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157356 [I][Elf] Total size of loaded sections: 652
157359 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157415 [I][Elf] Total size of loaded sections: 1104
157418 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
157422 [D][NfcSupportedCards] Loaded 17 plugins
157439 [D][Iso14443_4aPoller] Read ATS success
157449 [D][MfDesfirePoller] Read version success
157454 [D][MfDesfirePoller] Read free memory success
157458 [D][MfDesfirePoller] Read master key settings success
157463 [D][MfDesfirePoller] Read master key version success
157468 [D][MfDesfirePoller] Read application ids success
```

skotopes, 2024-02-12:
Looks like you got something really funky on your card. Do you have our wifi board or any other supported debugger?

I-Am-Skoot, 2024-02-12:
> Looks like you got something really funky on your card. Do you have our wifi board or any other supported debugger?

I do not have a wifi board or other debugger yet, but this crash is not only on this nfc card can, it is also on other apps too.   This card read fine on another flipper zero.  it is just this one.

Using Cross Remote, adding a new command chain.  The 11th IR entry crashed and rebooted this flipper with the latest prod Firmware.

Log File 
```

              _.-------.._                    -,
          .-"```"--..,,_/ /`-,               -,  \
       .:"          /:/  /'\  \     ,_...,  `. |  |
      /       ,----/:/  /`\ _\~`_-"`     _;
     '      / /`"""'\ \ \.~`_-'      ,-"'/
    |      | |  0    | | .-'      ,/`  /
   |    ,..\ \     ,.-"`       ,/`    /
  ;    :    `/`""\`           ,/--==,/-----,
  |    `-...|        -.___-Z:_______J...---;
  :         `                           _-'
 _L_  _     ___  ___  ___  ___  ____--"`___  _     ___
| __|| |   |_ _|| _ \| _ \| __|| _ \   / __|| |   |_ _|
| _| | |__  | | |  _/|  _/| _| |   /  | (__ | |__  | |
|_|  |____||___||_|  |_|  |___||_|_\   \___||____||___|

Welcome to Flipper Zero Command Line Interface!
Read the manual: https://docs.flipper.net/development/cli
Run `help` or `?` to list available commands

Firmware version: 0.98.2 0.98.2 (9744fd8f built on 01-02-2024)

>: log debug
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
13203004 [I][Loader] Loading /ext/apps/Infrared/infrared.fap
13203117 [I][Elf] Total size of loaded sections: 26512
13203120 [I][Loader] Loaded in 116ms
13203124 [I][AnimationManager] Unload animation 'L1_Mad_fist_128x64'
13204377 [I][FuriThread] Infrared allocation balance: 0
13204390 [I][Loader] App returned: 0
13204395 [I][Loader] Application stopped. Free heap: 141552
13204588 [I][AnimationManager] Select 'L1_Mad_fist_128x64' animation
13204594 [I][AnimationManager] Load animation 'L1_Mad_fist_128x64'
13204600 [I][SavedStruct] Loading "/int/.desktop.settings"
13207928 [D][BrowserWorker] Start
13207943 [D][BrowserWorker] Enter folder: /ext/apps items: 9 idx: -1
13207947 [D][BrowserWorker] Load offset: 0 cnt: 50
13209980 [D][BrowserWorker] Enter folder: /ext/apps/Infrared items: 2 idx: -1
13209984 [D][BrowserWorker] Load offset: 0 cnt: 50
13212966 [D][BrowserWorker] End
13212989 [D][DolphinState] icounter 34, butthurt 12
13212992 [I][Loader] Loading /ext/apps/Infrared/xremote.fap
13213083 [I][Elf] Total size of loaded sections: 22387
13213086 [I][Loader] Loaded in 94ms
13213091 [I][AnimationManager] Unload animation 'L1_Mad_fist_128x64'
13213118 [I][SubGhzSetting] File is not used /ext/subghz/assets/setting_user
13213124 [I][SubGhzKeystore] Loading keystore /ext/subghz/assets/keeloq_mfcodes
13213192 [I][SubGhzKeystore] Loading keystore /ext/subghz/assets/keeloq_mfcodes_user
13213200 [E][SubGhzKeystore] Unable to open file for read: /ext/subghz/assets/keeloq_mfcodes_user
13213206 [I][SubGhzProtocoNiceFlorS] Loading rainbow table from /ext/subghz/assets/nice_flor_s
13213210 [I][SubGhzProtocoCameAtomo] Loading rainbow table from /ext/subghz/assets/came_atomo
13213216 [I][SubGhzProtocoAlutechAt4n] Loading rainbow table from /ext/subghz/assets/alutech_at_4n
13213226 [D][PluginManager] Loading /any/apps_data/subghz/plugins/radio_device_cc1101_ext.fal
13213282 [I][Elf] Total size of loaded sections: 4472
13213285 [D][Fap] Library for subghz_radio_device, API v. 1 loaded
13213289 [D][Fap] Library for subghz_radio_device, API v. 1 loaded
13213292 [I][SubGhzDeviceRegistry] Loaded 2 radio device
13213669 [E][SubGhzDeviceCc1101Ext] Init failed
13213672 [D][SubGhz] completed TXRX alloc
13213674 [D][INT_INPUT] Resetting Model
[... 35 lines cut ...]
13269004 [D][BrowserWorker] End
13269016 [I][XremoteInfraredRemote] loading IR Remote: '/any/infrared/LEDs.ir'
13276113 [D][BrowserWorker] Start
13276120 [D][BrowserWorker] Enter folder: /any/infrared items: 2 idx: -1
13276124 [D][BrowserWorker] Load offset: 0 cnt: 50
13277157 [D][BrowserWorker] End
13277169 [I][XremoteInfraredRemote] loading IR Remote: '/any/infrared/LEDs.ir'
13282757 [D][BrowserWorker] Start
13282764 [D][BrowserWorker] Enter folder: /any/infrared items: 2 idx: -1
13282768 [D][BrowserWorker] Load offset: 0 cnt: 50
13284060 [D][BrowserWorker] End
13284072 [I][XremoteInfraredRemote] loading IR Remote: '/any/infrared/LEDs.ir'
13291718 [D][BrowserWorker] Start
13291725 [D][BrowserWorker] Enter folder: /any/infrared items: 2 idx: -1
13291729 [D][BrowserWorker] Load offset: 0 cnt: 50
13291949 [D][BtGap] set_non_discoverable success
13292994 [D][BrowserWorker] End
13293006 [I][XremoteInfraredRemote] loading IR Remote: '/any/infrared/LEDs.ir'
13298880 [D][BrowserWorker] Start
13298887 [D][BrowserWorker] Enter folder: /any/infrared items: 2 idx: -1
13298891 [D][BrowserWorker] Load offset: 0 cnt: 50
13300435 [D][BrowserWorker] End
13300447 [I][XremoteInfraredRemote] loading IR Remote: '/any/infrared/LEDs.ir'
13306005 [D][BrowserWorker] Start
```

I-Am-Skoot, 2024-02-13:
> @I-Am-Skoot we need backtrace and full logs. Full logs are only available on GPIO pins, backtrace only with debugger.
> 
> @doomwastaken try to reproduce it please

I have ordered the Wifi Debug board.  Will reproduce one it arrives.

I-Am-Skoot, 2024-02-13:
[flipper_2024_02_13_11-55.log](https://github.com/flipperdevices/flipperzero-firmware/files/14270985/flipper_2024_02_13_11-55.log)
I hooked up a serial monitor to the UART pins on the flipper, ran it in trace mode, and exported it.   Does this help?

I-Am-Skoot, 2024-02-13:
[flipper_2024_02_13_14.log](https://github.com/flipperdevices/flipperzero-firmware/files/14272247/flipper_2024_02_13_14.log)
Here is another capture from it crashing in Cross Remote.

I-Am-Skoot, 2024-02-20:
Updatee:   I just got the wifi debug module, and looks like the device logs are the same as when I connected to the serial pins.    

NFC
```
06835 [D][MfDesfirePoller] Read application ids success

[CRASH][NfcWorker] furi_check failed
        r0 : 0
        r1 : 20000f30
        r2 : 20000284
        r3 : 0
        r4 : 0
        r5 : 0
        r6 : 0
        r7 : 2001dd28
        r8 : 0
        r9 : 20000f04
        r10 : 0
        r11 : 20031330
        lr : 801213b
        stack watermark: 7632
             heap total: 185968
              heap free: 57024
         heap watermark: 53720
        core2: not faulted
Rebooting system<FF>0 [I][FuriHalRtc] Init OK
```

Cross Remote
```
142815 [I][XremoteInfraredRemote] loading IR Remote: '/any/infrared/LEDs.ir'

[CRASH][DialogsSrv] furi_check failed
        r0 : 0
        r1 : 20000f30
        r2 : 20000284
        r3 : 0
        r4 : 0
        r5 : 0
        r6 : 800
        r7 : 968
        r8 : 0
        r9 : 20000f04
        r10 : 808
        r11 : 20031330
        lr : 801213b
        stack watermark: 624
             heap total: 185968
              heap free: 2408
         heap watermark: 1648
        core2: not faulted
Rebooting system<FF>0 [I][FuriHalRtc] Init OK
```

skotopes, 2024-03-05:
@I-Am-Skoot can you use debugger to get stack trace?

Kuronons, 2024-03-10:
So I have the same issue and I tested on latest official DEV (Flip is frozen on NFC menu selection) but also RC 0.99 (Flip display error screen - cf screenshot). Not occuring on RELEASE 0.98.3.
Note that it also occurs on a custom firmware with an OUT OF MEMORY error screen.

[image]

Happens when the flip is set to play a certain pack of animations I made. (most probably happens with other anims or larger sets) 
I suppose, in my case, that those anims are a bit "heavier" than the ones I usually make because I don't have the issue with another set I did for instance.
It seems to me that the Flip runs out of memory, as if the RAM used to play anims is not cleared at entering NFC app.

To reproduce, and launch tests, simply add my set of anim (zip on GH link - manifest also included for quick test) to SD/dolphin/, reboot and try to access NFC app... :
https://github.com/Kuronons/FZ_graphics/blob/main/Animations/SF_Corporations_Logos/Kuronons_SFlogo_Collection%20(12%20animations).zip

To what I could hear, this is happening lately to many people, most of them have large custom anim packs set on their Flip.

Kuronons, 2024-03-10:
Complementary info.

On RC 0.99 build :
As I let my Flip alone for some time, the issue didn t occured when I tried NFC.

So to test, I rebooted OS, waited a few seconds and tried to enter NFC app -> crash.

My guess is that until Flip is loading anims in RAM, this will occur. Once all is fully loaded, it seems to be fine.
So the problem will occur each time an animation will switch, especially if the new displayed one is kinda "heavy".

Hope those feedbacks help finding a fix ! :)

PS : Maybe this should go a new issue thread... I'll create one if needed. Just lmk.

skotopes, 2024-03-11:
Yes, we do have problem with animations been unloaded too late, so app crashes because of not enough RAM. Will be fixed in next release

Kuronons, 2024-03-11:
> Yes, we do have problem with animations been unloaded too late, so app crashes because of not enough RAM. Will be fixed in next release

Many thanks for the prompt reply ! That's good news that you identified the issue and that it will be fixed soon !
Any idea when can be expected (roughly) the next release ?

skotopes, 2024-03-11:
in 2 weeks

Kuronons, 2024-04-02:
> in 2 weeks

Hey !

Wondering if there was already a commit pushed for this particular fix ? If so, could you point me to it ?
(didn t see anything related so far but I may have missed it)

Many thanks in advance !

skotopes, 2024-04-03:
> > in 2 weeks
> 
> Hey !
> 
> Wondering if there was already a commit pushed for this particular fix ? If so, could you point me to it ? (didn t see anything related so far but I may have missed it)
> 
> Many thanks in advance !

Still work in progress

gornekich, 2024-04-17:
Hello! The fix arrived in dev. Please update to latest dev and try again

ferretking, 2024-04-22:
Reading NFC Mifare DESFire cards still crashes with firmwares: 100.3 and 101.0-RC. Firmware Dev-c8768359 causes the read to froze on "Don't Move" the led blinks green but nothing.

skotopes, 2024-04-23:
@ferretking We need stack trace from your device. Do you have any supported ISP? Flipper WiFi board or st-link or jlink?

gornekich, 2024-04-23:
@ferretking also flipper logs will be useful

darryllee, 2024-06-08:
Ohey, I'm on latest Dev FW (ef0300d2 built on 08-06-2024). NFC can read various credit cards and also my iPhone Apple Pay just fine. But when I try to read an old hotel card (Disney Grand Californian), it crashes. (Stuck blinking green, have to reboot.)

Here's a log of a successful scan of the Clipper Card in my Apple Wallet and then the failure I'm seeing with the hotel card. Appreciate any ideas!

```
216045 [D][NfcScanner] Found 4 children
216086 [D][Nfc] FWT Timeout
216089 [D][Nfc] FWT Timeout
216122 [D][Nfc] FWT Timeout
216155 [D][Iso14443_4aPoller] Read ATS success
216201 [I][NfcScanner] Detected 1 protocols
216330 [D][Iso14443_4aPoller] Read ATS success
216350 [D][MfDesfirePoller] Read version success
216357 [D][MfDesfirePoller] Read free memory success
216364 [D][MfDesfirePoller] Read master key settings success
216372 [D][MfDesfirePoller] Read master key version success
216380 [D][MfDesfirePoller] Read application ids success
216382 [D][MfDesfirePoller] Selecting app 0
216390 [D][MfDesfirePoller] Reading app 0
216576 [D][MfDesfirePoller] Can't read file 6 data without authentication
216579 [D][MfDesfirePoller] Can't read file 7 data without authentication
216888 [D][MfDesfirePoller] Read applications success
216890 [D][MfDesfirePoller] Read success.
216893 [D][Nfc] FWT Timeout
216962 [I][Elf] Total size of loaded sections: 3428
216965 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
217127 [D][DolphinState] icounter 12, butthurt 0
227557 [D][ViewDispatcher] View changed while key press 20011E20 -> 20011730. Sending key: Back, type: Release, sequence: 000000D0 to previous view port
233160 [D][NfcScanner] Found 5 base protocols
233166 [D][DolphinState] icounter 13, butthurt 0
233203 [D][Nfc] FWT Timeout
233228 [D][Nfc] FWT Timeout
233279 [D][Nfc] FWT Timeout
233307 [D][Nfc] FWT Timeout
233323 [D][NfcScanner] Found 4 children
233378 [D][Nfc] FWT Timeout
233380 [D][Nfc] FWT Timeout
233413 [D][Nfc] FWT Timeout
233446 [D][Iso14443_4aPoller] Read ATS success
233478 [I][NfcScanner] Detected 1 protocols
233613 [D][Iso14443_4aPoller] Read ATS success
233621 [D][MfDesfirePoller] Read version success
233625 [D][MfDesfirePoller] Read free memory success
233629 [D][MfDesfirePoller] Read master key settings success
233633 [D][MfDesfirePoller] Read master key version success
233638 [D][MfDesfirePoller] Read application ids success
233641 [D][MfDesfirePoller] Selecting app 0
233645 [D][MfDesfirePoller] Reading app 0
233657 [D][MfDesfirePoller] Can't read file 0 data without authentication
233660 [D][MfDesfirePoller] Selecting app 1
233665 [D][MfDesfirePoller] Reading app 1
233681 [D][MfDesfirePoller] Can't read file 1 data without authentication
233684 [D][MfDesfirePoller] Selecting app 2
233689 [D][MfDesfirePoller] Reading app 2
```