flipperdevices/flipperzero-firmware #3441: read transport card in France

KaliStudio, 2024-02-11:
### Description of the feature you're suggesting.

I would like to be able to read my French transport ticket and why not modify it, here are 4 dumps, the first three sound of the same title with 4, 3 and 2 trips remaining and the second there are 0 trips remaining.

[Liatick10022024_18h20.nfc.txt](https://github.com/flipperdevices/flipperzero-firmware/files/14233522/Liatick10022024_18h20.nfc.txt)
[Liatick11022024_17h18.nfc.txt](https://github.com/flipperdevices/flipperzero-firmware/files/14233523/Liatick11022024_17h18.nfc.txt)
[Lia_10voy.nfc.txt](https://github.com/flipperdevices/flipperzero-firmware/files/14233524/Lia_10voy.nfc.txt)
[Liatick01022024_16h30.nfc.txt](https://github.com/flipperdevices/flipperzero-firmware/files/14233525/Liatick01022024_16h30.nfc.txt)

skotopes, 2024-02-12:
And we can't stop you from doing that. Just be responsible and report any found vulnerabilities to authorities.

closed 2024-02-12 by skotopes

KaliStudio, 2024-02-12:
This is a feature request, but if you don't want to, can you give me an example of how to do it?

skotopes, 2024-02-12:
@KaliStudio https://www.youtube.com/watch?v=1JT_lTfK69Q also make sure that you read https://en.wikipedia.org/wiki/Massachusetts_Bay_Transportation_Authority_v._Anderson first.