flipperdevices/flipperzero-firmware #3443: Toolchain version is not supported.

KaliStudio, 2024-02-12:
### Describe the bug.

I get the following error:

### Reproduction

I re-downloaded the github: git clone --recursive https://github.com/flipperdevices/flipperzero-firmware.git
Build firmware using Flipper Build Tool: ./fbt

### Logs

```Text
scons: *** Toolchain version is not supported. Allowed: (' 12.3.', ' 13.2.'), toolchain:    Stop.
```

hedger, 2024-02-12:
What's your OS and environment? Was it working before?

KaliStudio, 2024-02-12:
I am on windows. yes it worked before : New toolchain with gcc 12

hedger, 2024-02-12:
Please manually remove "toolchain" directory from firmware folder and try again.

KaliStudio, 2024-02-12:
`PS C:\flipperzero-firmware> ./fbt
Downloading Windows toolchain..done!

    Répertoire : C:\flipperzero-firmware

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----        12/02/2024     11:21                toolchain
Extracting Windows toolchain..moving..done!
Cleaning up temporary files..done!

scons: *** Toolchain version is not supported. Allowed: (' 12.3.', ' 13.2.'), toolchain:    Stop.
PS C:\flipperzero-firmware>`

hedger, 2024-02-12:
Works for me.
```
D:\Work\flipperzero-firmware>fbt
Downloading Windows toolchain..done!

    Directory: D:\Work\flipperzero-firmware

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----        12.02.2024     10:26                toolchain
Extracting Windows toolchain..moving..done!
Cleaning up temporary files..done!

fbt: warning: App folder 'D:\Work\flipperzero-firmware\applications\external': missing manifest (application.fam)
LoadAppManifest, line 130, in file "D:\Work\flipperzero-firmware\scripts\fbt_tools\fbt_apps.py"
        SDKCHK  targets\f7\api_symbols.csv
API version 55.2 is up to date
        AR      build\f7-firmware-D\mjs\libmjs.a
...
```

hedger, 2024-02-12:
Please dump file tree in firmware folder with `tree > files.log` command and attach it here.

KaliStudio, 2024-02-12:
[files.log](https://github.com/flipperdevices/flipperzero-firmware/files/14237299/files.log)

KaliStudio, 2024-02-12:
After a long search, I understood that the libwinpthread-1.dll file was missing, I downloaded it and put it in: C:\Windows\System32 :
[libwinpthread-1.zip](https://github.com/flipperdevices/flipperzero-firmware/files/14239628/libwinpthread-1.zip)

CookiePLMonster, 2024-02-12:
> After a long search, I understood that the libwinpthread-1.dll file was missing, I downloaded it and put it in: C:\Windows\System32 : [libwinpthread-1.zip](https://github.com/flipperdevices/flipperzero-firmware/files/14239628/libwinpthread-1.zip)

That's a horrible idea, you should never put randomly downloaded files in system32.

A quick search on my PC reveals that this DLL is shipped by many applications
[image]
so I suspect that it worked for @hedger but not you because it gets loaded from some PATH location. Flipper's toolchain should ship it instead.

KaliStudio, 2024-02-12:
Sorry, I didn't specify that the file comes from https://sourceforge.net/projects/mingw-w64/
I just shared the missing dll

CookiePLMonster, 2024-02-12:
> looking at your search I see that I must have a problem with Git

I don't think Git should be a pre-requisite for using the toolchain, it's most likely a legitimate missing file in the firmware toolchain.

hedger, 2024-02-13:
Yep, we've also identified missing libraries. We'll release a new toolchain version with fixes soon.
Thanks for the report.

AdonaiDiazEsparza, 2024-02-13:
> > After a long search, I understood that the libwinpthread-1.dll file was missing, I downloaded it and put it in: C:\Windows\System32 : [libwinpthread-1.zip](https://github.com/flipperdevices/flipperzero-firmware/files/14239628/libwinpthread-1.zip)
> 
> That's a horrible idea, you should never put randomly downloaded files in system32. If you really have to, put it in the toolchain directory and not system32.
> 
> A quick search on my PC reveals that this DLL is shipped by many applications ![image](https://private-user-images.githubusercontent.com/7947461/304155466-81918934-f222-4d01-963d-dfc31826f1d1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc4NTUwMzYsIm5iZiI6MTcwNzg1NDczNiwicGF0aCI6Ii83OTQ3NDYxLzMwNDE1NTQ2Ni04MTkxODkzNC1mMjIyLTRkMDEtOTYzZC1kZmMzMTgyNmYxZDEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTNUMjAwNTM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTkwYmM5NjMwMmYyNWRmMGVjY2YzMDQwZDI4MmI2MGM1NzE3NmEwNTc0MmU1OTQ4ZTg1OWU1NzRlNThjMmM0MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ar7UrCXQLHh_X1bU-JmogDDN3uA-taXWQXhUa9dcVQY) so I suspect that it worked for @hedger but not you because it gets loaded from some PATH location. Flipper's toolchain should ship it instead.

Hi, where can I find the Toolchain directory? I have the same issue, windows is warning that dll is missing

hedger, 2024-02-13:
@AdonaiDiazEsparza put 64-bit mingw .dlls in `toolchain/x86_64-windows/bin` folder.

Willy-JL, 2024-02-14:
FWIW ive seen `libstdc++-6.dll` and `libgcc_s_seh-1.dll` missing aswell, in this case for `protoc.exe`.

thanks for looking into this!

Kingwad, 2024-02-15:
Had the same issue, toolchain version 31 corrects the issue and builds.

Riamonos, 2024-02-15:
hola buenas tardes tengo el mismo problema y no se como solucionarlo 
scons: *** Toolchain version is not supported. Allowed: (' 12.3.', ' 13.2.'), toolchain:    Stop.
podria alguien ayudarme es mis primeros pasos con el flipper y tengo muchas ganas de aprender 
gracias de ante mano

Jigbo, 2024-02-18:
hey so, what is the solution here? i get this error,
scons: *** Toolchain version is not supported. Allowed: (' 12.3.', ' 13.2.'), toolchain:    Stop.
 delete the toolchain as another user suggested, and get another error
[image]

Willy-JL, 2024-02-23:
the fix is #3451 , you can wait until it is merged into dev branch here, or switch to `hedger/toochain_29` branch as that is what the PR fixing the issues is based on

AdonaiDiazEsparza, 2024-02-23:
> hola buenas tardes tengo el mismo problema y no se como solucionarlo scons: *** Toolchain version is not supported. Allowed: (' 12.3.', ' 13.2.'), toolchain: Stop. podria alguien ayudarme es mis primeros pasos con el flipper y tengo muchas ganas de aprender gracias de ante mano
> 
> por cierto descargue mingv y lo instale pero no se si tengo que seleccionar mas dll

Hola ya encontré algo que puede arreglar este error, resulta que esto es provocado a falta de una dll, que podemos obtener al descargar un GNU Arm Toolchain en esta página [,](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads), busca el archivo "arm-gnu-toolchain-13.2.rel1-mingw-w64-i686-arm-none-eabi.exe", lo descargas, lo instalas y después haces de nuevo el proceso de /fbt en el powershell o en el cmd (para windows).

Hi I found something could fix this error, there's missing a dll that we can get downloading a GNU Arm Toolchain in this page [,](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads), search the file "arm-gnu-toolchain-13.2.rel1-mingw-w64-i686-arm-none-eabi.exe", download it, install and after do again the same process of th /fbt in the powershell or prompt cmd (for windows)

Riamonos, 2024-02-23:
ahora me da este errror 
Excepción al llamar a "ExtractToDirectory" con los argumentos "2": "No se encuentra el registro de fin de directorio
central."
En C:\Users\riamo\Desktop\Nueva\flipperzero-firmware\scripts\toolchain\windows-toolchain-download.ps1: 47 Carácter: 1
+ [System.IO.Compression.ZipFile]::ExtractToDirectory("$toolchain_zip_t ...
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : InvalidDataException

AdonaiDiazEsparza, 2024-02-23:
mmmm que raro, a mi me resolvió ese problema, trata de hacerlo en otra carpeta el procedimiento de clonar el repositorio y activar el /fbt, también si tienes antivirus desactívalo, y si no funciona, puedo subir una carpeta con el proyecto ya con el build y las extensiones, a mi me funcionó eso con un compañero del trabajo que me paso a través de una memoria usb su carpeta donde estaba programando, ya hasta hace poco pude resolver ese problema con el GNU Arm Tool Chain.

AdonaiDiazEsparza, 2024-02-23:
> ahora me da este errror Excepción al llamar a "ExtractToDirectory" con los argumentos "2": "No se encuentra el registro de fin de directorio central." En [...]

mmmm que raro, a mi me resolvió ese problema, trata de hacerlo en otra carpeta el procedimiento de clonar el repositorio y activar el /fbt, también si tienes antivirus desactívalo, y si no funciona, puedo subir una carpeta con el proyecto ya con el build y las extensiones, a mi me funcionó eso con un compañero del trabajo que me paso a través de una memoria usb su carpeta donde estaba programando, ya hasta hace poco pude resolver ese problema con el GNU Arm Tool Chain.

Riamonos, 2024-02-23:
hecho un ufbt para actualizar el toolchain y ahora me da este error 

scons: Entering directory `C:\Users\riamo\.ufbt\current\scripts\ufbt'

fbt: warning: App folder 'C:\Users\riamo\Desktop\Nueva\flipperzero-firmware': missing manifest (application.fam)
LoadAppManifest, line 130, in file "C:\Users\riamo\.ufbt\current\scripts\fbt_tools\fbt_apps.py"
scons: *** No targets specified and no Default() targets found.  Stop.
Found nothing to build

Riamonos, 2024-02-23:
> mmmm que raro, a mi me resolvió ese problema, trata de hacerlo en otra carpeta el procedimiento de clonar el repositorio y activar el /fbt, también si tienes antivirus desactívalo, y si no funciona, [...]

dmitriy-belkin, 2024-02-23:
same problem((

AdonaiDiazEsparza, 2024-02-23:
> hecho un ufbt para actualizar el toolchain y ahora me da este error
> 
> scons: Entering directory `C:\Users\riamo.ufbt\current\scripts\ufbt'
> 
> fbt: warning: App folder 'C:\Users\riamo\Desktop\Nueva\flipperzero-firmware': missing manifest (application.fam) LoadAppManifest, line 130, in file "C:\Users\riamo.ufbt\current\scripts\fbt_tools\fbt_apps.py" scons: *** No targets specified and no Default() targets found. Stop. Found nothing to build
> 
> creo que hare un nuevo git clone

me avisas sino subo una carpeta con el proyecto al cabo creo que si se puede mover, sino ya no sabría como ayudarte :(

Riamonos, 2024-02-23:
muchas gracias adonai , tu que parte de flipper quieres estudiar

AdonaiDiazEsparza, 2024-02-23:
> muchas gracias adonai , tu que parte de flipper quieres estudiar

estoy haciendo mis practicas profesionales donde ando trabajando en un proyecto de Sniffing para una red CAN-Bus

El link del archivo está aquí: [link](https://drive.google.com/file/d/1PTF8yJGOkOKIwF3Y1gEyImeYJzO7lBxY/view?usp=sharing)}
solo ignora una carpeta que dice Scenes, son ejemplos de flipper Zero

CookiePLMonster, 2024-02-26:
This is fixed by https://github.com/flipperdevices/flipperzero-firmware/pull/3451 and can be closed.

hedger, 2024-03-01:
Closing, fixed in #3451.

closed 2024-03-01 by hedger