flipperdevices/flipperzero-firmware #3454: NFC (Bus card) Cant be emulated. Only scanned.

Killer74-hub, 2024-02-16:
### Describe the bug.

The Card is this card: https://www.rmv.de/c/en/tickets/your-ticket/tickets-overview/annual-season-tickets

It is a german Bus ticket.

### Reproduction

1. Clone Card
2. Save the Card/Emulate Directly
3. Emulate the card

### Target

German Bus Card(NFC)

### Logs

```Text
Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB
Device type: ISO14443-4A

(I will give you the rest through discord if you need it. contact me.)
```

### Anything else?

Contact me through Discord if you want the NFC Card number. I don´t want to give it to the public. 

discord: sniper74

mxcdoam, 2024-02-16:
It seems like f0 doesn't support writing operations on emulated card. So it will only emulate "frozen" card state, the reader will be unable to update emulated card's memory and the transaction will fail. 
By the way, this also affects intercoms in my area: reader writes new data on each pass, if it is unable to write - key is declined.

mxcdoam, 2024-02-17:
It might be not broken, but intended that way. Storing backups of transit passes on a pocket device might sound like a nice idea, but it usually goes against most transit authorities rules of service. As for the intercom keys - I am curious too if this is possible or not.

Killer74-hub, 2024-02-17:
> It might be not broken, but intended that way. Storing backups of transit passes on a pocket device might sound like a nice idea, but it usually goes against most transit authorities rules of service. As for the intercom keys - I am curious too if this is possible or not.

Yeah sure. I mean, once its expired, It will also not work anymore. So it doesn´t matter.

skotopes, 2024-02-18:
From all possible information you gave us the most useless. What card type was it?

Killer74-hub, 2024-02-18:
> From all possible information you gave us the most useless. What card type was it?

ISO 14443-4A (Unknown)

UID: 0E 11 3B 11 78 0D 03

Historical Bytes:

RAW: 5A 43 56 44 56

ISO14443-3A data
ATOA: 00 48 SAK: 20

Let me know, what you want next. I Don´t know if i´ts a mifare or so. I am new in tghe flipper zero community. What do you still need?

(sorry for my bad english)

skotopes, 2024-02-18:
Do you have proxmark?

Killer74-hub, 2024-02-18:
> Do you have proxmark?

No

Killer74-hub, 2024-02-18:
> Do you have proxmark?

Do you need something else?

Killer74-hub, 2024-02-18:
https://www.eticket-deutschland.de/en/

maybe this helps. They do have this symbol.

skotopes, 2024-02-18:
@Killer74-hub try to use any other tool you can find. For example mobile phone apps for nfc or any other reader. They may be using different detection technic and give some hints on card type.

Killer74-hub, 2024-02-18:
I found it. it´s a: Mifare SmartMX I think

Killer74-hub, 2024-02-18:
> @Killer74-hub try to use any other tool you can find. For example mobile phone apps for nfc or any other reader. They may be using different detection technic and give some hints on card type.

https://oliver-weikert.de/smart-cards/
[image]

Killer74-hub, 2024-02-18:
> @Killer74-hub try to use any other tool you can find. For example mobile phone apps for nfc or any other reader. They may be using different detection technic and give some hints on card type.

I have NFC Tools. I will try it right now

skotopes, 2024-02-18:
We don’t support smarts yet. We will take a look as soon as get some systems with it in our hands.

Killer74-hub, 2024-02-18:
> We don’t support smarts yet. We will take a look as soon as get some systems with it in our hands.

wait. Its not I think. My app says something different

Killer74-hub, 2024-02-18:
Can I send it to you on Discord? Or is it safe for here?

skotopes, 2024-02-18:
Yes you can find me on discord, search for skotopes.

skotopes, 2024-02-18:
But keep in mind that @gornekich is nfc maintainer and you probably want to talk with him instead.

Killer74-hub, 2024-02-18:
oh ok. I´ve sended it to you. Does @gornekich also have discord?

skotopes, 2024-02-18:
Yes, but please ping him on Monday. He deserves some weekends.

Killer74-hub, 2024-02-18:
> Yes, but please ping him on Monday. He deserves some weekends.

I cannot find him on Discord. what is his username?

skotopes, 2024-02-18:
He’ll tell it on Monday.

Killer74-hub, 2024-02-19:
> Hey @Killer74-hub . Thanks for information. We will get this card and start working on its support. However, I read datasheet and it's look like it is a well secured card, so flipper will read open data but full emulation might be impossible.

Hello! Thank´s for your reply @gornekich . It might be a Mifare SmartMx. But I don´t think so. Could I send you the screenshot that I have sended @skotopes on discord? If so, what is your discord Name?

gornekich, 2024-02-19:
> Hello! Thank´s for your reply @gornekich . It might be a Mifare SmartMx. But I don´t think so. Could I send you the screenshot that I have sended @skotopes on discord? If so, what is your discord [...]

Sure. My nick is gornekich

Killer74-hub, 2024-02-19:
> Sure. My nick is gornekich

sended you a Friend request

Killer74-hub, 2024-03-12:
> Sure. My nick is gornekich

Hello,

something new?

gornekich, 2024-04-15:
Still trying to get this card. Please, don't expect support for this card in nearest future

Killer74-hub, 2024-04-27:
> Still trying to get this card. Please, don't expect support for this card in nearest future

Okay. I am waiting. Thanks:D!

AdminEvilDRagon, 2024-09-19:
> Hey @Killer74-hub . Thanks for information. We will get this card and start working on its support. However, I read datasheet and it's look like it is a well secured card, so flipper will read open data but full emulation might be impossible.

Yo is it possible now, to scan it?

Killer74-hub, 2024-10-02:
Still Waiting. Keep me Up to Date please If you got it:)