flipperdevices/flipperzero-firmware #3458: Quitting a favorite (pinned) app very quickly crashes the OS (furi_check failed)

Hari-B6, 2024-02-17:
### Describe the bug.

When you try quitting an app very quickly _(by clicking/spamming the back button rapidly)_ the OS crashes and returns an error: **furi_check failed**. Keep in mind that the app **has** to be previously opened from the Favorites app section _(opening the app from the Apps menu would not return the same result)_.

[Logs](https://github.com/flipperdevices/flipperzero-firmware/files/14320250/22.26.12.17-02-2024.txt)

### Reproduction

1. Click the **DOWN** button to enter Favorites
2. Open any pinned app
3. Start clicking _(spamming)_ the **BACK** button rapidly until you go back to the home screen

### Logs

```Text
[22:26:12][info]: app version: 1.6.8(437)
[22:26:12][info]: log level: debug
[22:26:12][info]: connecting
[22:26:12][info]: session started
[22:26:13][info]: connected
[22:26:13][info]: synchronize
[22:26:13][debug]: > request(storage(stat(/ext/Manifest)))
[22:26:13][debug]: > request(storage(info(/int)))
[22:26:13][debug]: > request(system(property(devinfo.hardware.target)))
[22:26:13][debug]: > request(storage(read(/int/.region_data)))
[22:26:13][debug]: > request(system(setDate(2024-02-17 20:26:13 +0000)))
[22:26:13][debug]: < response(storage(stat(54936)))
[22:26:13][debug]: < response(storage(file(40 bytes)))
[22:26:14][debug]: < response(system(property(hardware.target: 7)))
[22:26:14][debug]: < response(ok)
[22:26:14][debug]: > request(system(property(devinfo.firmware.api)))
[22:26:14][debug]: < response(storage(info(28 KiB / 228 KiB)))
[22:26:14][debug]: > request(storage(info(/ext)))
[22:26:14][debug]: < response(ok)
[22:26:14][info]: syncing date: (0s)
[22:26:14][debug]: > request(storage(stat(/ext/nfc/.mfkey32.log)))
[22:26:14][debug]: < response(system(property(firmware.api.major: 54)))
[22:26:14][debug]: < response(system(property(firmware.api.minor: 0)))
[22:26:14][debug]: < response(ok)
[22:26:14][debug]: < response(storage(info(219 MiB / 29 GiB)))
[22:26:14][debug]: > request(system(property(devinfo.firmware.origin.fork)))
[22:26:14][debug]: < error(StorageError: File/Directory does not exist)
[22:26:14][debug]: > request(storage(list(/any, false, 0))
[22:26:14][debug]: < response(system(property(firmware.origin.fork: Official)))
[22:26:14][debug]: > request(system(property(devinfo.firmware.origin.git)))
[22:26:14][debug]: < response(ok)
[22:26:14][debug]: > request(storage(list(/ext/apps_manifests, true, 0))
[22:26:14][debug]: < response(storage(list([ibutton, subghz, nfc, infrared, lfrfid, favorites.txt, update, .tmp])))
[22:26:14][debug]: < response(storage(list([Manifest, apps, apps_data, badusb, dolphin, u2f, apps_manifests, wav_player])))
[22:26:14][debug]: < response(storage(list([apps_assets, unitemp])))
[22:26:14][debug]: > request(storage(list(/any/subghz, false, 0))
[22:26:14][debug]: < response(system(property(firmware.origin.git: https://github.com/flipperdevices/flipperzero-firmware)))
[22:26:14][debug]: < response(ok)
[22:26:15][debug]: < response(storage(list([paint.fim, ttt_multi.fim, arkanoid.fim, minesweeper.fim, asteroids.fim, bomberduck.fim, nupogodi.fim, doom.fim])))
[22:26:15][debug]: < response(storage(list([flappy_bird.fim, flipper_pong.fim, game_2048.fim, rubiks_cube_scrambler.fim, flipperzero_fnaf.fim, jetpack_game.fim, yatzee.fim, tetris.fim])))
[22:26:15][debug]: < response(storage(list([rock_paper_scissors.fim, slotmachine.fim, fuzzer_ibtn.fim, fuzzer_rfid.fim, servotester.fim, uart_terminal.fim, i2ctools.fim, avr_isp.fim])))
[22:26:15][debug]: < response(storage(list([signal_generator.fim, dap_link.fim, spi_mem_manager.fim, esp32_wifi_marauder.fim, esp_flasher.fim, flipperscope.fim, nrf24channelscanner.fim, xbox_controller.fim])))
[22:26:15][debug]: < response(storage(list([ir_scope.fim, morse_code.fim, wav_player.fim, music_player.fim, mfkey32.fim, dice_app.fim, picopass.fim, unitemp.fim])))
[22:26:16][debug]: < response(storage(list([pocsag_pager.fim, weather_station.fim, spectrum_analyzer.fim, calculator.fim, resistors.fim, nfc_rfid_detector.fim, multi_converter.fim, clock.fim])))
[22:26:16][debug]: < response(storage(list([mass_storage.fim, text_viewer.fim, gpio_7segment_output.fim, t_rex_runner.fim, tone_gen.fim, nfc_magic.fim, flashlight.fim])))
[22:26:16][debug]: < response(storage(list([assets, gates, tesla])))
[22:26:16][debug]: > request(storage(list(/any/lfrfid, false, 0))
[22:26:16][debug]: < response(storage(list([Home_elevator.rfid, A_batenberg_119.rfid, Momchil.rfid, Slunchev_brqg_topolite.rfid])))
[22:26:16][debug]: > request(storage(list(/any/nfc, false, 0))
[22:26:16][debug]: < response(storage(list([assets, Pw_tag.nfc])))
[22:26:16][debug]: > request(storage(list(/any/nfc, false, 0))
[22:26:16][debug]: < response(storage(list([assets, Pw_tag.nfc])))
[22:26:16][debug]: > request(storage(list(/any/infrared, false, 0))
[22:26:16][debug]: < response(storage(list([assets, 1205_ac.ir, 1205_tv.ir, Bulsatcom.ir, Home_tv.ir, Panasonic_home_tv.ir, Philips_tv.ir, Topalov_ac.ir])))
[22:26:16][debug]: < response(storage(list([Zibits_junk_bot.ir, Ziel_lamp.ir, Xmas_lights.ir])))
[... 253 lines cut ...]
[22:26:32][debug]: < response(system(property(gauge.fc: 0)))
[22:26:32][debug]: < response(system(property(gauge.otd: 0)))
[22:26:32][debug]: < response(system(property(gauge.otc: 0)))
[22:26:32][debug]: < response(system(property(gauge.sleep: 1)))
[22:26:32][debug]: < response(system(property(gauge.ocvfail: 0)))
[22:26:32][debug]: < response(system(property(gauge.ocvcomp: 1)))
[22:26:32][debug]: < response(system(property(gauge.fd: 0)))
[22:26:32][debug]: < response(system(property(gauge.dsg: 0)))
[22:26:32][debug]: < response(system(property(gauge.sysdwn: 0)))
[22:26:32][debug]: < response(system(property(gauge.tda: 0)))
[22:26:32][debug]: < response(system(property(gauge.battpres: 1)))
[22:26:33][debug]: < response(system(property(gauge.authgd: 0)))
[22:26:33][debug]: < response(system(property(gauge.ocvgd: 1)))
[22:26:33][debug]: < response(system(property(gauge.tca: 0)))
[22:26:33][debug]: < response(system(property(gauge.rsvd: 0)))
[22:26:33][debug]: < response(system(property(gauge.capacity.full: 2100)))
[22:26:33][debug]: < response(system(property(gauge.capacity.design: 2100)))
[22:26:33][debug]: < response(system(property(gauge.capacity.remain: 2100)))
[22:26:33][debug]: < response(system(property(gauge.state.charge: 100)))
[22:26:33][debug]: < response(system(property(gauge.state.health: 100)))
[22:26:33][debug]: < response(system(property(gauge.voltage: 4143)))
[22:26:33][debug]: < response(system(property(gauge.current: 0)))
[22:26:33][debug]: < response(system(property(gauge.temperature: 21)))
[22:26:33][debug]: < response(ok)
```

skotopes, 2024-02-18:
@doomwastaken would you kindly?

doomwastaken, 2024-02-18:
Does behavior change when you try another app?
I've tried this on latest release and dev and was not able to reproduce. 

Also what log level did you use? I used none/debug for testing purposes

Hari-B6, 2024-02-18:
I'm running the latest version of the official firmware (0.98.3).
My log level is **Default**.

I extracted the log using the mobile flipper app. The crash doesn't happen every time with every app. Sometimes the app would freeze the OS for a second or there won't be a crash at all. Apps that use the **BACK** button as pause don't count _(not enough time to reproduce the issue)_. It happens with official apps too _(like Snake, Music Player, Clock, and IR remote)_.

Demo video:
https://github.com/flipperdevices/flipperzero-firmware/assets/149514426/8d05122c-d15a-44d4-9d45-e20b5c3ba6a0

doomwastaken, 2024-02-19:
Thank you for details, will try to reproduce it now

hedger, 2024-02-19:
To be reassigned after triaging

Hari-B6, 2024-02-20:
Did you manage to reproduce the issue? I can change the debug settings and send you more logs.

doomwastaken, 2024-02-27:
oh I was able to reproduce it, but there is a twist! It depends on number of favorites that you have. Thank you, your video was actually really helpful

PR #3659 (2024-05-22): [FL-3775] Archive: fix condition race on exit
# What's new

- [FL-3775] Archive: fix condition race on exit, fix #3458

# Verification 

- Pin at least 6 apps in archive
- Start file upload with qFlipper in background
- Open archive, navigate to the app, start it, repeatedly press back button
- No crash should occurs

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix

[FL-3775]: https://flipperzero.atlass [... cut]
closed 2024-05-22 by hedger