flipperdevices/flipperzero-firmware #3460: NFC Emulation Time Limit

bluenazgul, 2024-02-18:
### Describe the enhancement you're suggesting.

After the NFC Refactoring there is an Emulation Time Limit of 5min, before the Refactoring there was No Time Limit.

With this Limit it is Impossible to Play an full audio Like Emulation of an SLIX-L "TommyBox".

Please add an Option to select If you want endless Emulation or 5min emulation

assasinfil, 2024-02-19:
Endless emulation is unsafe, there have been cases of severe heat from prolonged emulation.

assasinfil, 2024-02-20:
This is a feature of the unleashed firmware and the emulation time limit is disabled there by enabling debug mode.
For implementation issues, you can contact @Leptopt1los

hedger, 2024-02-20:
Not reproducible in ofw.

closed 2024-02-20 by hedger